dresden-elektronik/deconz-rest-plugin #6372: Xiaomi Aqara Plug EU lumi.plug.maeu01 shows wrong power values

grimmaldus, 2022-10-01:
## Describe the bug

Power sensors of Aqara power plugs EU show readings that are 10x higher than actual power consumption.

## Steps to reproduce the behavior

Use version deCONZ  2.18.2 with ConBee II fwversion 266B0700.

## Expected behavior

Expected value 10x lower than actual.

## Screenshots

[image]
## Environment

 - Host system: NAS
 - Running method:  Home Assistent deCONZ Add-on
 - Firmware version: (266B0700)
 - deCONZ version: (2.18.2)
 - Device: ConBee II

## deCONZ Logs

## Additional context

Smanar, 2022-10-01:
There is a DDF for this device but only enabled for some versions.

>   "comment": "DDF for device firmwares at least above 0.0.0_0022, paired with xBee I fw 0x26400500 / xBee II fw 0x266f0700 and above",
>   "matchexpr": "R.endpoints.includes(0x15) && R.endpoints.includes(0x1F)",

Do you have endpoint 0x15 and 0x1F on your ? it's visible on the GUI.
Else try direclty to edit the DDF devices\xiaomi\xiaomi_sp-euc01_smart_plug.json

```
        {
          "name": "state/power",
          "refresh.interval": 10,
          "read": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 21,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 21,
            "eval": "Item.val = Math.round(Attr.val);"
          }
        },
```

replace the "eval" line by "eval": "Item.val = Math.round(Attr.val / 10);"

Smanar, 2022-10-02:
Perhaps some infos here https://forum.phoscon.de/t/aqara-powersocket-consumption-issues/2494/5

grimmaldus, 2022-10-02:
> There is a DDF for this device but only enabled for some versions.
>
>
> Do you have endpoint 0x15 and 0x1F on your ? it's visible on the GUI. Else try direclty to edit the DDF [...]

Thank you, I've already fixed it in similar fashion. I just wanted to report it as a bug since it has introduced in 2.18.2.

Smanar, 2022-10-02:
So the patch is working ?

SwoopX, 2022-10-02:
DDF is working as expected, my Sonos SL is consuming 5W

```
"20:32:57:445": {
        "e": "changed",
        "id": "182",
        "r": "sensors",
        "state": {
            "lastupdated": "2022-10-02T18:32:57.435",
            "power": 5
        },
        "t": "event",
        "uniqueid": "54:ef:44:10:00:4d:da:0e-15-000c"
    }
```

[image]

grimmaldus, 2022-10-02:
> So the patch is working ?
My working solution was to alter the value right in the HA for the sensor. 

I wanted to do it via  ddf file within Home Assistant however there is no mapping for the addon container to this day to my knowledge (if you know any other way how to do it).  I will try to locate it right within container and alter it via DDF file and let you know.

erik3rik, 2022-10-05:
I have the exact same case, originally reported as a bug of the deCONZ addon to Home Assistant, but since that is only wrapping the deCONZ software the fault is in the underlying deCONZ rather than the addon itself.

https://github.com/home-assistant/addons/issues/2679

Power values increased with a 10x after the update, and was "fixed" by rolling back (to deCONZ 2.17.1 from 2.18.2)

[image]

grimmaldus, 2022-10-06:
> @manup so we can close this issue?

I have not have time to test ddf since I'm away from home atm but I will do it this weekend.

closed 2022-10-18 by grimmaldus