dresden-elektronik/deconz-rest-plugin #6374: unhandled APS-DATA.confirm id: 96 status 0xE1

mspbonn, 2022-10-01:
Hardware: Raspbee II
Version: 2.13.02 / 10.11.2021
Firmware: 26690700
Channel: 20
Nearby WiFi: Channel 1 and 13

Raspberry Pi 2 is connected by WiFi to the local network.

Debug log:
```
.
.
21:46:35:185 unhandled APS-DATA.confirm id: 73 status 0xE1
21:46:40:480 unhandled APS-DATA.confirm id: 97 status 0xE1
21:47:17:158 unhandled APS-DATA.confirm id: 80 status 0xE1
21:47:20:000 unhandled APS-DATA.confirm id: 96 status 0xE1
21:47:34:002 unhandled APS-DATA.confirm id: 179 status 0xE1
21:48:05:907 unhandled APS-DATA.confirm id: 107 status 0xE1
21:48:08:237 unhandled APS-DATA.confirm id: 119 status 0xE1
21:48:19:223 unhandled APS-DATA.confirm id: 182 status 0xE1
21:48:34:951 unhandled APS-DATA.confirm id: 10 status 0xE1
21:48:38:228 unhandled APS-DATA.confirm id: 28 status 0xE1
21:49:00:188 unhandled APS-DATA.confirm id: 149 status 0xE1
21:49:18:755 unhandled APS-DATA.confirm id: 6 status 0xE1
21:49:25:353 unhandled APS-DATA.confirm id: 42 status 0xE1
21:50:22:795 unhandled APS-DATA.confirm id: 123 status 0xE1
21:50:23:791 unhandled APS-DATA.confirm id: 128 status 0xE1
21:50:24:742 unhandled APS-DATA.confirm id: 133 status 0xE1
21:50:26:769 unhandled APS-DATA.confirm id: 143 status 0xE1
21:50:31:903 unhandled APS-DATA.confirm id: 167 status 0xE1
21:50:31:987 unhandled APS-DATA.confirm id: 166 status 0xE1
21:50:34:134 unhandled APS-DATA.confirm id: 179 status 0xE1
21:50:35:067 unhandled APS-DATA.confirm id: 186 status 0xE1
.
.
```

What' the reason for that ?
Some devices are unreactive from time by time.

Greetings

closed 2022-10-01 by close-issue-app[bot]

Mimiix, 2022-10-01:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log :)

Mimiix, 2022-10-01:
For further questions please use the forums