dresden-elektronik/deconz-rest-plugin #6378: Update 2.18.2 MLI tint Smart Switches loosing connection

ottelo9, 2022-10-02:
## Describe the bug
After deconz update from 2.17.1 to 2.18.2 my tint Smart Switches (Müller Licht Aldi) loosing connection randomly. I've 5 of them. One day  1-2 switches have problems, next day they work for days without any problems. They work again if I reset the power.
tint smart switch V2.5.3_r47. I now try to go back to 2.17.1. I use deconz inside Home Assistant with Conbee 2.

## Steps to reproduce the behavior
...

## Expected behavior
...

## Screenshots
[image]
[image]

## Environment
- deCONZ 2.18.2
- Conbee II 26720700
- Home Assistant 2022.9.6
- Deconz Addon 6.16.0
- Do you use an USB extension cable: yes

## deCONZ Logs

## Additional context

Mimiix, 2022-10-02:
Hi there,

As this is not a bug report as described in #5113 I am closing this issue. Feel free to open an topic on the forum 's.

Kind regards,
Dennis

closed 2022-10-02 by Mimiix

ottelo9, 2022-10-11:
Problem was the new OTA updates:
https://community.home-assistant.io/t/deconz-official-thread/100504/3305