dresden-elektronik/deconz-rest-plugin #6379: Tuya TS0202 motion sensor - _TZ3000_mcxw5ehu

Dods, 2022-10-02:
## Device
(https://www.aliexpress.com/item/1005003187399208.html)

- Product name: ZigBee3.0 human motion sensor IH012-RT01
- Manufacturer: _TZ3000_mcxw5ehu
- Model identifier: TS0202
- Device type:
  - Sensor

## Screenshots

[image]

[image]

[image]

[image]

[image]

[image]

[image]

I believe it's the same sensor as in the issue #5584 

That had a pull request #6098  

And got released in Yama Yama [Yama Yama v2.17.0-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.17.0-beta)

It would be great if we can have support for this one, i tried adding it manually but didn't succeed.

Smanar, 2022-10-03:
Hello, and if you edit this file https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/ts0202_presence_sensor.json

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": ["_TZ3000_msl6wxk9", "_TZ3000_otvn3lne", "_TZ3000_mcxw5ehu"],
   "modelid":["TS0202", "TS0202", "TS0202"],
   "product":"TS0202 Presence sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "default":0
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":60,
               "max":3600,
               "change":"0x00000001"
            }
         ]
      }
   ]
}

```

Dods, 2022-10-03:
Something's not working, with the json edit i can pair the sensor in phoscon app (previously couldn't pair but it would show up in deconz), i can rename the sensor, i can see if motion is detected or not, but phoscon shows this:

`Vendor - _TZ3000_mcxw5ehu
Product - Motion Sensor
Status - Not reachable
Version - unknown`

Also, home assistant pick it up, but it doesn't work.
First it reported unavailable, then motion detected, then it stays that way (even though it updates in phoscon):

`Logbook
October 3, 2022
Tuya Motion detected
23:14:35 - 10 minutes ago
Tuya Motion became unavailable
23:14:34 - 10 minutes ago
Tuya Motion cleared (no detected)
23:11:59 - 12 minutes ago`

I'll test it with some lights using phoscon sensor control and report back.

Smanar, 2022-10-04:
I think  you have a problem with config/reachable, you can take a look in phoscon/help/API Information/sensor

Dods, 2022-10-04:
This is a screenshot of what you suggested:
[image]

Also, seems like the only thing that "changes" is the icon for motion detection when motion is detected.
Nothing happens apart from that though,  phoscon doesn't control the lights or anything, it's like the value is never updated, and in Home Assistant status is unchanged from 23 hours ago when i added the sensor via edited json file from the first post.

Dods, 2022-10-04:
Just for example, this is a working motion sensor, screenshot of sensor info:
[image]

Smanar, 2022-10-04:
You have "reachable" = true, so not logic you have as error "not recheable" ....
And you are not comparing the same presence sensor type, one is using the alarm cluster, the other the presence cluster.

Alamr sensor only trigger when detection, you have a new notification but the state don't change, like for switch.

Try to use the legacy code, in the DDF
```

            {
               "name":"state/presence"
            }
```

Only.

Dods, 2022-10-04:
> You have "reachable" = true, so not logic you have as error "not recheable" .... And you are not comparing the same presence sensor type, one is using the alarm cluster, the other the presence cluster.

I'm sorry, i tried to understand what you meant to say but i can't. Where do i check what cluster does the sensor use? How do i change that?

> Alamr sensor only trigger when detection, you have a new notification but the state don't change, like for switch.

I think i understand, and that behavior would change if the tuya would use "presence" cluster instead of "alarm" cluster, but how do i change that?

 
> Try to use the legacy code, in the DDF
> 
> ```
> 
>             {
>                "name":"state/presence"
>             }
> ```
> 
> Only, remove all lines for "state/presence" (and the comma)
> 

Do you mean from this (from the file you told me to edit) :

`{
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }`

To this:

`{
               "name":"state/presence"
               }`

> And why the state is updated but not the "lastupdated" ...

What do you mean by that?

Smanar, 2022-10-05:
>I'm sorry, i tried to understand what you meant to say but i can't. Where do i check what cluster does the sensor use? How do i change that?

You can't change that ^^, it s the device firmware, it's visible on cluster list, your second capture on the first post, this one have the cluster 0x0500, I think the other have the cluster 0x0406

>What do you mean by that?

On your device json you have 
```
State: {
    "presence" :true
    "lastupated": "none"
}
```

and "lastupdated need to be updated in same time than "presence"

>Do you mean from this (from the file you told me to edit) :

Yep exaclty, like that you will use the legacy code insteaof the DDF one, can have a difference.

Dods, 2022-10-09:
After some trying and failing, I'm still not able to use the sensor. Changing the DDF did make it appear on phoscon and home assistant, but due to the use of "alarm" cluster instead of "presence" it seems like this won't work. The question is then, how are the other sensors that have a bit of a different name but still use the alarm cluster work?

Smanar, 2022-10-09:
```
   "manufacturername": ["_TZ3000_msl6wxk9", "_TZ3000_otvn3lne", "_TZ3000_mcxw5ehu"],
   "modelid":["TS0202", "TS0202", "TS0202"],
```

The 2 others are working with the same DDF.
I don't see what is missing for yours.

Monofin, 2022-10-12:
Wonder if your issues are related to:
#6368 ?
It can be worked around by forcing the device fingerprint in the DDF - root cause not yet fixed though.

Smanar, 2022-10-12:
>Wonder if your issues are related to:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6368 ?
It can be worked around by forcing the device fingerprint in the DDF - root cause not yet fixed though.

Ha fuck, right, try using this DDF

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": ["_TZ3000_msl6wxk9", "_TZ3000_otvn3lne", "_TZ3000_mcxw5ehu"],
   "modelid":["TS0202", "TS0202", "TS0202"],
   "product":"TS0202 Presence sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint": {
           "profile": "0x0104",
           "device": "0x0402",
           "endpoint": "0x01",
           "in": [
             "0x0000",
             "0x0001",
             "0x0500"
           ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "default":0
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":60,
               "max":3600,
               "change":"0x00000001"
            }
         ]
      }
   ]
}
```

Monofin, 2022-10-12:
.. and if that works, then #6398 may well fix an events/enrolement problem (after the DDF has been updated to include _TZ3000_mcxw5ehu). :-)

matze20211, 2022-10-16:
for me its working fine with the above ddf including #6367

Smanar, 2022-10-16:
Ok, soo DDf added here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

MrSolo570, 2022-10-18:
Hello, I have the very same motion sensor and it doesn't work anymore. I have two (maybe noobie) questions, can anyobody elaborate please, and dont mind my asking:

1. I have been using this sensor until today without any hassle. Due to some re-configuring I had to repair it again and now it does not work at all anymore. How did this happen? Like I am interested in the background about this here discussed issue.

2. Can anybode guide my step-by-step on how to make it work again? I see you all posting DDF codes but I dont know where to but it to make it work again?

btw my phoscon looks like this:
[image]

MrSolo570, 2022-10-18:
How come it doesn't use DDF and where can I change it?

Smanar, 2022-10-18:
DDF are something news. Your device is an old one.

DDF can overwrite legacy code, so yes if your device is relay broked a DDF can help.
Can make a try with this DDF.

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": "_TZ1800_fcdjzz3s",
   "modelid": "TY0202",
   "product":"lidl Presence sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint": {
           "profile": "0x0104",
           "device": "0x0402",
           "endpoint": "0x01",
           "in": [
             "0x0000",
             "0x0001",
             "0x0500"
           ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "default":0
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":60,
               "max":3600,
               "change":"0x00000001"
            }
         ]
      }
   ]
}
```

DDF for this device

```
   "manufacturername": "_TZ1800_fcdjzz3s",
   "modelid": "TY0202",
```

Dods, 2022-10-19:
> Ok, soo DDf added here #6244

I finally tried the DDF above, i was away for work, sorry.
Now it reports motion (and clears motion) in Phoscon but the sensor status is still "Not reachable":
[image]
[image]
[image]

Also, and i confirmed this now because it was a bit weird, the sensor doesn't report the state correctly, it only refreshes on reloading the Deconz integration in Home Assistant:

[image]

And in Deconz, when viewing the device map, the sensor isn't connected to anywhere (highlighted in blue):

[image]

Has anyone had this issue with anything, where the sensor "works" but "isn't" connected?

I'll try to add sensor control to some lights via Phoscon to see if that part is working, will report back.

Dods, 2022-10-19:
> I'll try to add sensor control to some lights via Phoscon to see if that part is working, will report back.

Just had it on sensor control in Phoscon for about 15 minutes, the sensor control doesn't work, maybe it's the same problem as with the no reports on Home Assistant. Somehow Phoscon can see the status changing from Detected to Cleared, but nothing happens after that.

evanhemmen, 2022-11-05:
Hi, a newbie here. To be honest I'm still struggling to understand DDF and Github/branches/pull requests. So I've read this issue completely but still a bit lost what to do.

I've bought a similar sensor as in the first post but with a different manufacturer number:

Model ID: TS0202
Manufacturer name: _TZ3040_6ygjfyll

[https://nl.aliexpress.com/item/1005003187399208.html?spm=a2g0o.order_list.0.0.d02679d2nu69ub&gatewayAdapt=glo2nld](url)

[image]

I've tried to pair it via Phoscon webinterface, but no result. Although the sensor indicated that is was connected. And indeed in the deconz-gui it is visible. Same story as above. But the id's and numbers are different. What are my next steps to get it working? Latest stable versions of conbee2 firmware and deconz are installed.

Smanar, 2022-11-05:
Hello, but the DDF in the PR you submitted haven't worked ?

Dods, 2022-11-05:
No idea, i moved on, using ZHA now, everything works.

evanhemmen, 2022-11-06:
I wasn't shure if I modified the DDF file properly, so I removed it. I tried it again, but still am a novice at github, pull-requests, etc. I actually don't understand the workflow and how I'm able to get a modified file on my Raspberry Pi.

Smanar, 2022-11-07:
Depend of the OS, if you don't have HA.

`All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.`

So you just need to create a text file called what_you_want.json with this contain

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": ["_TZ3000_msl6wxk9", "_TZ3000_otvn3lne", "_TZ3040_6ygjfyll"],
   "modelid":["TS0202", "TS0202", "TS0202"],
   "product":"TS0202 Presence sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "default":0
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":60,
               "max":3600,
               "change":"0x00000001"
            }
         ]
      }
   ]
}
```

Then restart deconz and re-include the device, or select the node "edit DDF" load the file, save and make a "hot reload"

evanhemmen, 2022-11-07:
Thanks, that helped me a step further.

In my case (Domoticz - Raspbian - RPi3b+) the location is: /home/pi/.local/share/dresden-elektronik/deCONZ/devices

Created a text file there with the script above followed by the procedure "edit DDF", load file and save. Hot reload did not work. But rebooting the pi and then re-pairing the sensor made it visible. Next thing is testing it in Domoticz.

Smanar, 2022-11-08:
Have added the device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

closed 2022-12-08 by github-actions[bot]