dresden-elektronik/deconz-rest-plugin #6381: _TZE200_cpmgn2cf

flowerbob, 2022-10-02:
## Device

- Product name: Zigbee Radiator Actuator
- Manufacturer: _TZE200_cpmgn2cf
- Model identifier: TS0601
- Device type : 
  - Thermostat

([aliexpress.com/item/1005003033044081.html](url))

## Screenshots
[image]
[image]
[image]
[image]

closed 2022-11-01 by github-actions[bot]

Piorko, 2022-11-22:
I've exactly the same issue with _TZE200_cpmgn2cf. HA deconz plugin: 6.17.0, deconz: 2.19.1.

p-git1, 2022-11-24:
The reason is an incomplete implementation in the API code. See [this thread in the phoscon forum](https://forum.phoscon.de/t/tuya-trv-tze200-cpmgn2cf-not-pairing-correctly/2698/4)

After adding the mentioned mapping line in the code and a recompile it works for me now.