dresden-elektronik/deconz-rest-plugin #6382: Hue Motion Sensor | new HW release / revision

andi-bar157, 2022-10-02:
There seems to be a new hardware revision of the Hue Motion Sensor (SML004) which seems to be creating a lot of APS delays - at least in my case. I am running a network of about 92 with about 50 routers, so my mesh is pretty stable and good.

I am running two old revisions (SML002) and two of the new one. Swapping the installation spots / sensor inbetween did not fix the issue. So i feel like this is related to the hardware revision and not

Clusters of the SML004 - hew hardware revision:

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Clusters of the the SML002 - old hardware revision:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

My environment:

[image]

-->

 - Raspberry Pi 4B  - 2GB 
 - Home Assistent deCONZ Add-on
 - Firmware version: 26780700
 - deCONZ version: 2.18.02 / 19.9.2022
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: no

## deCONZ Logs

Will be provided as needed!

andi-bar157, 2022-10-02:
dear @close-issue-app

fixxed as requested :)

ebaauw, 2022-10-02:
The 2022 revision of the Hue outdoor motion sensor has been around for quite some time.  I think support was added in May.  I refactored that to use DDF in July in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6164.  I have exactly the same model (same _Date Code_ and same _SW Build ID_) as you, and it's working just fine.

Mimiix, 2022-10-02:
@andi-bar157 I think it would be best to ask this on discord or the forus. As @ebaauw mentions it works for him and he added support himself.

closed 2022-10-02 by Mimiix