dresden-elektronik/deconz-rest-plugin #6390: Device name

Delusional112, 2022-10-03:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

SwoopX, 2022-10-03:
Can you please:
- update the issue name
- Correct the model identifier typo at the top
- Add the requested screenshot of the available clusters

Thanks!

Smanar, 2022-10-03:
Hello still missing the cluster list (the node picture), to see for exemple the Endpoint used by the device
But I think like all plug the on/off is working but not the power measurement ?

So generaly you can take a lambda DDF and just edit the model ID and the manufacture name, for exemple this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_ksw8qtmt_plug_nous_a1z.json

Delusional112, 2022-10-03:
> Can you please:
> 
> * update the issue name
> * Correct the model identifier typo at the top
> * Add the requested screenshot of the available clusters
> 
> Thanks!

Ok, did what you asked.

Delusional112, 2022-10-03:
> Hello still missing the cluster list (the node picture), to see for exemple the Endpoint used by the device But I think like all plug the on/off is working but not the power measurement ?
>
> So [...]

I'm sorry, just added it to the post. I tried something myself with de DDF editor, but it didn't work out. This is the first time a have a device that is not supported.

Smanar, 2022-10-03:
Still missing the cluster list ^^, it s the more important capture, it s the same than the first one here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

I can't help on incomplete issue, but to make fast, you just need to create a json file with the name you want in the "devices" folder with the same contain than the previous link but with replacing

```
  "manufacturername": "_TZ3000_ksw8qtmt",
  "modelid": "TS011F",
```

By the one from your devices.

The problem is creating the json file, it depend on OS.

SwoopX, 2022-10-03:
@Delusional112 Please test this one thoroughly

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "LDS",
  "modelid": "ZBT-ONOFFPlug-D0009",
  "vendor": "YNOA",
  "product": "Smart Plug zigbee 3.0",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0B04"
        ],
        "out": [
          "0x000A"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 360
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 360
        },
        {
          "name": "state/voltage",
          "refresh.interval": 360
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0702"
        ],
        "out": [
          "0x000A"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "eval": "Item.val = Attr.val * 10;"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Delusional112, 2022-10-09:
Thanks @SwoopX . I just made a JSON file. Where should I upload it? I have a samba share to the config folder of HA.

Smanar, 2022-10-10:
It' a little complicated on HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

Delusional112, 2022-10-11:
> It' a little complicated on HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

Ok, I read it. This is not something I have the skills for. I'm a man of GUI's and drag & drop :) 

So, I'll have to wait till it's implemented in the next Deconz version.

Smanar, 2022-10-12:
The problem is we need at least one user test it before adding the device :(

You don't have a way to create a file somewhere and load it using the DDF editor (with the GUI) ?

Delusional112, 2022-10-14:
@Smanar In the DDF editor you can open and save files. But the paths are in the container as far as I can see. But the problem is that I can't reach the json file that is outside of the container

Smanar, 2022-10-14:
Was one of my interrogation about the HA docker system.
I know the folder "data" is persistent, but nothing inside, and idk how to access it from inside the docker.

closed 2022-11-13 by github-actions[bot]