dresden-elektronik/deconz-rest-plugin #6393: Blitzwolf Power Meter update interval Conbee vs. Raspbee

dklinger, 2022-10-04:
## Describe the bug
After migrating from Raspbee to Conbee II/Docker, my Blitzwolf BW-SPH 13 Plug is only updating power meter info every 5 minutes. With Raspbee it updated 3 times per minute before.
Is this a limitation of the Conbee II or can this be configured? If the latter, how?

## Steps to reproduce the behavior
Look at update interval of Blitzwolf Smart Plug.

## Expected behavior
Update interval is at least once per minute. Better: Update interval is configurable.

## Screenshots
Not applicable

## Environment

 - Host system: Docker on Intel NUC
 - Running method: official docker container
 - Firmware version: 26780700
 - deCONZ version: 2.17.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2022-10-04:
Hi

This is not an bug, but simply has to do with how the device works and your network. The firmware of the Blitzwolf simply doesn't support reporting, so that's why polling is used. Depending on the network, polling is done more often or not. By switching, that might have been the case. Did you add more devices than before after moving?

Kind regards,
Dennis

dklinger, 2022-10-04:
Thanks for coming back. I wasn't sure where to ask/report his as there are a lot of forums.
It's the same network as before. No new devices. But the location of the deconz device changed.
Raspbee under the roof and the Conbee Stick is now in the basement.

dklinger, 2022-10-04:
Found it. I can change the DDF. Refresh.interval was set to 300 for this device.
So, click on device, right click, Edit DDF and change the value of power/refresh interval

closed 2022-10-04 by dklinger