dresden-elektronik/deconz-rest-plugin #6396: Heiman HS3AQ carbondioxide detector

batje, 2022-10-06:
## Device

- Product name: Carbon Dioxide Detector Heiman HS3AQ
- Manufacturer: HEIMAN
- Model identifier: HS3AQ-EFA-3.0
- Device type :  
  - Sensor
https://www.heimantech.com/product/?type=detail&id=106

## Screenshots
### Basic

[image]

### Identify
[image]

### Humidity
[image]

### Device Temperature

[image]

### Time
[image]

### Alarms

[image]
### Power Configuration

[image]

### Other clusters that are not mentioned above

The 2 Unknown groups are empty. It would be great if the actual CO2 value would be accessible, eventually.

batje, 2022-10-28:
Please issue, don't close

Mimiix, 2022-10-28:
@Smanar can this be done with a DDF?

Smanar, 2022-10-28:
I think can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "HEIMAN",
  "modelid": "HS3AQ-EFA-3.0",
  "product": "HS3AQ-EFA-3.0",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$ZHACarbonMonoxide",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x040D"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/carbonmonoxide",
          "awake": true,
          "parse": {
            "at": "0x0000",
            "cl": "0x040D",
            "ep": 1,
            "eval": "Item.val = Attr.val * 1000;",
            "fn": "zcl"
          },
          "read": {
            "at": "0x0000",
            "cl": "0x040D",
            "ep": 1,
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x040D",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

to test the unknow cluster, the DDf will use a ZHACarbonMonoxide

batje, 2022-10-28:
Can I test this before you commit this? How would I go about doing that?

Smanar, 2022-10-29:
You need to install the DDF, but it will depend of your OS (Generally is you have HA or not)

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

You just need to create a text file in the "devices" folder with this contain, and rename it what_you_whant.json.
Then restart deconz and re-include your device

batje, 2022-10-29:
So far, so good. The device now appears in the API:

```
```  "14": {
    "config": {
      "battery": 100,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "etag": "fcd2f708a4cafbdb83fc67202475085a",
    "lastannounced": null,
    "lastseen": "2022-10-29T12:08Z",
    "manufacturername": "HEIMAN",
    "modelid": "HS3AQ-EFA-3.0",
    "name": "Temperature 14",
    "state": {
      "lastupdated": "2022-10-29T12:08:24.390",
      "temperature": 2114
    },
    "type": "ZHATemperature",
    "uniqueid": "5c:02:72:ff:fe:e7:ad:79-01-0402"
  },
  "15": {
    "config": {
      "battery": 100,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "etag": "d5a202fee093b277dc029a681fc70bcc",
    "lastannounced": null,
    "lastseen": "2022-10-29T12:08Z",
    "manufacturername": "HEIMAN",
    "modelid": "HS3AQ-EFA-3.0",
    "name": "Humidity 15",
    "state": {
      "humidity": 6397,
      "lastupdated": "2022-10-29T12:08:24.781"
    },
    "type": "ZHAHumidity",
    "uniqueid": "5c:02:72:ff:fe:e7:ad:79-01-0405"
  },
  "16": {
    "config": {
      "battery": 100,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "etag": "219467da9cb3d365c10554373077eb93",
    "lastannounced": null,
    "lastseen": "2022-10-29T12:08Z",
    "manufacturername": "HEIMAN",
    "modelid": "HS3AQ-EFA-3.0",
    "name": "$ZHACarbonMonoxide 16",
    "state": {
      "carbonmonoxide": true,
      "lastupdated": "2022-10-29T12:08:27.224"
    },
    "type": "$ZHACarbonMonoxide",
    "uniqueid": "5c:02:72:ff:fe:e7:ad:79-01-040d"
  }
}

batje, 2022-10-29:
I found this: https://www.zigbee2mqtt.io/devices/HS3AQ.html which claims the co2 property is called, yes, co2

batje, 2022-10-29:
I changed the uuid of the humidity sensor from 0405 to 0400, the id of the 'unknown' group, and the 'state/humidity' to 'state/co2'.

Then removed the device, restarted deconz and re-added the device again. Now I do see the co2 value on the display * 100 in the measured value group of the 'relative humidity measurement' group.

So, it seems what's needed is a "$ZHACarbonDioxide", that is a copy of the humidity Sensor.

Smanar, 2022-10-29:
>I changed the uuid of the humidity sensor from 0405 to 0400

Why, it's the cluster 0x0405.

>the id of the 'unknown' group, and the 'state/humidity' to 'state/co2'. 

I don't understand, why you have edited the working humidity sensor and not the ZHACarbonMonoxide  ?
But there is a problem yes, there is no state/co2 field, and the one I have used (state/carbonMonoxide is a bool, so not usable for value.

So can you try this DDf instead 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "HEIMAN",
  "modelid": "HS3AQ-EFA-3.0",
  "product": "HS3AQ-EFA-3.0",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
[... 205 lines cut ...]
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x040D",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

I m using state/airqualityppb to test the device, if value are good I will add a state/co2 in the PR.

Smanar, 2022-11-03:
@batje this DDf work better ?

batje, 2022-11-07:
sorry for the late response.
I tried the above config and a few others that look like it, and the value for airqualityppb never changes from 0

Smanar, 2022-11-07:
You have made a "hot reload" or reinclude the device to configure bind ?

If you add this code to the general.xml file you will be able to use the GUI on the new cluster

```
    <cluster id="0x040D" name="Carbon dioxyde measurement">
      <description></description>
      <server>
        <attribute-set id="0x0000" description="Relative Humidity Measurement Information">
          <attribute id="0x0000" name="Measured Value" type="u16" access="r" default="0" required="m"></attribute>
          <attribute id="0x0001" name="Min Measured Value" type="u16" access="r" required="m" range="0x954d,0x7ffe"></attribute>
          <attribute id="0x0002" name="Max Measured Value" type="u16" access="r" required="m" range="0x954e,0x7ffe"></attribute>
          <attribute id="0x0003" name="Tolerance" type="u16" access="r" required="o" range="0x0000,0x0800"></attribute>
        </attribute-set>
      </server>
      <client>
      </client>
    </cluster>
```

Perhaps you will be able to read the value.

batje, 2022-11-07:
Progress!

[image]

In the API the value now is 1.

As to how I reloaded the template: I deleted the device, I rebooted the rasberry and then added the device.
Now, when playing with the attribute only, I only rebooted between edits of the json file.

Smanar, 2022-11-08:
>In the API the value now is 1.

If the value is updated only when ready the value in the GUI it mean there is a binding/reporting issue but the DDF is fine

And I have probably found the problem, the type, if you look on your capture, when you have read the value, it was chnaged to "float", mean in the xml file you need to use

    <cluster id="0x040D" name="Carbon dioxyde measurement">
      <description></description>
      <server>
        <attribute-set id="0x0000" description="Relative Carbon Dioxyde Information">
          <attribute id="0x0000" name="Measured Value" type="float" access="r" default="0" required="m"></attribute>
          <attribute id="0x0001" name="Min Measured Value" type="float" access="r" required="m" range="0x954d,0x7ffe"></attribute>
          <attribute id="0x0002" name="Max Measured Value" type="float" access="r" required="m" range="0x954e,0x7ffe"></attribute>
          <attribute id="0x0003" name="Tolerance" type="float" access="r" required="o" range="0x0000,0x0800"></attribute>
        </attribute-set>
      </server>
      <client>
      </client>
    </cluster>

And same for bind in the DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "HEIMAN",
  "modelid": "HS3AQ-EFA-3.0",
  "product": "HS3AQ-EFA-3.0",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
[... 214 lines cut ...]
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x040D",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x39",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

closed 2022-12-08 by github-actions[bot]

batje, 2023-01-23:
Just to let you know. I didnt get it working with the solution in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6396#issuecomment-1307469665
Since then an update of deconz wiped the changes in the xml & json files, so I recreated them from scratch *and* I updated the firmware on my ConBee II stick. With the firmware 26780700 the value for the ppb is read out correctly in the API.

```
"27":{"config":{"battery":100,"offset":0,"on":true,"reachable":true},"etag":"f61974288752d4a2a7622f442d0c4350","lastannounced":null,"lastseen":"2023-01-23T20:05Z","manufacturername":"HEIMAN","modelid":"HS3AQ-EFA-3.0","name":"AirQuality 27","state":{"airqualityppb":823,"lastupdated":"2023-01-23T20:05:49.071"},"type":"ZHAAirQuality","uniqueid":"5c:02:72:ff:fe:e7:ad:79-01-040d"},
```

Thanks so much and I hope this device can be added to the next release.

batje, 2023-02-15:
@Smanar You might have missed that this device is indeed working now.

Smanar, 2023-02-15:
Yep, totally ^^
PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6748