dresden-elektronik/deconz-rest-plugin #6401: Device name

mcf68k, 2022-10-08:
## Device

- Product name: E1 Wireless Remote Switch (Double Rocker), Model WXKG17LM manufactured by Aqara
- Manufacturer: LUMI
- Model identifier: lumi.remote.acn004
- Device type : Switch

https://zigbee.blakadder.com/Aqara_WXKG17LM.html

Screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2022-10-08:
Since you apparently have already a switch resource created, what does the deconz GUI debug output give you upon button press?

_Help -> Debug view_, then enable INFO and INFO_L2. The interesting messages should beginn with `[INFO] - ` for each button press (single, double, tripple?, hold?).

mcf68k, 2022-10-09:
Deconz doesn't recognise it as at all, the switch came from me messing around with ddf files. As you see with no success.

mcf68k, 2022-10-15:
@SwoopX I tried some ddf files from others components (H1 et E1 with a single button) but it doesn't work, any ideas ?
Thanks

SwoopX, 2022-10-15:
I could imagine this one would do:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.remote.acn004",
  "matchexpr": "R.endpoints.length === 3",
  "vendor": "Xiaomi",
  "product": "Aqara E1 dual rocker wireless switch WXKG17LM",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0012"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0000",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0012",
          "0xFCC0"
        ],
        "out": [
          "0x0006"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          }
        },
        {
          "name": "config/clickmode",
          "awake": true,
          "parse": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'highspeed' } else if (Attr.val == 2) { Item.val = 'multiclick' } else { Item.val = 'unknown' }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'highspeed') { 1 } else if (Item.val == 'multiclick') { 2 }",
            "fn": "zcl",
            "mf": "0x115f",
            "state.timeout": 2
          },
          "values": [
            ["\"highspeed\"", "Buttons emit only single click events, but fast"],
            ["\"multiclick\"", "Emits single, double and treble presses"] 
          ],
          "default": "multiclick"
        },
        {
          "name": "config/devicemode",
          "refresh.interval": 3000,
          "parse": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'compatibility' } else if (Attr.val == 2) { Item.val = 'zigbee' } else { Item.val = 'unknown' }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'compatibility') { 1 } else if (Item.val == 'zigbee') { 2 }",
            "fn": "zcl",
            "mf": "0x115f",
            "state.timeout": 2
          },
          "values": [
            ["\"compatibility\"", "Default mode for Xiaomi devices"],
            ["\"zigbee\"", "Closer to zigbee standard"] 
          ],
          "default": "compatibility"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

mcf68k, 2022-10-15:
@SwoopX thank you.

I loaded the ddf file and here what I'm getting as info log when I pushing button 1 and then button 2:

23:16:48:425 ZCL attribute report 0x54EF4410004F67B2 for cluster: 0x0012, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
23:16:49:051 ZCL attribute report 0x54EF4410004F67B2 for cluster: 0x0012, ep: 0x01, frame control: 0x18, mfcode: 0x0000 

but it's still not recognized as switch, any idea ?

SwoopX, 2022-10-16:
How do you determine the device is not recognized? And just sharing 2 preselected log lines is not exactly helpful 🙂

mcf68k, 2022-10-17:
@SwoopX because at phoscon App it's shows as empty switch and the events of pressing buttons are not transmitted to Domoticz. 
[image]

What king of log will be helpful ?

Smanar, 2022-11-08:
Hello can you try the previous DDF and give somes logs with "info" "ddf" and "info_l2" ?

sushiat, 2022-11-08:
[deconz log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9965404/deconz.log.txt)

Hope I did this right :) Restarted deconz container (switch wasn't bound), then enabled info, info_l2 and ddf in the deconz log viewer, started adding switch in phoscon and reset it, watched it come in hex first, then eventually all the way. In phoscon the switch has no buttons though and I can see that the battery reports as 0%. Log is attached, the device ID of the switch is 0x540F57FFFE1B4FEA.

Smanar, 2022-11-09:
>Log is attached, the device ID of the switch is 0x540F57FFFE1B4FEA.

```
{
	"attr": {
		"colorcapabilities": 0,
		"ctmax": 65279,
		"ctmin": 0,
		"id": "3",
		"lastannounced": "2022-11-08T19:15:13Z",
		"lastseen": "2022-11-08T21:24Z",
		"manufacturername": "IKEA of Sweden",
		"modelid": "TRADFRIbulbGU10WS345lm",
		"name": "Dining Table Top",
		"swversion": "1.0.012",
		"type": "Color temperature light",
		"uniqueid": "54:0f:57:ff:fe:1b:4f:ea-01"
	},
	"e": "changed",
	"id": "3",
	"r": "lights",
	"t": "event",
	"uniqueid": "54:0f:57:ff:fe:1b:4f:ea-01"
}
```

sure ? ^^, 54:ef:44:10:00:50:14:a9 will be better.

From your log it seem the fonction checkSensorButtonEvent() is never called, not sure this device use the cluster 0x0012.

BTW there is no bind for this cluster in the DDF, it's something normal ?

sushiat, 2022-11-10:
Yeah sorry, copied the wrong ID (was getting late and those IKEA lights are really creating a lot of entries).

Not sure about the rest, would love to understand more about Zigbee and how I would get startet looking at what this thing transmits and how to correct the DDF but tbh I don't have a clue :)

I just don't understand why they had to change that in the first place, sure the button design is slightly different from the older version, but why change the firmware too? Just to mess with us guys that don't want to also buy and use their hub...

Smanar, 2022-11-10:
The good question is why thoses fu**inkg brand like Xiaomi or Tuya are not able to respect standard ...

But ATM I m sorry but I realy don't see why the fonction checkSensorButtonEvent() is not trigger on the device.
The only event I m seeing for your device are on cluster 0xfcc0 and from that I m seeing your device need  to use the cluster 0x0012

mcf68k, 2022-11-11:
I tried some ddf files from others components but it didn't works for me, still inserting to make it work.

sushiat, 2022-11-17:
I wonder if this has something to do with the "double switch" mode they advertise. You can click any button 5 times and it is supposed to switch between the modes - but I think a compatible hub needs to send a reply to this. I picked up a Aqara E1 hub of Amazon for testing and it doesn't seem to be supported. The switch is working with it, but you can only get single clicks from it. No double clicks or long press events.

I also picked up one of their newer H1 double rocker switches and that thing behaves pretty much like my older D1 switches (will create a new issue to get them fully supported). But I can see the deconz events from it in HA and all the click modes work, and the event IDs looks identical to my D1's. The H1 also supposedly has the dual switching mode, but my guess is it defaults to the non "lighning fast single click mode" the E1 seems to default to and therefore works correctly with deconz.

mehmetakin1996, 2022-11-22:
I cant connect my E1 Wireless Remote Switch (Double Rocker).

- Model: WXKG17LM 

Fix this please...?

[image]

Smanar, 2022-11-23:
Hello can you try this DDF 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.remote.acn004",
  "vendor": "Xiaomi",
  "product": "Aqara E1 dual rocker wireless switch WXKG17LM",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0012"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0000",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0012",
          "0xFCC0"
        ],
        "out": [
          "0x0006"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          }
        },
        {
          "name": "config/clickmode",
          "awake": true,
          "parse": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'highspeed' } else if (Attr.val == 2) { Item.val = 'multiclick' } else { Item.val = 'unknown' }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'highspeed') { 1 } else if (Item.val == 'multiclick') { 2 }",
            "fn": "zcl",
            "mf": "0x115f",
            "state.timeout": 2
          },
          "values": [
            ["\"highspeed\"", "Buttons emit only single click events, but fast"],
            ["\"multiclick\"", "Emits single, double and treble presses"] 
          ],
          "default": "multiclick"
        },
        {
          "name": "config/devicemode",
          "refresh.interval": 3000,
          "parse": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'compatibility' } else if (Attr.val == 2) { Item.val = 'zigbee' } else { Item.val = 'unknown' }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'compatibility') { 1 } else if (Item.val == 'zigbee') { 2 }",
            "fn": "zcl",
            "mf": "0x115f",
            "state.timeout": 2
          },
          "values": [
            ["\"compatibility\"", "Default mode for Xiaomi devices"],
            ["\"zigbee\"", "Closer to zigbee standard"] 
          ],
          "default": "compatibility"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

And check if you have some logs when using it with log flag "info_l2"

`DBG_Printf(DBG_INFO_L2, "[INFO] - No button map for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",`

mehmetakin1996, 2022-11-23:
> Hello can you try this DDF
>
>
>
> ```
>
> {
>
>   "schema": "devcap1.schema.json",
>
>   "manufacturername": "$MF_LUMI",
>
>   "modelid": "lumi.remote.acn004",
>
>   "vendor": "Xiaomi",
>
>   "product": "Aqara E1 dual [...]

How can i change it?
Can you help me?

Smanar, 2022-11-23:
>How can i change it?

You mean how to install the DDF ?
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

On normal OS 
All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

On HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

You just need to create a text file called what_you_want.json in a "devices" folder. Then restart deconz and include the device.

mehmetakin1996, 2022-11-24:
> > How can i change it?
> 
> You mean how to install the DDF ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
> 
> On normal OS All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home//.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.
> 
> On HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
> 
> You just need to create a text file called what_you_want.json in a "devices" folder. Then restart deconz and include the device.

How can I paste the json file there.
This is quite difficult, can you help me with that?

Smanar, 2022-11-24:
I never use docker.
An user have made a more detailled tuto > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5864#issuecomment-1326314062

mehmetakin1996, 2022-11-24:
Aah thanks!
I will try it and i will report it.

mehmetakin1996, 2022-11-24:
> I never use docker. @ottelo9 have made a more detailled tuto > [#5864 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5864#issuecomment-1326314062)

i cant find the map deConz. Can you help me? 
[image]

ottelo9, 2022-11-24:
> > I never use docker. @ottelo9 have made a more detailled tuto > [#5864 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5864#issuecomment-1326314062)
> 
> i cant find the map deConz. Can you help me? [image]

You cannot find it there, because you are not inside the core-deconz docker! You see only the home assistant folders. Go to my tutorial and follow instructions B). If you are inside the deconz docker you can only use the terminal not the sftp browser (your image).

Smanar, 2022-11-25:
But you haven't the buttonmap file yet.
So just install the DDF, press button and give me the result (the error in log)
I will make the buttonmap.

mehmetakin1996, 2022-11-25:
> Maar je hebt het buttonmap-bestand nog niet. Dus installeer gewoon de DDF, druk op de knop en geef me het resultaat (de fout in het logboek). Ik zal de buttonmap maken, indien nodig kan ik waarschijnlijk zonder.

What exactly do you need?

mehmetakin1996, 2022-11-25:
> But you haven't the buttonmap file yet. So just install the DDF, press button and give me the result (the error in log) I will make the buttonmap, if needed, I can probably make without it.

What exactly do you need?

Smanar, 2022-11-25:
Just install the DDF, the device will be reconized but not working.

Go in deconz / help / debug view, enable flag "info_l2", and press button.
I need the error messeage you will have and the button/action done.

`DBG_Printf(DBG_INFO_L2, "[INFO] - No button map for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",`

mehmetakin1996, 2022-11-25:
> Just install the DDF, the device will be reconized but not working.
> 
> Go in deconz / help / debug view, enable flag "info_l2", and press buttons. I need the error messeage you will have and the button/action done.
> 
> `DBG_Printf(DBG_INFO_L2, "[INFO] - No button map for: %s%s, endpoint: 0x%02X, cluster: %s, command: %s, payload: %s, zclSeq: %u\n",`

Can you explain step by step what commands I need to execute? I am now in my SSH root.

Smanar, 2022-11-25:
You are on HA ?
I don't use their OS, sorry, but using SSH is not enought, you need to "connect" to their docker.
On the previous link, you have a tuto, and an app called "MobaXterm" perhaps this method is easier.

mehmetakin1996, 2022-11-25:
> You are on HA ? I don't use their OS, sorry, but using SSH is not enought, you need to "connect" to their docker. On the previous link, you have a tuto, and an app called "MobaXterm" perhaps this method is easier.

Yes on HA..
There are steps below but how and with what commands do I get into that path?

A) upload / download a DDF file

in deconz vnc select the smoke sensor, right click, Edit DDF
a draft ddf will be created, simply save it (/data/.local/share/dresden-elektronik/deCONZ/devices/)
download the DDF json file you want to use for the device or copy the code
install and start "SSH & Web Terminal" Addon and deactivate the permission mode. Set a password!
Then install and start "MobaXterm" on your pc. Create new SSH session for HA. Login with root und password.
Now you will see a ssh terminal and on the left side a sftp browser with files. Now enter this into the terminal to copy the created draft ddf from the deconz container to the accessible share folder.
e.g. (change file name)
docker cp addon_core_deconz:/data/.local/share/dresden-elektronik/deCONZ/devices/lumi.sensor_smoke.acn03.json /share
Now you can download (drag and drop) the draft ddf from the share folder and edit it.
-- upload new DDF file --
To upload the edited ddf oder a new downloaded one, type this into the terminal (first put the file into the share folder):
e.g. (change file name)
docker cp /share/aqara_smart_smoke_sensor.json addon_core_deconz:/data/.local/share/dresden-elektronik/deCONZ/devices/

Smanar, 2022-11-25:
So seem fine for me.
Direclty on the docker the path is `/data/.local/share/dresden-elektronik/deCONZ/devices/`

Just restart deconz to test, and try to include the device.

If you have the GUI (using VNC) you can select the node and take "edit DDF", you will see wich one file deconz is using in the window title bar.

mehmetakin1996, 2022-11-25:
> So seem fine for me. Direclty on the docker the path is `/data/.local/share/dresden-elektronik/deCONZ/devices/`
>
> Just restart deconz to test, and try to include the device.
>
> If you have the GUI [...]

I don't get it anymore.
Can you possibly reproduce this for me? 

Pretty complicated.

closed 2022-12-25 by github-actions[bot]