dresden-elektronik/deconz-rest-plugin #6407: Can't add a WSDCGQ11LM Aqara Humidity and Temperature sensor to deCONZ

notthebee, 2022-10-09:
## Describe the bug

One of my WSDCGQ11LM humidity and motion sensors suddenly started reporting a -100°C temperature, so I decided to reset it. However, it wouldn't react to the reset button at all, so I figured that the battery was dead.

After replacing the battery, I tried pairing the device to my ConBee II. The blue light appeared after holding the power button, and blinked rapidly after I let go of the button, indicating the pairing process. The deCONZ logs also indicated that the device was detected after it blinked blue.

However, nothing happened after that. The "Ready" button stayed greyed out in the deCONZ GUI, and after some time I got an error message saying "Failed to connect"

I've tried repeating the process multiple times with known working batteries as well, to no avail. The logs always show the device parameters, but then nothing happens.

All of my other WSDCGQ11LMs seem to be working fine, but I don't want to experiment with resetting and pairing them again, since I don't want to lose the humidity/temperature measurements in other rooms.

## Steps to reproduce the behavior

Try to pair a WSDCGQ11LM to deCONZ

## Expected behavior

The WSDCGQ11LM gets paired and the GUI indicates that a new device has been added

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Docker container
 - Firmware version: 26720700
 - deCONZ version: 2.18.02 / 9/19/2022
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system: no

## deCONZ Logs

```
19:36:51:153 ZCL attribute report 0x00158D000711084F for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000
19:36:51:154 0x00158D000711084F extract Xiaomi special attribute 0xFF01
19:36:51:154    01 battery 2945 (0x0B81)
19:36:51:154    04 unknown 424 (0x01A8)
19:36:51:155    05 RSSI dB (?) 29355 (0x72AB)
19:36:51:155    06 LQI (?) 4294967296 (0x0100000000)
19:36:51:155    64 temperature 2659
19:36:51:155    65 humidity 6362
19:36:51:156    66 pressure 98197 (982)
19:36:51:156    0a Parent NWK 0 (0x0000)
19:36:51:232 [1] get node descriptor for 0x00158d000711084f
19:36:51:233 ZDP get node descriptor for 0x7792
19:36:51:933 FP indication 0x0000 / 0x8002 (0x00158D000711084F / 0x7792)
19:36:51:934                       ...     (0x00158D000711084F / 0x7792)
19:36:51:934 ZDP indication search sensors 0x00158D000711084F (0x7792) cluster 0x8002
19:36:51:934 ZDP indication search sensors 0x00158D000711084F (0x7792) clear timeout on cluster 0x8002
19:36:51:940 [2] get active endpoints for 0x00158d000711084f
19:36:51:940 ZDP get active endpoints for 0x7792
19:36:54:878 ZCL attribute report 0x847127FFFEF42CA7 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
19:36:55:106 FP indication 0x0000 / 0x8005 (0x00158D000711084F / 0x7792)
19:36:55:107                       ...     (0x00158D000711084F / 0x7792)
19:36:55:107 ZDP indication search sensors 0x00158D000711084F (0x7792) cluster 0x8005
19:36:55:108 ZDP indication search sensors 0x00158D000711084F (0x7792) clear timeout on cluster 0x8005
19:36:55:185 ZCL attribute report 0x84FD27FFFED5A3AB for cluster: 0x0201, ep: 0x01, frame control: 0x08, mfcode: 0x0000
19:36:55:318 FP indication 0x0104 / 0x0402 (0x00158D000711084F / 0x7792)
19:36:55:318                       ...     (0x00158D000711084F / 0x7792)
```

## Additional context

None

notthebee, 2022-10-09:
Turns out that the issue was caused by applying this fix: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6276#issuecomment-1261850227
A clean reinstall seemed to have solved the issue. Closing now

closed 2022-10-09 by notthebee

arjannv, 2022-10-19:
I have the same -100°C readings on these sensors. Did you find a way to fix that as well?

Crayjin, 2022-10-21:
I have this issue as well. It is currently one of 6 sensors which provides -100°C every now and then (Mostly with a break between 1 - 3 days).

notthebee, 2022-10-21:
> I have the same -100°C readings on these sensors. Did you find a way to fix that as well?

Basically just remove the sensor from deCONZ and pair it again.