dresden-elektronik/deconz-rest-plugin #6409: Sonoff SNZB-02 Timeoutbug

Mephisto090, 2022-10-10:
## Describe the bug

Nach dem ich ein Sonoff SNZB-02 (https://www.amazon.de/gp/product/B08BFW697F/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1) erfolgreich einen Tag betrieben habe. Wurde nach circa 1. Tag keine Daten mehr gesendet und der Punkt der Node war nur noch Rot. Nachdem das Device gelöscht wurde lässt es sich nicht mehr einbinden und gibt auch nichts mehr zurück (Siehe Screens) Bei allen Anfragen bekomme ich ein Timeout

## Steps to reproduce the behavior

Siehe Text

## Expected behavior

Neue Verbindung mit dem Gerät bzw. kein Abbruch der Daten die gesendet werden

## Screenshots

[image]
[image]
[image]
[image]

## Environment

 - Host system: Windows
 - Running method: Windows
 - Firmware version: 26580700
 - deCONZ version: 2.18.02 / 19.9.2022
 - Device: ConBee II
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? If so: No

## deCONZ Logs

```
19:18:31:201 APS-DATA.confirm id: 196, status: 0x00 SUCCESS
19:18:31:201 APS-DATA.confirm request id: 196 -> confirmed, timeout 165168765
19:18:31:201 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 17, rssi: 21
19:18:31:201 APS-DATA.indication request id: 196 -> finished
19:18:31:201 APS-DATA.request id: 196 erase from queue
19:18:31:201 ZDP 0x00212EFFFF07F843 cluster: 0x8031 status = 0x00 -> SUCCESS
19:18:31:201 ZDP Mgmt_Lqi_rsp zdpSeq: 208 from 0x00212EFFFF07F843 total: 4, startIndex: 2, listCount: 2
19:18:31:201     * neighbor: 0x00124B00253946E9 (0x5177), LQI: 188, relation: 0x01, depth: 1, rxOnWHenIdle: 0
19:18:31:217     * neighbor: 0x00124B002543F65A (0x94F7), LQI: 252, relation: 0x01, depth: 1, rxOnWHenIdle: 0
19:18:36:807 Mgmt_Lqi_req zdpSeq: 209 to 0x00212EFFFF07F843 start index 0
19:18:36:807 APS-DATA.request id: 198, addrmode: 0x03, addr: 0x00212effff07f843, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
19:18:36:814 APS-DATA.confirm id: 198, status: 0x00 SUCCESS
19:18:36:814 APS-DATA.confirm request id: 198 -> confirmed, timeout 165174390
19:18:36:814 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 17, rssi: 21
19:18:36:814 APS-DATA.indication request id: 198 -> finished
19:18:36:814 APS-DATA.request id: 198 erase from queue
19:18:36:814 ZDP 0x00212EFFFF07F843 cluster: 0x8031 status = 0x00 -> SUCCESS
19:18:36:827 Mgmt_Lqi_req zdpSeq: 210 to 0x00212EFFFF07F843 start index 2
19:18:36:827 APS-DATA.request id: 200, addrmode: 0x03, addr: 0x00212effff07f843, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
19:18:36:830 ZDP Mgmt_Lqi_rsp zdpSeq: 209 from 0x00212EFFFF07F843 total: 4, startIndex: 0, listCount: 2
19:18:36:830     * neighbor: 0x00124B00251FF971 (0x4809), LQI: 252, relation: 0x01, depth: 1, rxOnWHenIdle: 0
19:18:36:830     * neighbor: 0x00124B002531B785 (0x9090), LQI: 160, relation: 0x01, depth: 1, rxOnWHenIdle: 0
19:18:36:830 APS-DATA.confirm id: 200, status: 0x00 SUCCESS
19:18:36:847 APS-DATA.confirm request id: 200 -> confirmed, timeout 165174412
19:18:36:852 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 17, rssi: 21
19:18:36:852 APS-DATA.indication request id: 200 -> finished
19:18:36:852 APS-DATA.request id: 200 erase from queue
19:18:36:857 ZDP 0x00212EFFFF07F843 cluster: 0x8031 status = 0x00 -> SUCCESS
19:18:36:857 ZDP Mgmt_Lqi_rsp zdpSeq: 210 from 0x00212EFFFF07F843 total: 4, startIndex: 2, listCount: 2
19:18:36:857     * neighbor: 0x00124B00253946E9 (0x5177), LQI: 188, relation: 0x01, depth: 1, rxOnWHenIdle: 0
19:18:36:857     * neighbor: 0x00124B002543F65A (0x94F7), LQI: 252, relation: 0x01, depth: 1, rxOnWHenIdle: 0
19:18:42:452 Mgmt_Lqi_req zdpSeq: 211 to 0x00212EFFFF07F843 start index 0
19:18:42:452 APS-DATA.request id: 202, addrmode: 0x03, addr: 0x00212effff07f843, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
19:18:42:467 APS-DATA.confirm id: 202, status: 0x00 SUCCESS
19:18:42:467 APS-DATA.confirm request id: 202 -> confirmed, timeout 165180038
19:18:42:483 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 17, rssi: 21
19:18:42:483 APS-DATA.indication request id: 202 -> finished
19:18:42:483 APS-DATA.request id: 202 erase from queue
19:18:42:483 ZDP 0x00212EFFFF07F843 cluster: 0x8031 status = 0x00 -> SUCCESS
19:18:42:483 Mgmt_Lqi_req zdpSeq: 212 to 0x00212EFFFF07F843 start index 2
19:18:42:483 APS-DATA.request id: 204, addrmode: 0x03, addr: 0x00212effff07f843, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
19:18:42:483 ZDP Mgmt_Lqi_rsp zdpSeq: 211 from 0x00212EFFFF07F843 total: 4, startIndex: 0, listCount: 2
19:18:42:483     * neighbor: 0x00124B00251FF971 (0x4809), LQI: 252, relation: 0x01, depth: 1, rxOnWHenIdle: 0
19:18:42:498     * neighbor: 0x00124B002531B785 (0x9090), LQI: 160, relation: 0x01, depth: 1, rxOnWHenIdle: 0
19:18:42:523 APS-DATA.confirm id: 204, status: 0x00 SUCCESS
19:18:42:524 APS-DATA.confirm request id: 204 -> confirmed, timeout 165180077
19:18:42:524 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 17, rssi: 21
19:18:42:524 APS-DATA.indication request id: 204 -> finished
19:18:42:540 APS-DATA.request id: 204 erase from queue
19:18:42:540 ZDP 0x00212EFFFF07F843 cluster: 0x8031 status = 0x00 -> SUCCESS
19:18:42:540 ZDP Mgmt_Lqi_rsp zdpSeq: 212 from 0x00212EFFFF07F843 total: 4, startIndex: 2, listCount: 2
19:18:42:545     * neighbor: 0x00124B00253946E9 (0x5177), LQI: 188, relation: 0x01, depth: 1, rxOnWHenIdle: 0
19:18:42:545     * neighbor: 0x00124B002543F65A (0x94F7), LQI: 252, relation: 0x01, depth: 1, rxOnWHenIdle: 0
```

## Additional context

Mimiix, 2022-10-10:
Hi,

We keep a English github, so please use English to keep everyone involved :)

Mephisto090, 2022-11-03:
> As there has not been any response in 21 days, this issue has been automatically marked as stale. **At OP: Please either close this issue or keep it active** It will be closed in 7 days if no further activity occurs.

I'm waiting for an answer from the manufacturer

closed 2022-12-04 by github-actions[bot]