dresden-elektronik/deconz-rest-plugin #6412: QBKGO3LM aqara added as switch instead of light

WladyX, 2022-10-11:
## Describe the bug
In the past this worked as light ok, but now it's seen as a switch for a reason, i've tried removing it from deconz/phoscon and readding it, but it gets added as a switch (which i cannot control from HA as far as i know) instead of a light.

## Steps to reproduce the behavior
adding the aqara switch results in a switch beeing added and not a light

## Expected behavior
in the past it was added as light

## Screenshots
[image]
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: (26720700)
 - deCONZ version: (2.18.02 / 9/19/2022)
 - Device: ConBee II
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system?  no

## deCONZ Logs
```
2022-10-11 19:41:43 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"247","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2022-10-11T16:41:43.158"},"t":"event","uniqueid":"00:15:8d:00:02:46:21:a4-04-0006"}
```
## Additional context
no additional context

Smanar, 2022-10-14:
It's a wired device, so it's a light.
If i m right you send order to this device ? to turn on or off the light ?

Smanar, 2022-10-14:
You have enabled the "dissociated" mode ?
If i m right you need to have 2 entries for this device 1 in "light" and 1 in "sensor"

WladyX, 2022-10-20:
What is dissociated mode and how can i enable/disable?
For the other working switches, i only have lights in deconz ( i see sensors only in HA ).
For this one in deconzi only see it as switch.
About the wire...yeah, i know, me too, made by my father (which btw is a electric engineer, so he should know better, but you know the saying that the shoeman does not have good shoes...)
DDF looks complicated as first sight :) And this exact switch used to work as light before without issues...

WladyX, 2022-10-21:
When you say GUI you mean Phoscon or deconz? I presume deconz, can you share more info on how to check this dissociate function please?
if I search for the mac, i see only the switch
[image]
Thank you very much so for your support, I have several lights that became switches and it drives me crazy, hope i can fix this and replicate for the rest.

Smanar, 2022-10-21:
Yep, it's using deconz, but If you don't know it, better to don't use it ^^
You have it in the basic cluster 0x0000

```
			<attribute id="0xff22" name="Xiaomi Disconnect 1" type="u8" showas="hex" access="rw" required="o" mfcode="0x115f">
				<description>Set to 0x12 (0xFE) to connect (disconnect) the left button to (from) the relay.</description>
			</attribute>
			<attribute id="0xff23" name="Xiaomi Disconnect 2" type="u8" showas="hex" access="rw" required="o" mfcode="0x115f">
				<description>Set to 0x22 (0xFE) to connect (disconnect) the left button to (from) the relay.</description>
			</attribute>
```

BTW, I m looking your capture, it's normal you have nothing at " manufacture name", second capture, attribute 0x0004 ?

I will link this issue on dev channel, but not sure we will have a solution.
I think DDF will be faster, so big question, what are your skill on the HA OS ^^ ?

- Can you share me the basic DDF auto created by deconz (right clic / "edit DDF" / copy paster the preview
- I can make a DDF from scratch, but you will need to instal it on HA
- I can explain you how to make one using the DDF editor

SwoopX, 2022-10-21:
That DDF should bring up the "lights". However, using DDF code, in this case here, would cease the button events, as the device doesn't use any button map but some other hardcoded parts which are then skipped. The map below should do the trick, but I haven't received confirmation on it yet

### DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.ctrl_neutral2",
  "vendor": "Xiaomi",
  "product": "Aqara double rocker switch QBKG03LM",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x03"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x04",
        "0x0006"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0000",
        "endpoint": "0x04",
        "in": [
          "0x0006"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/temperature"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

### Button map (button_maps.json)

```
"xiaomiSwitchCtrlNeutralMap": {
            "vendor": "Xiaomi",
            "doc": "QBKG03LM (dual rocker) and QBKG04LM (single rocker) switch",
            "modelids": ["lumi.ctrl_neutral1", "lumi.ctrl_neutral2"],
            "buttons": [
                {"S_BUTTON_1": "Left"},
                {"S_BUTTON_2": "Right"},
                {"S_BUTTON_3": "Both"}
            ],
            "map": [
                [1, "0x04", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press"],
                [1, "0x05", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press"],
                [1, "0x06", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press"]
            ]
        },
```

WladyX, 2022-10-26:
Total deconz n00b here :)
I saved ddf.json and button_maps.json, I went to deconz, edited the DDF for my device, opened the ddf.json, saved, reloaded DDF, but if I edit again it does not seem to be saved... how do I push the ddf?
Also what should I do with the buttons_map file? 
Will this be fixed in a future release? Also why did this happen, because it used to work, something changed in the code and broke it?
Thank you for your support!!

Smanar, 2022-10-26:
>I saved ddf.json

In wich one folder ? You have 2 usables, one used by deconz with all other DDF, one other for "user"
After that you just need to restart deconz and re-include the device.

>I went to deconz, edited the DDF for my device, opened the ddf.json

If all is fine, when you make "edit DDF" deconz will choose you file.

>Also what should I do with the buttons_map file?

In the deconz folder for DDF (the folder "devices)there is inside a file called button_maps.json, just open it with a text editor and you will understand, but you don't need it to use the device as light.

WladyX, 2022-10-28:
I'm on the hass docker addon, so i copied to the shared `/data` folder, which is mounted from the host.
I see in the container the `/usr/share/deCONZ/devices` folder, that's the one you are probably refering to, but if i copy it there it will be lost upon a container restart, as that is not persistent storage, only `/data` is.
What are my options? :)

Smanar, 2022-10-28:
Ha right, It probably possible to put the DDF in a persisitent folder, but can't change the buttonamp path, so not possible for this one.
But like I have said, you just need it if you use button return , as you can use light state, I don't think you need it.

You have a persistent folder for DDF here in "data"
`/data/.local/share/dresden-elektronik/deCONZ/devices/`

https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/20

WladyX, 2022-10-31:
WOW, that worked, thank you so much for your help!
In my case (I installed hassio via an unmaintained docker script) the folder was:
`/usr/share/hassio/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices`
Maybe somebody will need this.

Smanar, 2022-10-31:
@SwoopX your DDF is tested and working.

closed 2022-11-08 by manup