dresden-elektronik/deconz-rest-plugin #6415: Unexpected light color mode 2

dutch-erik, 2022-10-11:
After discussing with Robban (thank you!) he adviced ("this is definitely an issue on deCONZ side") to open this bug report for the warnings `Unexpected light color mode 2` below.
I get these (many) warnings for the two IKEA bulbs - TRADFRIbulbE27WSglobeopal1055lm - TRADFRI bulb E27 CWS opal 600lm

## Describe the bug
```
Logger: pydeconz.models.light.light
Source: components/deconz/light.py:167
First occurred: 21:09:49 (83 occurrences)
Last logged: 21:12:37

Unexpected light color mode 2
```
```
'11': {'colorcapabilities': 16,
       'ctmax': 454,
       'ctmin': 250,
       'etag': '8d9d99bd330e6080d94830cdbead5501',
       'hascolor': True,
       'lastannounced': None,
       'lastseen': '2022-10-08T12:46Z',
       'manufacturername': 'IKEA of Sweden',
       'modelid': 'TRADFRIbulbE27WSglobeopal1055lm',
       'name': 'Buiten 1',
       'state': {'alert': 'none',
                 'bri': 217,
                 'colormode': '2',
                 'ct': 454,
                 'on': False,
                 'reachable': True},
       'swversion': '1.0.012',
       'type': 'Color temperature light',
       'uniqueid': '04:cd:15:ff:fe:c6:da:19-01'},
```
```
'2': {'colorcapabilities': 8,
      'etag': '7b7b6f15f585f2b40148d2a128123aa7',
      'hascolor': True,
      'lastannounced': '2022-10-05T19:34:51Z',
      'lastseen': '2022-10-08T12:46Z',
      'manufacturername': 'IKEA of Sweden',
      'modelid': 'TRADFRI bulb E27 CWS opal 600lm',
      'name': 'Schuur 1',
      'state': {'alert': 'none',
                'bri': 250,
                'colormode': 'xy',
                'effect': 'none',
                'hue': 0,
                'on': False,
                'reachable': True,
                'sat': 0,
                'xy': [0.453, 0.432]},
      'swversion': '2.3.086',
      'type': 'Color light',
      'uniqueid': '68:0a:e2:ff:fe:16:01:55-01'},

```

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi)
 - Running method: (Home Assistent deCONZ Add-on)
 - Firmware version: (26780700)
 - deCONZ version: (2.18.02)
 - Device: (ConBee II)
[... 98 lines cut ...]
2022-10-02 12:32:48.275 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"10","r":"lights","state":{"alert":"none","on":true,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:06:75:8f-01"}
2022-10-02 12:32:50.300 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-02 12:32:50.341 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-02 12:32:50.381 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"colorcapabilities":16,"ctmax":454,"ctmin":250,"id":"11","lastannounced":null,"lastseen":"2022-10-02T10:32Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRIbulbE27WSglobeopal1055lm","name":"Buiten 1","swversion":"1.0.012","type":"Color temperature light","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-02 12:32:50.429 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-02 12:32:50.469 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-02 12:32:51.098 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"13","lastannounced":"2022-09-23T06:05:08Z","lastseen":"2022-10-02T10:32Z","manufacturername":"LUMI","modelid":"lumi.vibration.aq1","name":"Vibration Sensor","swversion":"20180130","type":"ZHAVibration","uniqueid":"00:15:8d:00:03:23:31:74-01-0101"},"e":"changed","id":"13","r":"sensors","t":"event","uniqueid":"00:15:8d:00:03:23:31:74-01-0101"}
2022-10-02 12:32:51.104 DEBUG (MainThread) [pydeconz.websocket] {"config":{"battery":100,"on":true,"pending":[],"reachable":true,"sensitivity":1,"sensitivitymax":21,"temperature":2300},"e":"changed","id":"13","r":"sensors","t":"event","uniqueid":"00:15:8d:00:03:23:31:74-01-0101"}
2022-10-02 12:32:54.596 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"id":"3","lastannounced":"2022-09-19T16:13:08Z","lastseen":"2022-10-02T10:32Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 CWS 806lm","name":"SKCasper 1","swversion":"1.0.021","type":"Extended color light","uniqueid":"ec:1b:bd:ff:fe:ce:71:c8-01"},"e":"changed","id":"3","r":"lights","t":"event","uniqueid":"ec:1b:bd:ff:fe:ce:71:c8-01"}
2022-10-02 12:32:56.275 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"9","lastannounced":null,"lastseen":"2022-10-02T10:32Z","manufacturername":"_TZ3000_g5xawfcq","modelid":"TS0121","name":"Wasmachine 1","swversion":"1.0.1","type":"Smart plug","uniqueid":"04:cd:15:ff:fe:06:6c:83-01"},"e":"changed","id":"9","r":"lights","t":"event","uniqueid":"04:cd:15:ff:fe:06:6c:83-01"}
2022-10-02 12:32:56.281 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"9","r":"lights","state":{"alert":"none","on":true,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:06:6c:83-01"}
2022-10-02 12:32:56.282 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"30","lastannounced":null,"lastseen":"2022-10-02T10:32Z","manufacturername":"_TZ3000_g5xawfcq","modelid":"TS0121","name":"Power 30","swversion":"1.0.1","type":"ZHAPower","uniqueid":"04:cd:15:ff:fe:06:6c:83-01-0b04"},"e":"changed","id":"30","r":"sensors","t":"event","uniqueid":"04:cd:15:ff:fe:06:6c:83-01-0b04"}
2022-10-02 12:32:56.282 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"29","lastannounced":null,"lastseen":"2022-10-02T10:32Z","manufacturername":"_TZ3000_g5xawfcq","modelid":"TS0121","name":"Consumption 29","swversion":"1.0.1","type":"ZHAConsumption","uniqueid":"04:cd:15:ff:fe:06:6c:83-01-0702"},"e":"changed","id":"29","r":"sensors","t":"event","uniqueid":"04:cd:15:ff:fe:06:6c:83-01-0702"}
2022-10-02 12:33:09.283 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"10","r":"lights","state":{"alert":"none","on":true,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:06:75:8f-01"}
2022-10-02 12:33:11.304 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"colorcapabilities":16,"ctmax":454,"ctmin":250,"id":"11","lastannounced":null,"lastseen":"2022-10-02T10:33Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRIbulbE27WSglobeopal1055lm","name":"Buiten 1","swversion":"1.0.012","type":"Color temperature light","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-02 12:33:11.310 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-02 12:33:11.344 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-02 12:33:11.386 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"colorcapabilities":16,"ctmax":454,"ctmin":250,"id":"11","lastannounced":null,"lastseen":"2022-10-02T10:33Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRIbulbE27WSglobeopal1055lm","name":"Buiten 1","swversion":"1.0.012","type":"Color temperature light","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-02 12:33:11.442 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-02 12:33:11.468 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
[... cut ...]
2022-10-02 12:33:19.389 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"colorcapabilities":8,"id":"2","lastannounced":"2022-09-24T20:09:29Z","lastseen":"2022-10-02T10:33Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 CWS opal 600lm","name":"Schuur 1","swversion":"2.3.086","type":"Color light","uniqueid":"68:0a:e2:ff:fe:16:01:55-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"68:0a:e2:ff:fe:16:01:55-01"}
```
## Additional context
If you need more information, please let me know.

Smanar, 2022-10-14:
Oups, sorry, Robban have explained the issue on dev channel, but I have completey forget it.

In the worst situation if I don't found the issue, it's possible the set and lock the value using DDF, and if you have acces to the GUI (so the the DDF editor, it's fast and without playing with dockers)

Just to be sure, this one is the not working 

```
'11': {'colorcapabilities': 16,
       'ctmax': 454,
       'ctmin': 250,
       'etag': '8d9d99bd330e6080d94830cdbead5501',
       'hascolor': True,
       'lastannounced': None,
       'lastseen': '2022-10-08T12:46Z',
       'manufacturername': 'IKEA of Sweden',
       'modelid': 'TRADFRIbulbE27WSglobeopal1055lm',
       'name': 'Buiten 1',
       'state': {'alert': 'none',
                 'bri': 217,
                 'colormode': '2',
                 'ct': 454,
                 'on': False,
                 'reachable': True},
       'swversion': '1.0.012',
       'type': 'Color temperature light',
       'uniqueid': '04:cd:15:ff:fe:c6:da:19-01'},
```

And the other is working ?

@iisworks have corrected so much of 'Color temperature light' too, recently, it seem there is more issue with this kind of device recently

dutch-erik, 2022-10-14:
Hi @Smanar, no problem at all. I'm glad that someone is looking at it. ;)

Just for my understanding, to which value needs the `colormode:` be changed in DDF (if you can't find the issue)?

Regarding id=2; same warning `Unexpected light color mode 2`, which seems to be weird I realize now with `'colormode': 'xy'`. So I just checked and noticed the warning is there as soon as I turn it on.

```
'2': {'colorcapabilities': 8,
      'etag': '7b7b6f15f585f2b40148d2a128123aa7',
      'hascolor': True,
      'lastannounced': '2022-10-05T19:34:51Z',
      'lastseen': '2022-10-08T12:46Z',
      'manufacturername': 'IKEA of Sweden',
      'modelid': 'TRADFRI bulb E27 CWS opal 600lm',
      'name': 'Schuur 1',
      'state': {'alert': 'none',
                'bri': 250,
                'colormode': 'xy',
                'effect': 'none',
                'hue': 0,
                'on': False,
                'reachable': True,
                'sat': 0,
                'xy': [0.453, 0.432]},
      'swversion': '2.3.086',
      'type': 'Color light',
      'uniqueid': '68:0a:e2:ff:fe:16:01:55-01'},
```
If you need more information, please let me know.

Smanar, 2022-10-14:
For the device id = 11
The type is
`'type': 'Color temperature light',`

so you need to have 
```
“colormode”: “ct”,
“colorcapabilities”: 16,
```

If you make "edit DDF", you will be able to set them and make them fixed, then save the DDF and make "hot realod"

But the json look fine for the 'Color light', what happen when you try to turn it "on" ?

dutch-erik, 2022-10-16:
Thank you for the information. I'm going to look at it.

Below the logs when I turn on device_id=2.
After that the warnings are popping up (Unexpected light color mode 2).
I'm not sure if this has anything to do with this device_id or is just related to device_id=11 `'type': 'Color temperature `light',

```
'2': {'colorcapabilities': 8,
      'etag': '2cda2b4b42d30e0cf216fbe7a2f72d69',
      'hascolor': True,
      'lastannounced': '2022-10-11T16:38:32Z',
      'lastseen': '2022-10-15T10:38Z',
      'manufacturername': 'IKEA of Sweden',
      'modelid': 'TRADFRI bulb E27 CWS opal 600lm',
      'name': 'Schuur 1',
      'state': {'alert': 'none',
                'bri': 250,
                'colormode': 'xy',
                'effect': 'none',
                'hue': 0,
                'on': False,
                'reachable': True,
                'sat': 0,
                'xy': [0.453, 0.432]},
      'swversion': '2.3.086',
      'type': 'Color light',
      'uniqueid': '68:0a:e2:ff:fe:16:01:55-01'},
```

```
2022-10-15 12:45:49.000 DEBUG (MainThread) [homeassistant.components.deconz] deCONZ event created: styrbar
2022-10-15 12:45:55.366 INFO (MainThread) [pydeconz.websocket] Connected to deCONZ (core-deconz)
2022-10-15 12:46:01.870 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"colorcapabilities":16,"ctmax":454,"ctmin":250,"id":"6","lastannounced":"2022-09-03T19:41:37Z","lastseen":"2022-10-15T10:46Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WS opal 1000lm","name":"Zolder 1","swversion":"2.3.091","type":"Color temperature light","uniqueid":"58:8e:81:ff:fe:ae:26:9f-01"},"e":"changed","id":"6","r":"lights","t":"event","uniqueid":"58:8e:81:ff:fe:ae:26:9f-01"}
2022-10-15 12:46:02.639 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"10","lastannounced":null,"lastseen":"2022-10-15T10:46Z","manufacturername":"_TZ3000_g5xawfcq","modelid":"TS0121","name":"Smart plug 2","swversion":"1.0.1","type":"Smart plug","uniqueid":"04:cd:15:ff:fe:06:75:8f-01"},"e":"changed","id":"10","r":"lights","t":"event","uniqueid":"04:cd:15:ff:fe:06:75:8f-01"}
2022-10-15 12:46:02.644 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"10","r":"lights","state":{"alert":"none","on":true,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:06:75:8f-01"}
2022-10-15 12:46:02.645 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"32","lastannounced":null,"lastseen":"2022-10-15T10:46Z","manufacturername":"_TZ3000_g5xawfcq","modelid":"TS0121","name":"Power 32","swversion":"1.0.1","type":"ZHAPower","uniqueid":"04:cd:15:ff:fe:06:75:8f-01-0b04"},"e":"changed","id":"32","r":"sensors","t":"event","uniqueid":"04:cd:15:ff:fe:06:75:8f-01-0b04"}
2022-10-15 12:46:02.649 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"31","lastannounced":null,"lastseen":"2022-10-15T10:46Z","manufacturername":"_TZ3000_g5xawfcq","modelid":"TS0121","name":"Consumption 31","swversion":"1.0.1","type":"ZHAConsumption","uniqueid":"04:cd:15:ff:fe:06:75:8f-01-0702"},"e":"changed","id":"31","r":"sensors","t":"event","uniqueid":"04:cd:15:ff:fe:06:75:8f-01-0702"}
2022-10-15 12:46:02.687 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"32","r":"sensors","state":{"current":0,"lastupdated":"2022-10-15T10:46:02.684","power":0,"voltage":232},"t":"event","uniqueid":"04:cd:15:ff:fe:06:75:8f-01-0b04"}
2022-10-15 12:46:02.709 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"32","r":"sensors","state":{"current":0,"lastupdated":"2022-10-15T10:46:02.707","power":0,"voltage":232},"t":"event","uniqueid":"04:cd:15:ff:fe:06:75:8f-01-0b04"}
2022-10-15 12:46:02.749 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"32","r":"sensors","state":{"current":0,"lastupdated":"2022-10-15T10:46:02.730","power":0,"voltage":233},"t":"event","uniqueid":"04:cd:15:ff:fe:06:75:8f-01-0b04"}
2022-10-15 12:46:04.689 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-15 12:46:04.716 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-15 12:46:04.759 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"colorcapabilities":16,"ctmax":454,"ctmin":250,"id":"11","lastannounced":null,"lastseen":"2022-10-15T10:45Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRIbulbE27WSglobeopal1055lm","name":"Buiten 1","swversion":"1.0.012","type":"Color temperature light","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-15 12:46:04.799 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-15 12:46:04.853 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-15 12:46:17.282 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"9","r":"lights","state":{"alert":"none","on":true,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:06:6c:83-01"}
[... cut ...] da:19-01"}
2022-10-15 12:49:34.750 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
2022-10-15 12:49:34.791 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":"none","bri":217,"colormode":"2","ct":454,"on":false,"reachable":true},"t":"event","uniqueid":"04:cd:15:ff:fe:c6:da:19-01"}
```

ebaauw, 2022-10-16:
The problem is in the [item definition](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/generic/items/state_colormode_item.json), which was updated in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6320.

The `parse` function should translate the numeric value into the corresponding string, as the `ResourceItem` has been defined as string:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/fb9b322a2aa09317bb5123c7d7776e335646640d/resource.cpp#L329
<br>
Alternatively, the `ResourceItem` could be redefined as numeric, and the C++ code could translate the numeric value to string, very much like translating the values of `RStateX` and `RStateY` into `xy`.  This leads to the philosophical question whether resource items should reflect the Zigbee attribute value or the API attribute value.  We're not really consistent there.

I'm not too thrilled adding default `read` and `parse` actions to the item definitions, as they affect all devices that haven't overwritten these.  This is still a way for a bad change for one device to break multiple devices.  Also, for some lights, `colormode` should be based on _Enhanced Color Mode_ (attribute 0x4001) instead.

Smanar, 2022-10-16:
But for the model "TRADFRIbulbE27WSglobeopal1055lm"
The DDF use only
```
         {
          "name": "state/colormode"
        },
```

So it use the legacy code ?

@dutch-erik your both device already have DDF
```
devices\ikea\tradfribulbe27wsglobeopal1055lm.json
devices\ikea\e27_cws_opal_600lm_light.json
```

So you can lock value for all.
- For devices\ikea\tradfribulbe27wsglobeopal1055lm.json

```
    {
      "name": "state/colormode",
      "static": "ct"
    },
    {
      "name": "config/colorcapabilities",
      "static": 16
    },
```

- For devices\ikea\e27_cws_opal_600lm_light.json

```
    {
      "name": "state/colormode",
      "static": "xy"
    },
    {
      "name": "config/colorcapabilities",
      "static": 8
    },
```

ebaauw, 2022-10-16:
he DDF use only
```
       {
          "name": "state/colormode"
        },
```
So it use the legacy code ?

No, it uses the code from the item definition.

Smanar, 2022-10-16:
>No, it uses the code from the item definition.

Ha seriously ? arrrrrrr, can explain lot of thing.
In my mind the item definition is only used to generate the DDF, not during the working mode.

hanskroner, 2022-10-17:
I ended up here after noticing a similar issue: '_Unexpected light effect type 0_' being repeated for quite a few of my bulbs. The item definition for `RStateEffect` seems to also have been updated to be a string.

ebaauw, 2022-10-17:
Please open a new issue for that, with:
- The web socket notification sent by deCONZ with the numeric value;
- A listing of the `lights` resource that's causing this;
- The version of deCONZ.