dresden-elektronik/deconz-rest-plugin #6420: Tuya Zigbee Water Leak Sensor ZW08

P-etr, 2022-10-14:
## Device

- Product name: Tuya Zigbee Water Leak Sensor
- Manufacturer: Tuya
- Model identifier: ZW08
- Device type :
  - Sensor

Info about sensor: https://www.aliexpress.com/item/1005003485422125.html

## Screenshots

[image]
[image]

Mimiix, 2022-10-14:
Hi,

Can you please add all screenshots as requested?

Smanar, 2022-10-14:
And here you have a working DDF for a sensor like your https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_d16y6col_water_sensor.json

Can try this DDF after having compelted the issue.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_d16y6col", "_TZ3000_85czd6fy"],
  "modelid": ["TS0207", "TS0207"],
  "vendor": "Tuya",
  "product": "Tuya water sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "ZHAWater",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/tampered",
          "awake": true
        },
        {
          "name": "state/water",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

P-etr, 2022-10-14:
I am attaching more screens.
[image]
[image]
[image]
[image]

Smanar, 2022-11-05:
N one have try the DDF ?

Duskeren, 2022-11-29:
> N one have try the DDF ?

I have not tried that device in particular, but my guess is that all these tuya water leak sensors are the same. I have this one: https://www.aliexpress.com/item/1005003923445751.html
And I just added _TZ3000_ocjlo4ea as manufacturername to a copy of the existing _TZ3000_d16y6col_water_sensor.json DDF as a test, and voila, it works. Battery, sensor, everything. So almost no doubt that DDF you posted, would work too.

Smanar, 2022-11-29:
Right, but I prefer wait for a confirmation, I often forget something.
For exemple I think this DDF can provoke a battery drain.

But yes, I have a PR with 20 DDF, 1 or 2 manufacture name in more ....

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

karsie1994, 2022-12-12:
Can you normally pair the sensor in phoscon deconz? And then upload the dfd? I cannot get it to pair in the normal UI.

Smanar, 2022-12-12:
Nope, the API can't reconize it without the DDF (so can't be visible in phoscon), but you can include it in deconz without the DDF.

>I cannot get it to pair in the normal UI

Not normal, a device can always be included in deconz (the GUI) even without support.

flopp999, 2022-12-21:
I do have one of those at home now. I added the above DDF, rebooted deconz(home assistant).
It doesn't choose the DDF automatically.
I open the DDF, Hot reload.
When I press the send button on the device I get this message in Debug

```
22:33:40:924 void DeRestPluginPrivate::handleIasZoneClusterIndication(const deCONZ::ApsDataIndication&, deCONZ::ZclFrame&),196: assertion 'itemIasState' failed
22:33:40:926 void DeRestPluginPrivate::handleIasZoneClusterIndication(const deCONZ::ApsDataIndication&, deCONZ::ZclFrame&),197: assertion 'itemPending' failed
```

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/9c334b0a2b0a881f6fd8e5fa0d3fa2aded0f266f/ias_zone.cpp#L196
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/9c334b0a2b0a881f6fd8e5fa0d3fa2aded0f266f/ias_zone.cpp#L197

flopp999, 2022-12-21:
IAS

```
22:43:25:457 [IAS ZONE] - Address 0xA4C1380D34AC0248, Payload 000000000000, Command 0x00
22:43:25:459 [IAS ZONE] - 0xA4C1380D34AC0248 No IAS sensor found for endpoint: 0x01
```

flopp999, 2022-12-21:
After I deleted the Node in deconz and add it again I get
```
22:50:12:023 [IAS ZONE] - Address 0xA4C1380D34AC0248, Payload 010000312a00, Command 0x01
22:50:12:025 [IAS ZONE] - 0xA4C1380D34AC0248 No IAS sensor found for endpoint: 0x01
22:50:14:879 [IAS ZONE] - Address 0xA4C1380D34AC0248, Payload 000000000000, Command 0x00
22:50:14:880 [IAS ZONE] - 0xA4C1380D34AC0248 No IAS sensor found for endpoint: 0x01
```

flopp999, 2022-12-21:
Now mine is also working. After I changed to Gold it showed up in Phoscon

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZ3000_ocjlo4ea",
    " _TZ3000_d16y6col",
    " _TZ3000_85czd6fy"
  ],
  "modelid": [
    "TS0207",
    " TS0207",
    " TS0207"
  ],
  "vendor": "Tuya",
  "product": "Tuya water sensor",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/_TZ3000_ocjlo4ea_water_sensor.json",
  "subdevices": [
    {
      "type": "ZHAWater",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/tampered",
          "awake": true
        },
        {
          "name": "state/water",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

Smanar, 2022-12-22:
But the status was "gold" at start ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6420#issuecomment-1279034630

On your capture we can see the device is not reconised, he have as name something like 0xXXXX

flopp999, 2022-12-22:
Yes, I played around soooo much that I don't know each step I did 😀 
Yes, I noticed the spaces and did change it in my DDF.
Also spaces in modelid.
I made a PR for this DDF and there I removed the spaces

closed 2023-01-21 by github-actions[bot]