dresden-elektronik/deconz-rest-plugin #6421: Device name

Schnabulation, 2022-10-14:
## Device

- Product name: Bosch Smoke Alarm II
- Manufacturer: clusters not showing (see screenshot)
- Model identifier: clusters not showing (see screenshot)
- Device type : Please remove all unrelated device types. 
  - Sensor
  - Siren

## Screenshots

[image]

[image]

## Problem

This is the new Bosch Smoke Alarm II which is now ZigBee based. However it seems that deCONZ can not retrieve the clusters of the ZigBee device. Is it possible to reveal these somehow?

I did find a Zigbee EUI-64 and an install-code on the device itself. Does this help?

Mimiix, 2022-10-14:
Hi,

It's not paired correctly. Did you repair? Or, this might be one of those that need a unlock code.

Schnabulation, 2022-10-14:
Yes, I did re-pair it: I tried to pair it through the Phoscon app which didn't work. However the device was then shown in deCONZ - but without the cluster infos. I deleted the device, restarted deCONZ, removed the battery of the device and reset it. Then re-paired it - to the same result as before.

There is an "install-code" physically printed on the device and it normally pairs with the Bosch Smart Home Controller so it might be possible that it requires the install-code. Is it possible to supply it when pairing to deCONZ?

Mimiix, 2022-10-14:
Asked devs to check in.

Schnabulation, 2022-10-14:
Appreciated, thank you!

I'd really like to be able to get this thing working. I think it's long overdue that we get a good, reliable Zigbee based smoke detector. Also I know that this Bosch device is able to trigger the siren based on Zigbee commands - because with the Smart Home Controller you can use it as an intruder alert.

manup, 2022-10-14:
We had recently been looking into a Bosch joining problem reported via mail which might be similar here. Side note: there was a Bosch device in the past which could only be added via install code (at least that's what we thought). But the problem might be a different one: we have seen in the sniffer log that the Bosch sensor reports the deviceId `0xffff` in the Simple Descriptor instead of one the Zigbee standard defined ones.

This `0xffff` value is also used by deCONZ to mark the Simple Descriptor as invalid and causes it to repeatedly query it not coming to the point to actually integrate it in the system.

There is a small workaround in the upcoming version but I'm not sure if it resolves the problem all together, would be cool if you can test it with v2.19.0-beta when it's out.

Schnabulation, 2022-10-14:
Of course! I'm more than willing to provide help and support as best as I can.

I am using deCONZ as a Home Assistant add-on at the moment. Would you kindly steer me into the right direction to where I could get the beta for this setup?

manup, 2022-10-18:
The stable version will be out this week, usually it's available in HA shortly after release and can be tested.

Schnabulation, 2022-10-19:
Thank you @manup - I'll report back once the HA Add-On has been updated.

In the meantime: I have noticed that there is also a device request for this Bosch Smoke Alarm II for the ZHA integration where it seemed to be possible connect the device and extract the model identifier and clusters. Are these informations maybe helpful too? Link to said device request: https://github.com/zigpy/zha-device-handlers/issues/1562

Schnabulation, 2022-11-18:
> As there has not been any response in 21 days, this issue has been automatically marked as stale. **At OP: Please either close this issue or keep it active** It will be closed in 7 days if no further activity occurs.

I am still waiting for the implementation of this device: At the moment it sits on my desk unused.
If I can provide further information please let me know.

Puncher4444, 2022-11-22:
Same here - also waiting for implementation and happy to help for testing (Home Assistant, deconz).

Schnabulation, 2022-12-01:
May I ask: what would be the next steps to make this device work? I work in IT and have a test lab at my disposal. If I can help I'd gladly do so.

antaron1973, 2022-12-16:
I'm also not sure if Bosch Smoke Alarm II is being under work or not. I saw a release note for v2.19.3 which I run, but when trying to pair the Bosch Smoke Alarm II via Phoscon/deCONZ it wasn't found nor displayed in deCONZ.
Are there any new developments?

Schnabulation, 2023-01-09:
I am still waiting for the implementation of this device: At the moment it sits on my desk unused.
If I can provide further information please let me know.

antaron1973, 2023-01-29:
Is it possible to implement Bosch Rauchwarnmelder II (Smoke Alarm II) - as this is Bosch and they have a successful eco-system for smartphone I feel this important, as this is one of the few certified and affordable smoke alarms. I have one device here, but it's not even pairing. My Bosch bridge allows it of course, but I don't want to use a second Zigbee network, but instead want to use my strong Conbee II net.

Schnabulation, 2023-01-30:
I resorted to using the Bosch Smart Home Controller (ZigBee gateway for the Bosch Rauchwarnmelder II): it integrates well into Home Assistant and works perfectly fine. I wanted my Rauchwarnmelder to work but as I see no progression in this ticket I bought the gateway. Would recommend the same because I doubt we will see any integration in the next time from deCONZ.

antaron1973, 2023-01-30:
@Schnabulation I did the same, bought a Bosch SHC II. The thing for me is: I have a strong network of Zigbee router devices in Conbee II and the Bosch smoke alarms are passive, because of the batteries. That means, they all need to connect directly to the Bosch SHC II, which is a problem in my house with many walls. So still, my wish would be to integrate them into my Conbee II Zigbee network. I know, Bosch, Gardena, Philips all want their own nets and bridges, but in truth this would just create several week networks. This is why we need integration of important devices manufacturers like Bosch, rather than strange and intransparent brands from Asia.

Schnabulation, 2023-01-30:
I 100% agree and that was basically the reason I wanted to integrate the smoke alarms in my existing ZigBee net. Also I didn't want to convolute my existing ZigBee network with a separate one. Also I wanted to use the siren for my alarm system - which is not possible with the Bosch integration.

But I have to say the SHC is pretty strong with its signal: it lives in my basement and has a perfect connection to all of my alarms - even the one in the top most floor.

tkintscher, 2023-01-30:
I have bought a Bosch smart plug (#6689) and have the same issue. The device is not even pairing, even with the install code set beforehand.

It did work immediately with another stick and zigbee2mqtt, thus it must be possible...

It also seems that folks such as @SwoopX got these devices to work. Maybe they can point into the right direction?

Puncher4444, 2023-01-31:
Actually, same on my side...
I´ve bought a couple of "Smoke Detector II" devices and have a Bosch Smart Home Controller in the meanwhile to use it in my Home Assistant system. Nevertheless, as I have a Conbee II Stick with lots of other Zigbee devices, it´s not really useful to have another Zigbee Gateway installed. It costs money for no reason and I really like to get rid of all the crappy gateways in my house.
Thats the reason why I really think these Bosch II devices should been added into the Deconz supported list. I´m really willing to help testing etc.!

Schnabulation, 2023-02-22:
As always: issue not resolved - don't close!

Puncher4444, 2023-03-06:
@manup I‘ve seen you guys have merged the DDF for the Bosch Thermostat II in PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6565. 
Did you managed to get the issue with the install code for the Bosch devices to be solved!? If yes, how? Really want to help here to get the Rauchmelder II device of Bosch working.

Schnabulation, 2023-03-06:
Also: I noticed that zigbee2mqtt has the Bosch Smoke Alarm II working perfectly fine with using the install code to pair it.

Puncher4444, 2023-03-23:
Seems that there is some movement on the install-code topic - cant wait for it!

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6791

Schnabulation, 2023-03-23:
YES, finally! It looks like a Smoke Alarm II was paired successfully using the install code. So if the PR gets merged we can suspect to finally be able to integrate our devices.

burpy, 2023-04-02:
I was able to connect the Bisch Smoke Alarm II successfully with the install code because of [this tutorial](https://forum.phoscon.de/t/bosch-water-sensor-bwa-1-how-to-set-up-using-install-code/1698/7). But I am not sure how to do the last step (set the MAC address of the Conbee II in the Smoke Alarm device).

There a two more problems. After stopping the Deconz-GUI the smoke alarm device disconnects and needs to pair again manually. And the clusters are not showing (see screenshot).

[image]

vjonas, 2023-04-03:
> I was able to connect the Bosch Smoke Alarm II successfully with the install code (and with deconz version 2.21.02) because of [this tutorial](https://forum.phoscon.de/t/bosch-water-sensor-bwa-1-how-to-set-up-using-install-code/1698/7). But I am not sure how to do the last step (set the MAC address of the Conbee II in the Smoke Alarm device).
> 
> There a two more problems. After stopping the Deconz-GUI the smoke alarm device disconnects and needs to pair again manually. And the clusters are not showing (see screenshot).
> 
> [image]

Hello Burpy

Can you point me in the direction on how to install the deconz-gui on a PI? 
I tried using the deconz-docker container and then routing to the phoscon app(http://phoscon.de/pwa/login.html?1679567896558) did not allow me to open it.
Any suggestions?

burpy, 2023-04-03:
> Can you point me in the direction on how to install the deconz-gui on a PI? I tried using the deconz-docker container and then routing to the phoscon app(http://phoscon.de/pwa/login.html?1679567896558) did not allow me to open it. Any suggestions?

Maybe this helps you: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1405. The web GUI alone isn't able to do this.

Schnabulation, 2023-04-25:
Still waiting for the integration into deCONZ GUI.

Mimiix, 2023-04-25:
You are able to pair it, as support for install codes has been added.

Mimiix, 2023-04-25:
See: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6791

Schnabulation, 2023-04-25:
@Mimiix Where exactly in the deCONZ GUI do I find the field to insert the install code?

Mimiix, 2023-04-25:
> @Mimiix Where exactly in the deCONZ GUI do I find the field to insert the install code?

You need to use the RestAPI. See the PR i linked.

Puncher4444, 2023-04-25:
Par was merged and is already available in beta version, will be available in Home Assistant once next stable version is released and published to HA, See changelog: https://github.com/dresden-elektronik/deconz-rest-plugin/releases

Schnabulation, 2023-04-25:
> Still waiting for the integration into deCONZ GUI.

Then I'd like to quote my initial reply: Still waiting for integration into the GUI as we don't have access to the RestAPI of the HA plugin, I believe.

Mimiix, 2023-04-25:
> > Still waiting for the integration into deCONZ GUI.
> 
> Then I'd like to quote my initial reply: Still waiting for integration into the GUI as we don't have access to the RestAPI of the HA plugin, I believe.

You do :) You can use the restAPI in HA. Simply by forwarding the ports in the addon (Default is 40850 IIRC). After that, you can simply access the restapi using yourhaip:40850/api/ But as @Puncher4444 mentions: HA decides to wait for stable releases. Small fun fact: The HA integration uses the rest api ;) Additionally, Once this device has been integrated by someone using the install code, i am not sure if there's a DDF present, which would be the next step.

@tsloms Mind commenting why there's a downvote on my comment?

tsloms, 2023-04-25:
@Mimiix as that Was no answer to the question. As @Schnabulation already state, we are awaiting a function with in the gui. But as you stated in you latest Post that is something that you believe can take even longer.

Mimiix, 2023-04-25:
> @Mimiix as that Was no answer to the question. As @Schnabulation already state, we are awaiting a function with in the gui. But as you stated in you latest Post that is something that you believe can take even longer.

As far as i am concerned: This issue is to get a device work with deCONZ. A Developer created the possibility, a good an proper one, to integrate it. In that sense, we are on the right track.

Requesting to have that feature in the gui, be it phoscon, be it deCONZ, is a feature request and that has to be moved to the forums.   

We need to keep these 2 things seperated :)

Mimiix, 2023-04-25:
> > I wonder why it should take longer as I thought this Software IS developed by DE? Correct me if I'm wrong!
> 
> Because DE is not that really fast in developing things. If you need proof: This feature was added by a volunteer dev, as most of the development is (including myself).
> 
> **Nevertheless, we are getting offtopic. I am happy to discuss anything on their appropriate places (Discord / Forums). But i'd really like to have this issue on-topic for the device integration. To proceed: Anyone that is able to use the Rest API (after V2.20.0) and Pair the device using the code, please comment and provide the screenshots that are required to make this device work :)**

Puncher4444, 2023-04-25:
@Mimiix as soon as the stable version is available in HA, be sure I get my hands on. Currently I have no chance to use beta, as HA doesnt support Beta of deconz addon and my only Conbi stick is in productional use

Schnabulation, 2023-05-17:
Still waiting for integration into the GUI

Mimiix, 2023-05-17:
@Schnabulation Not going to happen with this issue as mentioned. I recommend opening an request on the forum for that.

I am happy to wait until the current way of using the Install codes goes Stable. If anyone is able to provide Screenshots after that, we can proceed here.

Schnabulation, 2023-06-08:
[image]

closed 2023-07-09 by github-actions[bot]

tsloms, 2023-07-09:
Bought a Sonoff stick. That integrates much better and even Supports install Codes. Migrated all devices one by one. Had no issues since than and the setup is much more stable.
BTW I'm using z2m