dresden-elektronik/deconz-rest-plugin #6422: MOES Smart Dimmer Switch  ZS-EUD 1 gang (TS0601  _TZE200_ip2akl4w)

imammedo, 2022-10-14:
## Device

- Product name: MOES Smart Dimmer Switch  ZS-EUD.
- Manufacturer: _TZE200_ip2akl4w
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Other:  Dimmable light controller 

Info about sensor: https://www.aliexpress.com/item/1005003572427505.html?spm=a2g0o.order_list.0.0.16b81802eQtguy
[image]

Basic DDF:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_ip2akl4w",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": false,
  "status": "Draft",
  "path": "/devices/ts0601.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/group"
        },
        {
          "name": "state/alert",
          "public": false,
          "default": "none"
        },
        {
          "name": "state/bri",
          "refresh.interval": 84000,
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 2,
            "dt": "0x2b",
            "eval": "(Item.val / 254.0) * 1000.0;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 2,
            "eval": "Item.val = (Attr.val / 1000.0) * 254.0;",
            "fn": "tuya"
          }
        },
        {
          "name": "state/on",
          "refresh.interval": 300,
          "read": {
            "fn": "tuya"
          },
          "write": {
            "dpid": 1,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```
with it, it's possible to control device individually via web GUI, however 
group control doesn't work (either web GUI or when group is controlled by sensor)

## Screenshots
[image]

### Basic
[image]

### Tuya specific
[image]

### Groups
[image]
[image]

imammedo, 2022-10-25:
"config/group" was my invention to try and see if it helps, but it's the same with without this entry.

When I say group, I mean light group in phoscon web gui, where one assigns some lights, sensors and specify rules.
Whether that translates into some kind of bind or into zigbee group, I don't know.

Manually I haven't tried to bind anything (I have only vague idea what it is).

In my case 'light group' consists this dimmer, one IKEA bulb and IKEA motion sensor.
The sensor can send on/off command to group address but it will have effect only on bulb that has corresponding cluster,
and I assume since the dimmer doesn't have on/off cluster the command gets ignored buy it.
The same happens when one toggles state of whole 'light group' in Phoscon GUI.

And looking at DDF examples, I'm not sure if it possible to make deconz control such devices individually or send an extra on/off command to group using tuya cluster. (Yep, that will make dimmer work only if deconz is running, but it's probably not possible to convince standard motion sensor to manage tuya specific cluster directly)

Smanar, 2022-10-25:
Haaaa yes, I think you have found something ^^
If you take a normal remote, it will make brodcast on cluster on/off for exemple the cluster 0x0006. Deconz do same.

But your device don't have this cluster, so even it have the group cluster, it's not possible make broadcast request to tuya device.

And different device can have too different request on the tuya cluster for the same command ^^

imammedo, 2022-10-26:
Then the only route that will work with non standard devices is for deconz to translate group on/off or level commands to unicast ones (as described in DDF) if the device doesn't have target cluster that is used by broadcast.

i.e. workflow could look like:

1.  light group gets on/off request (either via phoscon GUI, REST request or event from motion sensor (zigbee source))
2. deconz sends broadcast on/off to zigbee group if there is at least a device that supports on/off cluster
3. if light group contains lights without on/off cluster, deconz should send unicast requests to such lights if DDF provides alternative 'state/on' behavior.

the same goes for level or 'state/bri' control.

Lets see what @manup can say if it's possible to implement in deconz .

PS:
as workaround I can probably specify a custom rule via REST API that would do above steps, triggered by group/sensor state change. But that adds extra manual configuration to maintain, so it would be better if deconz could do it transparently
(current behavior for folks unaware of zigbee internals, it looks like a bug)

imammedo, 2022-10-26:
> It's not good at all for me, and realy bad for tuya.

It doesn't have to be a unicast, it could be a series of group requests 1 req per nonstandard model/vendor light combination.

> Group request are here to lighter request. Making a broadcast request is better for big zigbee network than making unicast.

it depends on the size of the group, if it's group of one light and one sensor + deconz itself,
then cost of unicast should be the same as a group broadcast to the same group.

> If the device can work only with unicast, making group is totally useless for it.
yep, if light doesn't support groups then making zigbee group for it is useless.

But if one would think of a group as more abstract thing ( i.e. REST API group, ex: 'room')
then having groups makes sense as one still can put a bunch of other of devices in there
and from API point of view it doesn't really matter if devices within REST group are broadcast
or unicast capable, all should be turned on/off when the group is turned on/off  via REST API.

>If you use direct bind on a group, it's remote that send the brodacast request, and it will never make unicast for not compatible device.

it's true only in case of pure zigbee devices.
However that's where software gateway can mediate and help. For example:
1. motion sensor triggers on/off command
2. deconz receives it and translates to a series of broadcast or in worst case unicast zigbee messages to all lights within the group.

Actually deconz already does above for sensors that do no have on/off cluster, the problem is that gateway sends only on/off message to the controlled group, where as it could send several commands (depending on devices within group) to the same group or unicast if device(s) doesn't support groups.

Smanar, 2022-10-26:
But in your exemples, better to use simple automation with third app, no need zigbee group. The first utility of zigbee group IS broadcast.

imammedo, 2022-10-27:
Lets say the less apps in control chain the better from reliability/maintenance point of view.
Moreover if I have several such dimmers in the room, it would very much make sense to put them in the same zigbee group and make it work with gateway as expected.

Using third party app, would work as workaround too. The same as creating custom rule(s) in deconz via REST API, which is also workaround. However with that I loose phoscon's 'light groups' with its sensor/schedule/scenes/switch controls
and a lot of overhead maintaining it in working condition which quickly becomes pian to manage with more devices added. 

Workarounds also don't fix currently broken group control via REST API. (zigbee group != REST API group)
Point of gateway is hide zigbee impl. details from end user (whether deconz will use zigbee broadcast or
unicast behind the scenes, end user doesn't really care). What user cares about if dimmer(s) is put in REST API group
it should be correctly controlled when on/off command issued to the whole group via REST API.

But this doesn't look like request new device anymore but more like a bug fix in deconz internals, maybe I should open a new issue asking for bugfix.

Smanar, 2022-10-28:
zigbee group == REST API group.
Third app group != zigbee group

And it s not a bug but an improvement. the working mode is normal for me.

closed 2022-11-26 by github-actions[bot]

cornim, 2023-03-10:
The same config works for me with smart dimmer (rotating knob) from tuya having "manufacturername": "_TZE200_p0gzbqct"

Smanar, 2023-03-10:
Ha yep thx, I will make a PR to add thoses 2 devices

cornim, 2023-03-10:
I de-installed the dimmer switch for now but I can check it in about 3 weeks if that's ok.  
I did look at the other json though and can say that it looks basically identical.

I do have a question about both of them though:  
the data type `dt` is specified as `0x10` and `0x2b` for them but the [Tuya developer docs](https://developer.tuya.com/en/docs/iot/tuya-zigbee-universal-docking-access-standard?id=K9ik6zvofpzql#title-3-Clusters%2C%20attributes%2C%20and%20commands) state under DP data format that the data types have values from `0x00` to `0x05`. I also tried `dt` as `0x01` and `0x02` and it also seems to work.  

Any ideas as to why?

Smanar, 2023-03-11:
Yes ^^, it's not the tuya type but the deconz ype

```
    - dataType: string hex value

          bool           0x10
          s32 value      0x2b
          enum           0x30
          8-bit bitmap   0x18
          16-bit bitmap  0x19
          32-bit bitmap  0x1b
```

But strange it can work with bad value, deconz probably use a defaut one if the one used in the DDF is not reconised.

>I did look at the other json though and can say that it looks basically identical.

Yeah for me, we just need to edit the DDF to add the both device, like that

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_9cxuhakf", "_TZE200_p0gzbqct" "_TZE200_ip2akl4w"],
  "modelid": ["TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya Genric",
  "product": "Smart Dimmer Switch 1 gang",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none",
          "public": false
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"},
          "refresh.interval": 300
        },
        {
          "name": "state/bri",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = (Attr.val / 1000.0) * 254.0;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "(Item.val / 254.0) * 1000.0;"},
          "read": {"fn": "none"},
          "refresh.interval": 84000
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

cornim, 2023-03-11:
Thanks for the explanation regarding the data types.  
Could I have found that by myself, i.e., is this documented anywhere so I don't have to bug you with these kinds of questions in the future?

Smanar, 2023-03-12:
Honnestly, documentation on deconz need to be improved.
This comment is from the code itself, so invisible for you. Better to ask, not sure you will have an answer, but cost nothing.