dresden-elektronik/deconz-rest-plugin #6423: Aubess 20A smart plug Zigbee 3.0

Delusional112, 2022-10-15:
## Device

- Product name: Aubess Smart Plug
- Manufacturer: _TZ3000_gvn91tmx
- Model identifier: TS011F
- Device type : Please remove all unrelated device types. 
 
  - Switch
  

## Screenshots

### Basic

[image]

[image]

[image]

### Identify

[image]

### Alarms

### Device Temperature

### Groups

[image]

### Scenes
[image]

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering
[image]

[image]

### Diagnostics

### Other clusters that are not mentioned above

Delusional112, 2022-10-15:
Hi guys, I already tried myself with the DDF editor and managed to get something to work. But I'm really a noob to this. I tried replicating a working tuna plug. But it looks like I did something wrong because the consumption value in HA can't be right.

[image]
[image]
[image]
[image]

[image]

Delusional112, 2022-10-16:
Ok, I found the problem with the KWH not showing the right value. 

3dccc90bca324d3983a7669a0e596034

3dccc90bca324d3983a7669a0e596034

Smanar, 2022-10-17:
It's because the value switch from 0 to the real value ?
Like having 10/0/10/0/11/0/11/0/12/0 ?

You can probably see it using the flag "DDF" in deconz log.

Delusional112, 2022-10-17:
@Smanar Ok, seems logical. Is it to be fixed with changing a value in the DDF file? Cause the Tuya device I copied the code from doesn't have this issue.

Smanar, 2022-10-18:
It's a question ^^. And will be nice if you can see the defaut on logs, to see how to reconize it and ignore it in the "eval" part for exemple.

We can't ignore the 0 value, because sometime it can be 0, hopping for something more usefull in logs.
Can check for exemple the current value.
Or can be an issue too on deconz code.

Delusional112, 2022-10-18:
[image]

@Smanar here is the log. No idea what to look for

Smanar, 2022-10-18:
Arf my idea was to check state/current, to see if the 0 value is good or not, but bad luck state/current have the same issue, the 0 value.

And I don't see how to see the real value in deconz logs, but was probably 0 realy.

Have you try to set a bigger report interval ? Try to double all value for the cluster 0x0b04 "min" "max" and "reportable change"

Delusional112, 2022-10-20:
I have no clue how to interpret al these values. But I'll try to follow your advice. 

I changed these values as seen in the screenshots, but no succes. Still the same.

[image]

[image]

Smanar, 2022-10-21:
Ha fuck, I have bad read your logs

You have the 0 value only for 1 device, the one that have the mac adress finishing by 71
Your actual device finish by 53 and on your log you never have the 0 value

If you use phoscon / help / API Information / event / sensors  / ID , you can see all websckets notification for a sensor.

Try looking for 0 value in notification, but on your capture there isn't one

For the reporting/binding setting try to use bigger values for "min" it will decrease report (but it's just to make a test)

Delusional112, 2022-10-30:
@Smanar Just hooked some other devices to the same brand sockets. 

In this log you can see the 0 value. What can I do about it?

{
    "16:10:33:821": {
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "state": {
            "current": 4829,
            "lastupdated": "2022-10-30T15:10:33.831",
            "power": 266,
            "voltage": 224
        },
        "t": "event",
        "uniqueid": "a4:c1:38:c8:c9:e2:ed:ce-01-0b04"
    },
    "16:10:42:110": {
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "state": {
            "current": 4829,
            "lastupdated": "2022-10-30T15:10:42.094",
            "power": 288,
            "voltage": 224
        },
        "t": "event",
        "uniqueid": "a4:c1:38:c8:c9:e2:ed:ce-01-0b04"
    },
    "16:10:44:135": {
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "state": {
            "current": 4985,
            "lastupdated": "2022-10-30T15:10:44.092",
            "power": 288,
            "voltage": 224
        },
        "t": "event",
        "uniqueid": "a4:c1:38:c8:c9:e2:ed:ce-01-0b04"
    },
    "16:10:47:445": {
        "attr": {
            "id": "28",
            "lastannounced": null,
            "lastseen": "2022-10-30T15:10Z",
            "manufacturername": "_TZ3000_gvn91tmx",
            "modelid": "TS011F",
            "name": "Vaatwasser",
            "swversion": null,
            "type": "On/Off plug-in unit",
            "uniqueid": "a4:c1:38:7d:d4:7f:94:63-01"
        },
        "e": "changed",
        "id": "28",
        "r": "lights",
        "t": "event",
        "uniqueid": "a4:c1:38:7d:d4:7f:94:63-01"
    },
    "16:10:57:77": {
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "state": {
            "current": 4985,
            "lastupdated": "2022-10-30T15:10:57.088",
            "power": 263,
[... 920 lines cut ...]
            "name": "Power 28",
            "swversion": null,
            "type": "ZHAPower",
            "uniqueid": "a4:c1:38:c8:c9:e2:ed:ce-01-0b04"
        },
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:c8:c9:e2:ed:ce-01-0b04"
    },
    "16:16:30:209": {
        "e": "changed",
        "id": "28",
        "r": "sensors",
        "state": {
            "current": 128,
            "lastupdated": "2022-10-30T15:16:30.210",
            "power": 34,
            "voltage": 224
        },
        "t": "event",
        "uniqueid": "a4:c1:38:c8:c9:e2:ed:ce-01-0b04"
    }
}

Smanar, 2022-10-30:
```
"state": {
"current": 47,
"lastupdated": "2022-10-30T15:15:00.195",
"power": 0,
"voltage": 223
},
```

```
"state": {
"current": 1767,
"lastupdated": "2022-10-30T15:15:59.206",
"power": 170,
"voltage": 224
},
```

BTW, you don't need to divide your current by 1000 ?

But the 0 value in this logs is fine, look at the current value 0,047 A, when it's a bug the current value is 0 too. For me here the 0 vaue is fine, the power was < 1 W, measured by the device.

Delusional112, 2022-10-30:
So, what to do about those strange graphs? Because the devices draw continuous power from the sockets. And still it is measuring 0 watt a lot of times.

Smanar, 2022-10-30:
The device precision ?
You have more consumption between 15h06 and 15h20, but the value move with same "amplitude".

What is the device connected to it ? something stable without electronic ?

Delusional112, 2022-10-30:
@Smanar. I hooked a heating blanket to one of the plugs. Constant power drawn. Check the logs and look at the graph. 

Are you living in Holland by chance? I'll send you one of the plugs so you can experiment with them. 

{
    "17:50:36:861": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "current": 0,
            "lastupdated": "2022-10-30T16:50:36.915",
            "power": 53,
            "voltage": 217
        },
        "t": "event",
        "uniqueid": "a4:c1:38:74:20:53:22:1f-01-0b04"
    },
    "17:50:36:881": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "current": 0,
            "lastupdated": "2022-10-30T16:50:36.939",
            "power": 0,
            "voltage": 217
        },
        "t": "event",
        "uniqueid": "a4:c1:38:74:20:53:22:1f-01-0b04"
    },
    "17:50:51:977": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "current": 239,
            "lastupdated": "2022-10-30T16:50:51.921",
            "power": 0,
            "voltage": 217
        },
        "t": "event",
        "uniqueid": "a4:c1:38:74:20:53:22:1f-01-0b04"
    },
    "17:50:51:978": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "current": 239,
            "lastupdated": "2022-10-30T16:50:51.939",
            "power": 53,
            "voltage": 217
        },
        "t": "event",
        "uniqueid": "a4:c1:38:74:20:53:22:1f-01-0b04"
    },
    "17:50:55:795": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "current": 239,
            "lastupdated": "2022-10-30T16:50:55.867",
            "power": 53,
            "voltage": 217
        },
        "t": "event",
        "uniqueid": "a4:c1:38:74:20:53:22:1f-01-0b04"
    }
}

[image]

Smanar, 2022-10-30:
```
"state": {
"current": 0,
"lastupdated": "2022-10-30T16:50:36.939",
"power": 0,
"voltage": 217
},
```
>Are you living in Holland by chance? 

Nope I m french, but I can't do lot of thing, voltage is stable so not possible to use it to detect a bad 0 value.
And we can't ignore all 0 value, because, when you will turn off the plug and have a real 0 value, it will be ignored.
No changes if you increase the report setting ?

Delusional112, 2022-10-30:
I already had it on min. interval of 15. I set it to 40.

[image]

Smanar, 2022-10-30:
On Z2M they are using 10 as value for the reportable change for power, 5 for voltage, and 50 for current.

They are using the tuya unlock sequence but IDK if it have an impact on this issue.

Delusional112, 2022-10-31:
> On Z2M they are using 10 as value for the reportable change for power, 5 for voltage, and 50 for current.
> 
> They are using the tuya unlock sequence but IDK if it have an impact on this issue.

Tuya unlock sequence. Is that something I can activate in DDF?

Delusional112, 2022-10-31:
Ok. I’m out of options I think and need to accept this plug won’t work correctly with deconz. Too bad is guess

Delusional112, 2022-11-05:
> Please don't give up on this.

I'm not a developer. Just a bit handy with computers. 

So it's up to the guys of Deconz to heat his fixes I guess.

Mimiix, 2022-11-05:
I am reading all issues and redirect the most pressing ones. Just because it seems that Dresden doesn't have the capacity (for years) to respond to every issue and fix them all. Most devs are volunteers (myself included ) who do this in their spare time as a hobby. 

In this case Smanar needs something implemented and that seems to be taking way to long.

Sorry guys , trying to drive a car with flat tires here :(

Mimiix, 2022-11-27:
> 

Sorry, missed this one. 

He's waiting for this pr https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244.

thelvyn777, 2022-12-12:
Does someone already test this device with another add-on like Zigbee2MQTT, is-it OK ?

Delusional112, 2022-12-20:
> Sorry, missed this one.
> 
> He's waiting for this pr #6244.

HI @Mimiix Is something expected of me at this time?

Mimiix, 2022-12-20:
Nothing on you I'm afraid

1100101, 2023-02-04:
@Mimiix @Smanar 
#6244 has been merged, but this specific plug is still not recognized fully.
Is there any way we can assist in getting power/current reporting to work with this device?

Smanar, 2023-02-04:
The device "_TZ3000_gvn91tmx" is not in the PR you have linked.
It seem this one need the tuya unlock sequence to work else we have zero values for somes fields.

Mimiix, 2023-02-04:
So we have to reopen :)

1100101, 2023-02-04:
Zigbee2MQTT has the same issue, pointing to an issue with the firmware: https://github.com/Koenkk/zigbee2mqtt/issues/15333

Mincka, 2023-02-26:
Keep it open

Smanar, 2023-02-27:
Hello, can you try this DDF, it use the tuya unlock sequence

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "_TZ3000_gvn91tmx",
    "modelid": "TS011F",
    "product": "Random tuya plug",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
      {
        "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
        "restapi": "/lights",
        "uuid": [
          "$address.ext",
          "0x01"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "state/alert"
          },
          {
            "name": "state/on"
          },
          {
            "name": "state/reachable"
          },
        {
          "name": "config/configured",
          "refresh.interval": 86400,
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": [
              "0x0004",
              "0x0000",
              "0x0001",
              "0x0005",
              "0x0007",
              "0xfffe"
            ]
          },
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0xfffe",
            "eval": "Item.val = (Attr.val != 0)"
          },
          "public": false,
          "default": false
        }
        ]
      },
      {
        "type": "$TYPE_CONSUMPTION_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0702"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
          "name": "state/consumption"
        },
          {
            "name": "state/lastupdated"
          }
        ]
      },
      {
        "type": "$TYPE_POWER_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0b04"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
            "name": "state/current"
          },
          {
            "name": "state/lastupdated"
          },
          {
            "name": "state/power"
          },
          {
            "name": "state/voltage"
          }
        ]
      }
    ],
    "bindings": [
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0006",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x10",
            "min": 1,
            "max": 1800
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0702",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x25",
            "min": 5,
            "max": 300,
            "change": "0x0000000A"
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0B04",
        "report": [
          {
            "at": "0x0505",
            "dt": "0x21",
            "min": 5,
            "max": 500,
            "change": "0x00000003"
          },
          {
            "at": "0x0508",
            "dt": "0x21",
            "min": 5,
            "max": 300,
            "change": "0x0000000F"
          },
          {
            "at": "0x050B",
            "dt": "0x29",
            "min": 5,
            "max": 300,
            "change": "0x00000003"
          }
        ]
      }
    ]
  }
```

1100101, 2023-03-02:
I placed the json file according to the DDF Cheat sheet, but it does not look like it works:
```
2023-03-02 14:04:48 HUEDevice plug.aubess.HomeOffice.Power voltage: 236
2023-03-02 14:04:48 HUEDevice plug.aubess.HomeOffice.Power current: 96
2023-03-02 14:07:30 HUEDevice plug.aubess.HomeOffice.Power power: 0
2023-03-02 14:07:30 HUEDevice plug.aubess.HomeOffice.Power current: 0
2023-03-02 14:07:35 HUEDevice plug.aubess.HomeOffice.Power current: 88
2023-03-02 14:07:35 HUEDevice plug.aubess.HomeOffice.Power power: 9
```
The devices still send `0` values in-between.

Smanar, 2023-03-02:
You can see them on the GUI (on your missing capture ^^), take a look on cluster 0b04 for exemple, it's called mutiplier or divider, but not supported on all devices.

1100101, 2023-03-02:
[image]
[image]
As you can see, for cluster 0x0702 no Multiplier or Divisor is specified, which I believe defaults them to 1.
But the value provided in ID 0x0000 "Current Summation Delivered" cannot be kWh:
I installed this plug only about 2 weeks ago, it measures the consumption of an LED display and a laptop. 
There's no way these 2 devices consume 55kWh, especially considering that I haven't used that arrangement much in these 2 weeks.

That said, 550W is a more realistic value, wich is the existing value multiplied by 10.

Smanar, 2023-03-03:
So bad luck, this device don't have value, need to make test and use a fixed value in the DDF

W is power not consumption.

A laptop can consume 100 W so for 2 weeks : 100 * 24 * 14  =33 Kw/h

Mincka, 2023-03-08:
For anyone only interested by the switch feature of this plug, I can confirm that **it already works just fine when added as a light in Deconz 2.20.1**, and also in Home Assistant. I prefer to share this since it's not clear what's still unsupported in the original post.

closed 2023-04-07 by github-actions[bot]

simical, 2023-06-20:
Maybe there is a chance to get the ddf to deconz? On/Off works fine, but not the power measurement. And in HA all individual DDFs are gone after a restart.

Smanar, 2023-06-20:
Have you tried the DDF ?
You have the 0 value issue too ?