dresden-elektronik/deconz-rest-plugin #6426: ADUROSMART ERIA LIGHT BULB

wifiuk, 2022-10-16:
# Device

- Product name: ERIA
- Manufacturer: ADUROSMART
- Model identifier: doesn't show in deconz
- Device type : 
  - Light
  

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

The product says ZIGBEE CERTIFIED PRODUCT, but when it is in pairing mode its not detected at all with deconz/phoscon.

https://www.vesternet.com/products/zigbee-aduro-b22-bulb-warm-white

Mimiix, 2022-10-16:
Hi,

As this template is basically empty ,I am closing this issue.

Feel free to open a new one with the required screenshots.

Kind regards,

closed 2022-10-16 by Mimiix

wifiuk, 2022-10-16:
I moved over to zigbee2mqtt instead