dresden-elektronik/deconz-rest-plugin #6427: Aqara Curtain Driver E1

dxa13, 2022-10-16:
## Device

- Product name: Aqara Curtain Driver E1
- Manufacturer: LUMI
- Model identifier: lumi.curtain.agl001
- Device type : Please remove all unrelated device types. 
  - Other: Battery Powered curtain driver with light sensor

## Screenshots

[image]
[image]

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Window Covering:
[image]
[image]
[image]
[image]

dxa13, 2022-10-16:
Addendum to the above:  I tried adding the device repeatedly but ultimately does not show up in the Phoscon App.  If it helps, this new device may be a close relative to the Aqara Roller Shade Driver E1 device described on the following issue, except it adds a light sensor:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5330

Smanar, 2022-10-16:
>If it helps, this new device may be a close relative to the Aqara Roller Shade Driver E1 device described on the following issue, except it adds a light sensor:

Yes, it seem, it's the issue.
The device is an enddevice, but covering are router by defaut.

I think it can solved with mimic a DDF for another covering with the same issue, like the `devices\robb_smarrt\rob_200-029-0_covering.json`

Smanar, 2022-10-29:
cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6469

closed 2022-11-28 by github-actions[bot]