dresden-elektronik/deconz-rest-plugin #6429: _TZ3000_typdpbpg_smart_plug_eu

tompijls, 2022-10-17:
I have above plug. It seems to be sold by multiple vendors. 

There is a DDF available for Tuya, see here: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/153908be947a18ce5f9104778e6e5adc3126f8dd

Mine requires vendor to be TS011F instead of Tuya. Everything else remains unchanged

FYI: the original file is missing the json extension in the repository.

## Screenshots
[image]

closed 2022-10-17 by close-issue-app[bot]

VivantSenior, 2022-10-18:
I created a copy of file `_TZ3000_typdpbpg_smart_plug_eu`, add it to `/data/.local/share/dresden-elektronik/deCONZ/devices/_TZ3000_typdpbpg.json` and the device started to share power and energy usage in Home Assistant.

tompijls, 2022-10-18:
Possibly. I first changed vendor and then appended .json so you may be right.
Anyway, the json extension should be added.

Cheers

VivantSenior, 2022-10-18:
@tompijls could you please reopen this ticket (and change label)? I think the issue with missing extension still exists for majority of users.

Best

tompijls, 2022-10-18:
How do I do that? I don't see an option to re-open (and got discouraged by the auto-close in all fairness)

Mimiix, 2022-10-18:
You can't re open when the topic does not comply with the template.

VivantSenior, 2022-10-18:
Ok, no need to reopen. I've created a PR #6434 to add missing extension.

Best