dresden-elektronik/deconz-rest-plugin #6436: Aqara Roller Shade Driver E1 - Shows up as Light and does not update cover position.

oneseventhree, 2022-10-19:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: PC
 - Running method: Windows
 - Firmware version: 26660700
 - deCONZ version: 2.19.00 / 19/09/2022
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
N/A

## Additional context

Mimiix, 2022-10-21:
Hi,

can you show what it shows in the RestAPI when you moved it?

oneseventhree, 2022-10-21:
> Hi,
> 
> can you show what it shows in the RestAPI when you moved it?

Do you mean this:
```
    "5": {
        "etag": "d5a9e68292a8055a84dcaf324515629c",
        "hascolor": false,
        "lastannounced": "2022-10-20T23:21:27Z",
        "lastseen": "2022-10-21T10:04Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.curtain.acn002",
        "name": "Living Room Blinds",
        "state": {
            "alert": "none",
            "bri": 38,
            "lift": 15,
            "on": true,
            "open": true,
            "reachable": true,
            "speed": 2
        },
        "swversion": "0x00000E1B",
        "type": "Window covering device",
        "uniqueid": "54:ef:44:10:00:44:ee:85-01"
    }
}
                    
                ```

Mimiix, 2022-10-21:
Did you try to re-pair already? In the original topic i dont see anyone mentioning this.

oneseventhree, 2022-10-21:
> Did you try to re-pair already? In the original topic i dont see anyone mentioning this.

I've repaired it 3 times

Mimiix, 2022-10-21:
I'm pinging you on discord :)

Mimiix, 2022-10-21:
Got it working, in the end, the Firmware update on the Conbee made it work. 

For future reference:

- Reset device.
- Remove from Phoscon.
- Stop deCONZ
- Update Conbee firmware to latest version
- Start deCONZ
- Pair Blind again.
- ???
- Profit!

closed 2022-10-21 by Mimiix