dresden-elektronik/deconz-rest-plugin #6437: Missing current and voltage measurement from Tuya TS011F _TZ3000_typdpbpg

VivantSenior, 2022-10-19:
## Describe the bug

Hi,

I have Tuya TS011F _TZ3000_typdpbpg plug which reports to Home Assistant power but according to the Electrical Measurement cluster information:
[image]
this plug also supports current and voltage information.
I tried to compare different DDFs (supposedly to handle current and voltage) e.g. [BlitzWolf SHP13](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/blitzwolf/bw_shp13_smart_plug.json) but I wasn't able to find any relevant differences based on which I could modify DDF of _TZ3000_typdpbpg to add current and voltage handling.

This fix could be possibly added to other DDFs of TZ3000 plug e. g. _TZ3000_rdtixbnu (I have this plug and could test it).
_TZ3000_wxtp7c5y doesn't have power management capabilities (neither current nor voltage).

## Steps to reproduce the behavior

1. Add Tuya TS011F _TZ3000_typdpbpg plug to deCONZ.
2. Wait for electrical data to be transferred to Home Assistant.

## Expected behavior

Current and voltage data to be moved to Home Assistant.

## Screenshots

This is what Home Assistant receives from deCONZ (using [this DDF](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6434)):
[image]

Basic cluster:
[image]

Simple Metering:
[image]

Electrical Measurement:
[image]

## Environment

 - Used DDF: [here](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_typdpbpg_smart_plug_eu) (I've created PR #6434 to fix missing extension)
 - Home Assistant OS: 9.2
 - Home Assistant: 2022.10.4
 - Supervisor: 2022.10.0
 - Home Assistant deCONZ add-on version: 6.16.0
 - Host system: Raspberry Pi 4B
 - Running method: Home Assistant deCONZ Add-on
 - Firmware version: (26780700)
 - deCONZ version: (2.18.02 / 19.09.2022)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes: SSD drive (it doesn't cause any interferences with Zigbee network)

## deCONZ Logs

## Additional context

Mimiix, 2022-10-19:
Hi can you please open a device request for this?

VivantSenior, 2022-10-19:
@Mimiix, I added a comment in previously created device request (#5639).

Smanar, 2022-10-19:
But on your capture, values are 0 too ?
Mean the device don't support them or use the Xiaomi cluster instead, and the DDF is made for classic cluster.

VivantSenior, 2022-10-19:
The values are 0 because there was no load attached to the plug. Voltage value is correct (between 230 and 240).

Sorry I don’t know what do you mean by “Tuya unlock sequence”.

Mimiix, 2022-10-20:
What does the rest API show you:? I remember the voltage (in HA) is a attribute of one of the two.

Smanar, 2022-10-20:
For me the DDF is fine.
But I think this device need the tuya unclock sequence to work (so will be not for tommorow)

VivantSenior, 2022-10-20:
> What does the rest API show you:? I remember the voltage (in HA) is a attribute of one of the two.

You are right. I was able to find current and voltage as attributes to the Power entity:
[image]

Normally in other ESP based plugs I can see these values as separate entities (configured using [ESPHome](https://esphome.io/)):
[image]

This is how the Tuya based plug looks like in HA:
[image]

The value of current is off. Multiplying current (A) and voltage (V) should give power (W).

Mimiix, 2022-10-20:
> You are right. I was able to find current and voltage as attributes to the Power entity: [image]
>
> Normally in other ESP based plugs I can see these values as separate entities (configured using [...]

So just to be sure: 

- voltage is there but not as a own sensor which is correct. 
- current is off. Note that the values in the deconz gui is what the device reports. If it's not correct , it's a fault in the device. Scaling we can adjust tho. @Smanar can help with that.

Smanar, 2022-10-20:
But the binding is fine for me, in the cluster 0x0B04

```
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
```

can try more aggresive with replacing
```

        {
          "name": "state/current",
          "refresh.interval": 360
        },
```

by

```
        {
          "name": "state/current",
          "refresh.interval": 300,
          "read": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "eval": Item.val = Attr.val;"
          }
        },
```

But for me the DDF is fine

VivantSenior, 2022-10-21:
Yes, you are right - "basic" DDF (after this change - PR #6434) shows current and voltage as attribute of power entity (current is off by a factor of ~1 000 - probably that's how Tuya handles fraction).

After adding "more aggressive" approach nothing has changed in terms of the values. I think this will address the 1000 factor:

>  "eval": "Item.val = Math.round(Attr.val / 1000 * 100) / 100"

Smanar, 2022-10-21:
This code is reading the attribute 0x0508 (the current) and use it as it, so can use convertion, but you need to have other value than 0

closed 2022-11-20 by github-actions[bot]