dresden-elektronik/deconz-rest-plugin #6438: Nous Termperature and Humidity Sensor E5

cornim, 2022-10-19:
## Device

- Product name: Nous Termperature and Humidity Sensor E5
- Manufacturer: _TZ3000_lbtpiody
- Model identifier: TS0201
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

cornim, 2022-10-19:
I wasn't sure what to do with all the headings at the bottom of the ticket, as it said in the ticket not to delete any headings but lots of them don't really relate to my device.
Also the forum seems to be offline so I couldn't read the relevant post there.
If I need to fix something, please let me know.

Smanar, 2022-10-19:
No if all is working for you I will make the PR.
So all is fine ?

cornim, 2022-10-19:
The connection seems to work fine.
Temp readings are fine as well but the humidity is way to high. 74% where all other sensors show 55-60%.
Might be that the sensor is broken. I bought a second one and will try it out. Should take only a couple of minutes.

Smanar, 2022-10-19:
It s from the device, I m just reading value.
There is a config/offset to correct the value but 74 % to 55 % it s realy hugde, I don't think it will be reliable using an offset.

cornim, 2022-10-19:
Yes, so in summary, the deconz integration works fine but the humidity sensor seems broken.
I also tested it in a really humid environment and there it shows 100%. So the max value is not somewhere else but the measured values are just wrong.

cornim, 2022-10-20:
yeah, offset won't work because the error isn't linear
58% -> 74%
but 100% -> 100%
I'll just send those back.

But yes, go ahead and make the PR. Reading out the values via deConz works just fine.
And thanks again for the quick response.

Smanar, 2022-10-20:
>I put it in a large pot with a small pot of boiling hot water next to it. :)

Lol pro tester.

closed 2022-11-19 by github-actions[bot]