dresden-elektronik/deconz-rest-plugin #6440: Merten Wiser Socket outlet (working DDF attached)

jensflorian, 2022-10-19:
## Device

- Product name: MEG2380-0325
- Manufacturer: Schneider Electric
- Model identifier: SOCKET/OUTLET/1
- Device type : 
  - Sensor
  - Switch

There is a new wall socket outlet from Schneider Electric
(https://www.se.com/ww/en/product/MEG2380-0325/socketoutlet-merten-system-m-wiser-single-schuko-16-a-connected-active-white-glossy/)
Its pretty similar to the Fuga LK/SOCKET/1 

To join it in Zigbee network press button 3x in less than 0,5sec.
To reset, press 3x then hold button for 10sec

## Screenshots
[image]

### Basic
[image]

### On/Off
0x06
Nothing special here

### Simple Metering
0x06
[image]

### Electrical measurement
0x06
[image]

I used the LK fuga JSON as template.
With this modified DDF the device is fully supported. @Smanar : Could you create a PR for this? Or even add it to the existing LK fuga JSON to support both devices from this commit: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6095/commits/384236da792d36e57a453a77a07559f8485b89e2
?
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "SOCKET/OUTLET/1",
  "product": "SOCKET/OUTLET/1",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x06"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5,
          "read": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 6,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 6,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x06",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300,
          "read": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 6,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 6,
            "eval": "if (Attr.val != 65535) { Item.val = Attr.val; }"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300,
          "read": {
            "at": "0x0400",
            "cl": "0x0702",
            "ep": 6,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0400",
            "cl": "0x0702",
            "ep": 6,
            "eval": "if (Attr.val != -32768) { Item.val = Attr.val / 1000; }"
          },
          "default": 0
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300,
          "read": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 6,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 6,
            "eval": "if (Attr.val != 65535) { Item.val = Attr.val; }"
          },
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x06",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 300,
          "read": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 6,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 6,
            "eval": "Item.val = Attr.val"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 6,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 6,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 6,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0400",
          "dt": "0x2A",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}
```

Smanar, 2022-10-19:
Sure thx, have mixed both DDF here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6441

jensflorian, 2022-10-20:
I just found this in the logs. Do you think it is relevant? Device is working anyway.

20:14:12:308 [INFO] - No button map for: SOCKET/OUTLET/1, unicast to: 0x0000, endpoint: 0x06, cluster: 0x0702, command: 0x0A, payload: 00042AD91C00, zclSeq: 105
20:14:12:309 	payload: 00042ad91c0020:14:24:248 [INFO] - No button map for: SOCKET/OUTLET/1, unicast to: 0x0000, endpoint: 0x06, cluster: 0x0702, command: 0x0A, payload: 00042AC71C00, zclSeq: 111

Smanar, 2022-10-21:
No don't worry, button map is only used for switches.

>  cluster: 0x0702, command: 0x0A

But btw, I have checked the documentation and nothing about this command for this cluster ...

closed 2022-10-21 by manup

SwoopX, 2022-10-21:
> No don't worry, button map is only used for switches.
> 
> > cluster: 0x0702, command: 0x0A
> 
> But btw, I have checked the documentation and nothing about this command for this cluster ...

A simple attribute report comand perhaps... 😉

Smanar, 2022-10-21:
> A simple attribute report comand perhaps... 😉

Ha right, 0x0A is the report attribute command.