dresden-elektronik/deconz-rest-plugin #6443: Switch 2 gang (_TZ3000_zmy4lslw TS0002) both channels are switched in parallel

Paalap, 2022-10-20:
## Describe the bug

Both channels are working independly using the hardware switches/contacts. But via Phoscon both channels are always triggered together - the channels are "binded". Is this a knowd issue? At least I was not able to find anything in the closed issues.

Of course it would be really good to have the proper control of each channel. 

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian (headless)
 - Firmware version:  26720700
 - deCONZ version: 2.17.01
 - Device: ConBee II 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Mimiix, 2022-10-20:
I think it's just a case of support not being added for it properly.

I'd recommend opening a device request.

Mimiix, 2022-10-20:
As simple as stopping and starting :)

Paalap, 2022-10-21:
New DEvice REquest #6446 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6446#issue-1418303761

closed 2022-10-21 by Paalap

GitMouth, 2022-12-28:
Same behavior here for _TZ3000_yf8iuzil TS0002 ... Any help ?