dresden-elektronik/deconz-rest-plugin #6446: 2 gang Switch module (_TZ3000_zmy4lslw - TS0002)

Paalap, 2022-10-21:
## Device

- Product name: 2CH Zigbee Switch Module (Tuya Smart ZigBee 3.0 Switch Module)
- [AliExpress](https://de.aliexpress.com/item/1005004371232755.html)
- Manufacturer: _TZ3000_zmy4lslw 
- Model identifier: TS0002
- Device type : 2 gang/ch Switch

The device is connected reports the states properly but switching on/off is always applied to both channels (fun fact  the 4ch model from the same source above (with _TZ3000_ltt60asa / TS0004) is working fine).

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]

### Scenes
[image]
[image]
[image]

### On/Off
1st
[image]

2nd
[image]

### Simple Metering
[image]

Thank you in advance!

Paalap, 2022-10-24:
Is there a way to support the development?

There is/was a [similar issue on zigbee2mgtt](https://github.com/Koenkk/zigbee2mqtt/issues/14523#issue-1412091591) - just in case this is helpful

And would it it be possible to expose "power on behavior"?

Thanks

whit33r, 2022-10-25:
I have a 3 CH version of this switch, and that have same issue.

[image]

[image]

[image]

Paalap, 2022-11-16:
Would be happy to keep it active

Smanar, 2022-11-16:
This device need a "tuya unlock sequence" to work.
So not possible to add it for the moment.

Smanar, 2022-12-07:
Nope, need to add the feature on c++ side.
And as its not a priority ...

OndraM, 2022-12-07:
@Smanar I don't see `0xfffe` attribute in the deconz GUI in the basic cluster, is there any manual way how to show it?

Or maybe is it possible to send the write commands via some API? This way it would be possible to send the sequence.

Smanar, 2022-12-08:
>Or maybe is it possible to send the read commands via API?

Nope, sorry, the only way is using the GUI or the c++ code.

>is there any manual way how to show it?

If you are not using HA yes (I don't think this file was persistent, but have never tried), there is an xml file in your config folder called general.xml

I dont see the manufacture number on the capture, it's visible on the "node information" but probably can miss here

```
		<attribute-set id="0xFF00" description="Xiaomi specific" mfcode="0x115f">
			<attribute id="0xFF08" name="Unknown" type="u16" access="rw" required="o" mfcode="0x115f"></attribute>
			<attribute id="0xFF51" name="Unknown" type="ostring" access="rw" required="o" mfcode="0x115f"></attribute>
			<attribute id="0xFF52" name="Unknown" type="cstring" access="rw" required="o" mfcode="0x115f"></attribute>
			<attribute id="0xFF53" name="Unknown" type="ostring" access="rw" required="o" mfcode="0x115f"></attribute>
			<attribute id="0xFF54" name="Unknown" type="ostring" access="rw" required="o" mfcode="0x115f"></attribute>
			<attribute id="0xFFF0" name="Unknown" type="ostring" access="rw" required="o" mfcode="0x115f"></attribute>
		</attribute-set>
```

ranelazari, 2022-12-10:
Hi,
is there ny devalopment of thias issue?

OndraM, 2022-12-14:
@Smanar Ok, thanks. Any idea how to know whether the sequence succeeded? How can I recognize the device started to work normally after receiving the unlock sequence? Will this "just happen" or I need for example to switch some attribute to enable the "individual switch" mode?

Smanar, 2022-12-14:
Ha yes right.
No it's realy something magic, it work or not. return are classic.

You can use the GUI, choose a endpoint "01" or "02", select the cluster 0x0006 on/off, and in the cluster panel try the "on" or "off" command.
If it have action on 1 channel/part/gang it have worked, else it will switch the 2 channel/part/gang

OndraM, 2022-12-15:
Ok, so I spend like two hours on this, and I'm pretty confident it cannot be done using UI. 

- I edited general.xml, added the `0xfffe` attribute to the basic cluster. I removed all other attributes in the cluster, leaving just those required for tuya magic spell, in the right order.
- Then I copied the file into deconz docker image on startup, to override the default one.
- I opened the Basic cluster of the device in the UI.
- I tried using the "Read" button above all attributes (because it then reads all attributes in the cluster) but with no luck. The logs shows that deconz is sending request to some other attributes right after it goes through the visible, so maybe it breakes the signature? Who knows.
- I tried going through those six attributes and pressing read individually in their respective windows, with no luck.
- I even write script ("macro") which does the same automatically. Even though it is able to cycle through all the attributes in like ~two seconds, there is still no change in behavior of the device.

So either I'm missing something (I don't know when should this sequence be played? Right when the device is pairing? Or anytime later?). Or the two seconds are just to slow and it don't get recognized by the device and don't unlock it.

Also, I really thing this should be prioritized in the deconz, as there is already ~80 devices (according to z2m source code) by various manufacturers (tuya but also lidl, moes, zemismart...) using the magic packet, which are broken because of this.

OndraM, 2022-12-15:
Also, I found [this reference](https://community.smartthings.com/t/four-button-tuya-zigbee-device-dth-dth-link-in-post-7-same-device-from-multiple-brands-zemismart-yagusmart-losonho-dyhf-etc/206149/120):

> It is important to send the readAttribute commands for the Basic and Power clusters at once. Sending individual readAttribute commands for each attribute didn’t work.

So yes, it seems its not possible to achieve this in deconz UI, because it sends the commands individually.

ThiemeNL, 2022-12-25:
I have the same issue with [_TZ3000_bvrlqyj7](https://forum.phoscon.de/t/tuya-tz3000-bvrlqyj7-switch/2427) sending an ON command is turning on both channels.

GitMouth, 2022-12-28:
Hi, any suggestion to make them works ? ....
_TZ3000_yf8iuzil  TS0002
Same behavior....

Linqman, 2023-01-09:
I have the same issue with TS0012 _TZ3000_jl7qyupf, Switch Module 2 Channel No Neutral

OndraM, 2023-01-09:
> Hi, any suggestion to make them works ? ....

I solved it by migrating to zigbe2mqtt. According to information Smanar gave (that it won't be worked on this anytime soon) this is the only solution. I spend many hours trying to get this work with deConz, but finally I gave up and switched to zigbe2mqtt. So far, I didn't have a single issue, everything works fine and I definitely won't regret the extra time the migration took.

Paalap, 2023-01-09:
Same here, running 2 ZigBee networks in parallel. Not what I really like but it is ok for now.

micsey, 2023-01-13:
I have the same issue with 4 gang Switch module ( _TZ3000_ltt60asa - TS0004). In zigbee2mqtt i can find this device as supported.

GitMouth, 2023-01-13:
thx for your experience. I'm working with Deconz, and don't want to reappair and make my automations with mqtt... I'll selling them...

ThiemeNL, 2023-02-11:
Please keep this issue open

BabaIsYou, 2023-02-23:
Hi folks, 

@Smanar did a great job in doing this Tuya Magic sequence to be integrated in a [DDF](https://forum.phoscon.de/t/including-standard-tuya-magic-spell-sequence-during-joining/2916/15). That's perhaps juste a workaround waiting for something more optimized but can try to add the partin the referenced issue in DDF.

Smanar, 2023-02-23:
Probably something like

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_zmy4lslw",
  "modelid": "TS0002",
  "product": "Tuya 2 output module",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        },
        {
          "name": "config/interfacemode",
          "refresh.interval": 86400,
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": [
              "0x0004",
              "0x0000",
              "0x0001",
              "0x0005",
              "0x0007",
              "0xfffe"
            ]
          },
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0xfffe",
            "eval": "Item.val = Attr.val"
          },
          "public": false,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

closed 2023-03-25 by github-actions[bot]

BlauerPulli, 2023-03-27:
hey, just leave a comment to re-open this again. have the same device and it also not working for me. i will test @Smanar s suggestion soon.

ThiemeNL, 2023-04-01:
I tested @Smanar suggestion, but it's not working for me.

Smanar, 2023-04-02:
Uou have re-included the device ? (else need to wait 24h)

commonslibs, 2023-05-04:
Same problem here.

Smanar, 2023-05-05:
You have used the new DDF and even after a re-inclusion both channels are linked ?

ThiemeNL, 2023-05-05:
Yes, I tried the new DDF and re-inclusion. The behavior is now different. Looks like one channel is still connected to both outputs and but the other one channel is not working anymore. Reading the channel status is working fine.

commonslibs, 2023-05-05:
How can I put that DDF on my system? Any links that explain it? 

By the way, I understand that the "issue" should have an "open" state and not like the one that has to be "closed", right?

ThiemeNL, 2023-05-05:
Maybe this link is helpful: https://dresden-elektronik.github.io/deconz-dev-doc/modules/ddf/

For me its a while ago. I've upload the DDF file to `/home/{user}/.local/share/dresden-elektronik/deCONZ/devices`

The ticket was automatically closed due to inactivity.

commonslibs, 2023-05-06:
Thank you all very much and especially @Smanar and @ThiemeNL

The problem is already **solved**.

I have uploaded the DDF file that @Smanar had posted some answers before and it has worked perfectly for me.

Regarding the file that @Smanar put, I have modified very few things in the initial part of it, especially to put the vendor and some other minor change, which I understand has not been what has solved the problem, but I liked leaving the DDF more clear.

Once I have created the DDF file with the extension ".json" I have uploaded it to the folder "..."\devices\WETEN\TZ3000_zmy4lslw.json" and I have restarted the deconz docker container.

Once the container has restarted I have gone to home assistant and now I have been able to turn the bulbs on and off individually without problems.

I paste the screenshot of the DDF in deconz:
[image]

I attach the DDF file with the small changes that I have made.

[TZ3000_zmy4lslw.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/11411900/TZ3000_zmy4lslw.zip)

As said, thank you very much.

Smanar, 2023-05-06:
>By the way, is it possible that in future versions of "deconz" this device will be integrated without problems without having to carry out this "workaround" that consists of uploading the DDF file to "deconz"?

Yep, ofc https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6953

But I haven't used your modification because tuya is just clone of clones of clones, so better to use generic DDF.
Have used too a simplified config for the tuya unbclock command, so you can have an unused field now on your side ("config/tuya_unlock" instead of "config/interfacemode")

ThiemeNL, 2023-05-07:
Unfortunately your solution is not working on the`_TZ3000_bvrlqyj7` 
The device has a `Tuya special light mode` which is set to `Mode 1` [image] 
I think this is preventing the solution to work, since the 2 channel mode is set on the devices it selves. 
Any advise how to change the light mode?

BabaIsYou, 2023-05-07:
Did you try to change it into deconz GUI ?

ThiemeNL, 2023-05-07:
Yes, I tried, but it's a read only field.

Smanar, 2023-05-07:
Nothing special for your devices, all of them are clone and use the same code on Z2M

```
        fingerprint: [{modelID: 'TS0002', manufacturerName: '_TZ3000_01gpyda5'}, {modelID: 'TS0002', manufacturerName: '_TZ3000_bvrlqyj7'},
            {modelID: 'TS0002', manufacturerName: '_TZ3000_7ed9cqgi'}, {modelID: 'TS0002', manufacturerName: '_TZ3000_zmy4lslw'}],
```

Can you share the used DDF ?

ThiemeNL, 2023-05-07:
Currently I have loaded this file: https://gist.githubusercontent.com/ThiemeNL/ce8f0db07da2c4d78b99ad7ab576ccfe/raw/990585a12a6ed5fe6519a639db18df2656287a3f/TZ3000_bvrlqyj7.json
I change the vendor to something custom, so I could see that my changes where loaded.

I also tried your PR version and only changed the `manufacturername` to `_TZ3000_bvrlqyj7` but it didn't help
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6953/commits/8abab56069e65076cbdd157e65f4e8ddb3d7b824#diff-3051f341cfb05084eed8f8c89b841ebc3ed4430fab03b4da6047cd1fc781f650

Smanar, 2023-05-07:
The model id is "TS0002" too ?

ThiemeNL, 2023-05-07:
Yes, I deleted the lights, rebooted my Pi and then added the lights again: 
[image]

Mimiix, 2023-05-07:
Please open a seperate request for this, as this is not the same device.

sciorty, 2024-02-17:
I have the same problem with this device _TZ3000_nuenzetq:

[image]

[image]

Smanar, 2024-02-17:
You mean not reconised ?
You can try with adding the modelID+Manufacture name on the DDF.

sciorty, 2024-02-18:
> You mean not reconised ? You can try with adding the modelID+Manufacture name on the DDF.

I mean that when i try to turn one of the switches on, they both turn on.
If the solution would still be to add it into the ddf, which ddf is it?

sciorty, 2024-02-18:
It’s Ts0002 _TZ3000_nuenzetq with 2 gangs.
So i just have to add this json file where? Even if it’s named differently? Or do i have to change/add something on the inside?

Smanar, 2024-02-18:
As you want, depend of OS.
You can just add your model

```
  "manufacturername": ["_TZ3000_zmy4lslw", "_TZ3000_nuenzetq"],
  "modelid": ["TS0002", "TS0002"],
```

On the actual DDF.
Else  (if you have docker for exemple) you need to create a new DDF in a different folder.

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

sciorty, 2024-02-19:
Sorry that's my lack but I miss some informations. This is what I see if I try to edit the DDF through deCONZ
[image]
[image]

Should I modify it? Or should I create a file here (Home assistant supervised)?
[image]
How the file should be created? Just a copy/paste of [this](https://github.com/Smanar/deconz-rest-plugin/blob/581120bc632b4b35068f3edba37c8b08b4d7623f/devices/tuya/_TZ3000_zmy4lslw_2channel_module.json) ?

sciorty, 2024-02-19:
OK now it's working. I created the ddf on the path that I mentioned and called it _TZ3000_nuenzetq_2channel_module.json
Took [this ddf](https://github.com/Smanar/deconz-rest-plugin/blob/581120bc632b4b35068f3edba37c8b08b4d7623f/devices/tuya/_TZ3000_zmy4lslw_2channel_module.json) that apparently is the same device but of a different vendor (right?) and just changed the "manufacturername" field with mine.
On deCONZ I went to the same screenshot I posted and gone to File->Open and selected this newly created file so that it is loaded as a ddf for the device. Then did a Hot reload and it's working!

Is this the processo on how to manually "set" the ddf of the devices when I have some problems? How to simply modify the existing ones in case I have problems? What's the difference in the file's folder locations? How do I know that this device will be using the ddf I loaded? Eg: future releases will contain this .json and my "manually" created will not be used anymore and I can delete it?

Smanar, 2024-02-20:
>Is this the processo on how to manually "set" the ddf of the devices when I have some problems?

Depend of OS, HA supervised don't use docker ?
On normal OS deconz will use 2 "devices" folder, 1 is for user, untouched (so you will need to remove the file yourself), the second one is reset a every deconz update, on docker it's another story.
Make a right clic, then "edit DDF", on the editor titlebar you will see the DDf file used.

Have added your device in list  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6446