dresden-elektronik/deconz-rest-plugin #6449: The file /deCONZ.tar.gz couldn't be found (https://xxxx:xxx/deCONZ.tar.gz)

snhnic, 2022-10-21:
I have synology DSM 7.x with docker deconz conbee 2.  Everytime I do an update dsm or image conbee II the data seems to be old and I have to put everything form the ast months back in the conbee II system. I can make an backup form within the app but can not doanload it. So I search and fount issue 3664 and thought to ask there. But  @Mimiix telle me tp make a new issue. I truly don't like it to get to learn how to make isues and stuff. I got a life you know. But okay. @Smanar I can not find error in log file docker. Colud it be that data is stored in the stick? If so I could try to put in in my windows machine to create a backup?  

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3664#issuecomment-1284177465
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3664#issuecomment-1285766265

closed 2022-10-21 by close-issue-app[bot]

snhnic, 2022-10-21:
Why is this? Thanks @Mimiix for breaking my issue

Mimiix, 2022-10-21:
> Why is this? Thanks @Mimiix for breaking my issue

Ehm? 

The reason is provided in the post. We like to help people, but we have a life too you know? So we want to make it efficient as possible for everyone involved .

Mimiix, 2022-10-21:
Please use the forums for your question as this is not a bug probably.

snhnic, 2022-10-21:
> Reply: dresden-elektronik/deconz-rest-plugin
> deCONZ.tar.gz couldn't be found (https://xxxx:xxx/deCONZ.tar.gz) (Issue
> #6449)

I believe it is. But whatever your not helping.

Mimiix, 2022-10-21:
Hi,

We are all happy to help, but we need users to help us. That's why we do it this way. 

A bug needs to comply with #5113 , otherwise, help is at the forums :) So really, we want to help.

Kind regards,

snhnic, 2022-10-21:
If that was true, you would have left the issue open. But I guess your
eager to use your moderator righrs.
Many thanks for not helping at all.

Date: 21 October 2022 at 17:28:34

Mimiix, 2022-10-21:
Thank you for helping too :)