dresden-elektronik/deconz-rest-plugin #6454: Change brightness and colour white light is off

ebaauw, 2022-10-23:
## Device

- Product name: All lights
- Manufacturer: All manufacturers
- Model identifier: All models
- Device type : 
  - Light

Reading the latest ZCL spec (from the Matter site), attribute 0x000F in the _Level Control_ cluster (listed as "IKEA specific" in `general.xml`) is defined as _Options_, a bitmap with (commissioning-time) light configuration options.  The _Color Control_ cluster defines the same attribute.  Currently, _Options_ only defines bit 0: _Execute If Off_.  Looks like when this is set, the light should accept commands to change the level of colour while off.

This looks like granting a long-time wish by many users, being able to change brightness or colour while the light is off.  

Some quick test on a Hue light shows that, when _Execute If Off_ is set, it updates _Current Level_ on _Move_.  Is doesn't when _Execute If Off_ is cleared (which, unfortunately, is the default).

Unfortunately, the deCONZ REST API won't send any commands to a light when (the API thinks) it's off.  Instead, it issues an "api error 201: parameter, bri, is not modifiable. Device is set to off."  Last I checked, the Hue API issues a similar warning, but still sends the command.

I'll add the _Options_ attribute to `general.xml` (see screenshot below), but the API needs to be updated to handle this.  Not sure if we want to expose the _Execute If Off_ setting under the newly introduced (PR ) light `config`, or just force it to be set.  Of course, the API should send the commands when _Execute If Off_ is set, and probably should forego the warning.  For simplicity, we might simply skip the test whether the light is on altogether, having no sympathy for heretic lights that don't support this setting, and forego the warning.

Can I have a vote (by adding a thumbs up to the comments below) what we would prefer?
- Exposing the settings (one for brightness and one for colour) under `config vs simply forcing them to be set?
- Always skip the check if light is on altogether vs checking only of _Execute If Off_ hasn't been set?

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control
[image]

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

ebaauw, 2022-10-23:
Please give this comment the tumbs up if you want to expose _Execute If Off_ under `config`.

ebaauw, 2022-10-23:
Please give this comment a thumbs up, if you want to force _Execute If Off_, instead of exposing it under `config`.

ebaauw, 2022-10-23:
Please give this comment a thumbs up, if you want to skip the check if the light is on altogether.

ebaauw, 2022-10-23:
Please give this comment a thumbs up, if you want to keep the check if the light is on, when _Execute If Off_ isn't set (or supported by the light).

Smanar, 2022-10-23:
Honneslty I prefer 

> if you want to force Execute If Off, instead of exposing it under config.

But the information is usefull, so need  to be visible in config, so if the field exsit why don't use it.

LeoeLeoeL, 2022-10-23:
This feature would be useful with ADAPTING LIGHT in HA.

ebaauw, 2022-10-23:
Thanks @Smanar, I updated the description of the choices.  We'll still set it on pairing, even when exposed.  Looks like the setting is persisted across power cylces of the light.

wvuyk, 2022-10-23:
Several users of my plugin asked for such functionality. It is hard to explain you cannot set values before sending the actual on or bri commands

LeoeLeoeL, 2022-10-26:
Any possibility to set lamps (by adapting lighting) at minimum brightness when you switch them off?

ebaauw, 2022-10-26:
> Any possibility to set lamps (by adapting lighting) at minimum brightness when you switch them off?

Not sure what you mean by adapting lightning.

Setting lights to minimum brightness while switching them off has been supported for a long time.  Just put `{"bri": 1, "on": false}` to the light `state`.  The issue is probably that the light turns on at full brightness.  This is controlled by the _On Level_ attribute, which isn’t yet exposed by the API, see #149.  I’ll be including it in my PR, now that we add `config` to lights.  Not all lights support this, though.  Amongst those are Hue lights, but they behave like _On Level_ is 255 (meaning previous level).

closed 2022-12-18 by github-actions[bot]