dresden-elektronik/deconz-rest-plugin #6459: FDTEK Power Plug NAS-WR01B

sven2c, 2022-10-23:
## FDTEK Power Plug NAS-WR01B

- Product name: FDTEK Power Plug
- Manufacturer: FDTEK
- Model identifier: TS011F
- Device type : 
  - Light: smart plug (ON/OFF supported, or works with Phoscon and Node Red)
  - Sensor: power measurement NOT SUPPORTED / NOT WORKING

[image]
[image]

  Screenshots help to identify the device and its capabilities. Please refer to:
  https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support
  for examples of the required screenshots.

  Required screenshots:
  - Endpoints and clusters of the node

Node Info
[image]

Cluster Info
[image]

Looking for Data  marked with red arrow
0x0505  RMS Voltage
0x0508  RMS Current
0x050B  Active Power   
These Data not available for example in Node Red

Smanar, 2022-10-23:
Hello, can you try to add your model at an existing DDF to test ?
For exemple this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_typdpbpg_smart_plug_eu.json

Mimiix, 2022-10-23:
@sven2c can you please add all requested screenshots ?

sven2c, 2022-10-25:
Hello,

 **@Smanar:** I'm not familiar how to add my model to the DDF? I do not know what DDF means. I'm sorry I'm a HW guy.

I update to latest Deconz version 
Version:    v2.19.1 
Firmware: 0x26780700
Anyhow, I can not see the Power Consumption

Now I got a bit different information
Node Info
[image]

Cluster Info
[image]

Smanar, 2022-10-25:
Here you have an explanation about DDF https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

It will depend of you OS, but you just need to acces too folder that contain the DDF and edit one with a text editor.
Or can just create a new one using an old one.

sven2c, 2022-10-27:
@Smanar 
Thank you for the instruction. I updated the DDF File - but I just see the smart plug as on/off switch at phoscon and node red. The sensor information is not accessable

[image]

I can see the copied code in here
[image]

Smanar, 2022-10-28:
The DDF seem fine, have yiu tried "hot reload" in the editor menu ?
Nothing more visible in phoscon/help/API information/sensors ?

sven2c, 2022-10-31:
I have tried "hot reload" and "restart". No Senor Information available.

No information of this plug in phoscon/help/API information/sensors
Th plug is just shown in /lights only with ON/OFF data.

Please find my deconz version
[image]

Mimiix, 2022-10-31:
@sven2c you should find them under /sensors in the api.

Mimiix, 2022-10-31:
Did you check the api?

sven2c, 2022-10-31:
I have checked the Rest API.
I just can see the value showing in  Deconz
- here without load, so the value is 0
[image]

Smanar, 2022-10-31:
Can you make a "hot reload" but with logs enabled in deconz ? (In help/ debug view with flag "DDF" and "info")

sven2c, 2022-11-01:
Please find the Debug view
[image]

Smanar, 2022-11-01:
Ok so the DDF is reconized and used.
I realy have no idee to explain why the sensors don't appear on phoscon / help / API information / sensors.

Nothing usefull if you use too "info_l2" ?

sven2c, 2022-11-04:
I don't know. I delete the plug  and reload, ... Nothing helps
I just can read the Power/ Voltage / Current data only in the deCONZ gui -- fustrating

I just found this in Info_L2
-------------------------------------------------------------
21:20:16:265 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('10', 'normal', 'a4:c1:38:03:79:13:48:36-01', 'On/Off plug-in unit 10', '65520,1', '1', 'TS011F', '_TZ3000_rdfh8cfs', '77', '{"attr/id":"10","attr/lastannounced":null,"attr/lastseen":"2022-11-04T20:19Z","attr/manufacturername":"_TZ3000_rdfh8cfs","attr/modelid":"TS011F","attr/name":"On/Off plug-in unit 10","attr/swversion":"77","attr/type":"On/Off plug-in unit","attr/uniqueid":"a4:c1:38:03:79:13:48:36-01","state/alert":null,"state/on":true,"state/reachable":true}')

Smanar, 2022-11-05:
I have perhaps an idea ^^
Have you set deconz in permit join ? "add new light" or "add new sensor" when making the procedure ?
I think the new devices addition addition is blocked if it's not enabled.

sven2c, 2022-11-05:
Hello,
now it works but it was a strange way to get it running.

1) I delete the "switch/plug" code in the DDF and I only used the ZHAPower part of the code and then I triggered a Hot Reload
[image]

2) Next I paired the Sensor using Phoscon Tool
 It tells Sensor connected, but does not appear in the Sensor Overview. Anyhow now I'm able to see the sensor at API Info
[image]

3) Now I add the ON/OFF plug-in unit to the DDF and trigger a Hot Reload. The switch appears now as well in the Phoscon Tool
[image]

In the DDF .json File finally the Code is just switched. I do not really know why it works now

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_rdfh8cfs",
  "modelid": "TS011F",
  "vendor": "FDTEK",
  "product": "Smart plug EU",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/FDTEK_Power_Plug_001.json",
  "subdevices": [
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 360,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 360,
          "default": 0
        },
        {
          "name": "state/voltage",
          "refresh.interval": 360,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {`

sven2c, 2022-11-05:
@Smanar: Thank you for your strong support.

Smanar, 2022-11-06:
>Thank you for your strong support.

Seriously it's not my fault, your procedure is just magic. I think delete the device and make a full re-inclusion could work ...

I have found a PR with a clone of your device, I will ask him to add your model too

iisworks, 2022-11-06:
Can you remove the old ddf and try this DDF file (delete the device and pair it again)?
https://github.com/iisworks/deconz-rest-plugin-1/blob/patch-6/devices/neo/NAS-WR01B_TS011F.json

Smanar, 2022-11-06:
BTW I haven't asked what is your OS ? (to install the DDF)

sven2c, 2022-11-06:
My OS
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

sven2c, 2022-11-06:
I delete the DDF  File  in  ...
/home/pi/.local/share/dresden-elektronik/deCONZ/devices
... but if I delete and reconnect the plug always my old configuration is loaded. I do not know were this comes from?

Smanar, 2022-11-06:
>my old configuration is loaded

You mean your first DDF auto created ?
You haven't the path/file name in the editor title ? When making "edit DDF"

After a restart deconz scan all DDF and choose one according to manufacture name and model id. As it's a clone there is perhaps an other DDF with the same value

iisworks, 2022-11-06:
Make sure to restart deconz as well. 

Or from the device "Edit DDF", then Open the new DDF and choose "Hot reload":
[image]

sven2c, 2022-11-06:
After full restart now the new DDF is running -- but now I lost the sensor information again.
Reason is the unique ID  is with FF now instead of 01: 
    "uniqueid": "a4:c1:38:03:79:13:48:36-ff-0b04"

iisworks, 2022-11-06:
Do you see all expected sensor info in phoscon / help / API information / sensors?

sven2c, 2022-11-06:
Yes, I used /help/ APi in... but this seems this is  a virtual sensor or somtehing  - no data changing when I read from this adress 
Additionally I have now a strange behavior.
I delete the plugs and when I restart always the two nodes with power appear -- independent if the plugs powered or not.
They appear always after restart as well the plugs are not connect 

Ahhh it is a nightmare

sven2c, 2022-11-06:
[image]

The plugs are not supplied and anyhow Deconz remember the ID and create two Power39 and 40 -- I do not understand this

sven2c, 2022-11-06:
[image]

iisworks, 2022-11-06:
Not sure, I'm just a Deconz user as well... 
I guess you should see one device and pretty much the same clusters as on the Innr plugs. Can you remove these 2 Power nodes, wait a bit (restart deconz maybe)  and try to pair it again? 

You do see the correct ddf path being loaded in the debug log on startup?

sven2c, 2022-11-06:
I connected the Plug 
[image]

but all values are not Ok
[image]

iisworks, 2022-11-06:
So when you delete all these orphaned nodes and re-add the device it ends up the same way? Not just one Smart Plug device in deconz?

Smanar, 2022-11-06:
Yes the 0xFF endpoint is an issue, idk how it can happen.
But the DDF is fine for me, if you delete the device and make a new inclusion, still have the 0xFF ?

sven2c, 2022-11-06:
Finally, I did a full Gateway RESET.
I had to connect all my diveces again from the scratch.
The Power Plug I connect by searching for Sensors and it was recognized, bit not shown in the Deconz Sensor Overview.
Then I add in the DDF the ON/OFF Plug Unit and did a Hot reload.
I can see now the Switch in Phoscon Lights. 
In the Help/Rest API it was recognized as Power23.
This is not fully fine but it works again

Smanar, 2022-11-06:
>Finally, I did a full Gateway RESET.

Arf, sorry, you haven't backup ?

>This is not fully fine but it works again

Still have issue ?

sven2c, 2022-11-06:
Somehow it works now.
Two thinks are not nice but nor really important
1) Sensor Value not shown in Phoscon 
2) Different naming for Plug (ON/OFF Plug 08) and Sensor (Power 23)
Mainly I use NodeRed for controlling. This works now - Ok with me

Smanar, 2022-11-07:
>Sensor Value not shown in Phoscon

Yeah Phoscon don't support all sensor, but it's not the same project

>Different naming for Plug (ON/OFF Plug 08) and Sensor (Power 23)
Mainly I use NodeRed for controlling

This is a long story ^^, you can add a field in the DDF to set a priority, else deconz use as node name the last device loaded/created, generaly "on/off plug" is prioritary on "Power X".
So you can change the name in the API if you reconize the entry used by deconz, but it can choose another one at next restart, it's event based, and event order can change.

But I think you are using the device ID to reconize it in Nodered ?