dresden-elektronik/deconz-rest-plugin #6461: Since versions v2.19.0-beta and v2.19.1, several devices do no longer work properly

Rolf56, 2022-10-24:
## Describe the bug
Since versions v2.19.0-beta and v2.19.1, the following devices do no longer work properly:

- Hersteller: Philips, Produkt: Dimmschalter, Version: 6.1.1.28573
- Hersteller: Philips, Produkt: Dimmschalter, Version: 5.45.1.17846
- Hersteller: LUMI, Produkt: lumi.switch.n2aeu1, Version 0.0.0_0022

These are recognized by the system, but can no longer be switched locally.
The relay in the Lumi switch clicks and the blue lamp comes on, but the light command is not executed.
The LED on the Dimmschalter lights up red.

## Steps to reproduce the behavior
it works with version v2.18.2. After the installation of one of the newer version it doesn't work anymore.
Deconz is conected to FHEM and controld by it.

## Expected behavior

## Screenshots

## Environment
 - Host system: PC
 - Running method:  Windows
 - Firmware version: 26400500
 - deCONZ version: 2.19.01
 - Device: ConBee I
 - Do you use an USB extension cable:  no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

SVH-Powel, 2022-10-28:
I have a lot of these dimmers and I stay with version 2.18.2 for the moment. Better safe than sorry....

Can anyone confirm what @Rolf56 reports? Is this an issue with 2.19? (or conbee I or perhaps Rolf56 system only)

AndreHamburger, 2022-10-28:
Confirmed. After updating to 2.19 HUE Button still appears as a device / switch, but without function. Button is blinking  2 times green, but no assigned actions will be performed. Deleting the Button is also not working properly. Seems cache is beating and device is back visible as a switch, after deleting it and rebooting the Rasperry PI. Homebridge is sometimes providing websocket errors. I my perception is, the Raspberry slows down and CPU usage increased, once several button clicks has been made. Other switches (Opodo) working properly and lights and scenes are fine too

le-clu, 2022-11-02:
Same issue. NONE of my devices seem to work.

Using RaspBee with Home Assistant.

Is there any known and good way to downgrade?

Mincka, 2022-11-02:
> Is there any known and good way to downgrade?

Restore a backup. If don't use the Google Backup Add-on, don't uncheck the backup option when you update an add-on. Good luck.

therealhalifax, 2022-11-02:
Same for me with the HUE SMART BUTTON. Also pairing dosnt work. Downgrade helped.

ebaauw, 2022-11-02:
My Hue smart button works without any issue in v2.19.1.

For the Hue smart button, please:
- Please double-check in the GUI that deCONZ receives button action notifications on the 0xFC01 cluster.  If not, double-check that the 0x0031 _Configuration_ attribute in the _Basic_ cluster is 0x000b;
[image]

- Double-check that it reports `manufacturername` of "Signify Netherlands B.V." (not: "Philips").  If not, read the _Basic_ cluster attributes, while pressing the button, ensuring that it's awake;
[image]

- Double-check that it uses the correct DDF, `philips_rom001_smart_button.json`.  The DDF should define four bindings.

If all these check out, and the API still doens't emit buttonevent notifications, please:
- List the API ZHASwitch `sensors` resource for the button;
```
{
  "config": {
    "battery": 100,
    "group": "10",
    "on": true,
    "reachable": true
  },
  "etag": "99282fc5781bd2fb7fc2812f30b69bc7",
  "lastannounced": "2022-11-02T17:02:23Z",
  "lastseen": "2022-11-02T17:15Z",
  "manufacturername": "Signify Netherlands B.V.",
  "mode": 1,
  "modelid": "ROM001",
  "name": "Living Room Button",
  "state": {
    "buttonevent": 1002,
    "eventduration": 1,
    "lastupdated": "2022-11-02T17:07:10.297"
  },
  "swversion": "2.47.8_h2F968624",
  "type": "ZHASwitch",
  "uniqueid": "00:17:88:01:06:07:b6:c4-01-fc00"
}
```
- Capture the APS and APS_L2 log in deCONZ, while pressing the button:
```
Nov 02 18:07:10 pi2 deCONZ[25970]: 18:07:10:111 APS-DATA.indication srcAddr: 0xb302, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xFC00, lqi: 255, rssi: -58
Nov 02 18:07:10 pi2 deCONZ[25970]: 18:07:10:112 	asdu: 1d0b104a000100003000210000
Nov 02 18:07:10 pi2 deCONZ[25970]: 18:07:10:143 [INFO] - No button map for: ROM001, unicast to: 0x0000, endpoint: 0x01, cluster: 0xFC00, command: 0x00, payload: 0100003000210000, zclSeq: 74
Nov 02 18:07:10 pi2 deCONZ[25970]: 18:07:10:144 Websocket 127.0.0.1:47430 send message: {"attr":{"lastseen":"2022-11-02T17:07Z"},"e":"changed","id":"229","r":"sensors","t":"event","uniqueid":"00:17:88:01:06:07:b6:c4-01-fc00"} (ret = 2044751104)
Nov 02 18:07:10 pi2 deCONZ[25970]: 18:07:10:150 Websocket 127.0.0.1:47430 send message: {"e":"changed","id":"229","r":"sensors","state":{"buttonevent":1000,"eventduration":0,"lastupdated":"2022-11-02T17:07:10.126"},"t":"event","uniqueid":"00:17:88:01:06:07:b6:c4-01-fc00"} (ret = 2044751104)
```

Pirol62, 2022-11-03:
Hi, I'm using these switches in Home Assistant.
Opening the DDF in deConz providing these Info:
[image]
IT's definately another json
That's the preview:
[image]
What shall I do now?

ebaauw, 2022-11-03:
That’s an old-model Hue dimmer switch, not a Hue smart button.  Afaik there were no changes for this device between 2.18 and 2.19.  I think you might be running into a generic issue, not one related to a particular device.  There are known issues with send queues filling up in 2.19, causing outgoing commands to be dropped by the gateway.  In this case, the ZHASwitch state would be updated alright (and the corresponding web socket notifications would be sent), but any actions by related rules would be lost.

This would affect all switches that behave like sensors, reporting button events to the gateway and requiring some automation (rules) to control lights.  It would not affect switches that behave like controllers, sending commands directly to bound groups or lights, without going through the gateway.

Pirol62, 2022-11-03:
By the way: I'm working with deConz not ZHA. Maybe that's a difference and the hue dimmer switch doesn't provide events anymore compared to my Aquara Switches, which are working as usual.

therealhalifax, 2022-11-03:
I have also deconz, not zha and just by the update some devices not working anymore and not pairable anymore. After downgrade it works again. So no change in anything excapt the deconz addon.

AndreHamburger, 2022-11-05:
Just FYI. I updated just Node.js to the latest version for Rasperry Pi to 18.12.1, as while i updated Deconz to 2.19.1 at the same time Node was updaten to 18.2.0. But no effect for the HUE Button issue. Still not working, same websocket errors occurred in Homebridge

IgnacioHR, 2022-11-05:
I’ve the same issue with ROM001 (Smart Switch) devices. I documented here
https://forum.phoscon.de/t/hue-smart-switch-rom001-no-longer-work-in-version-2-19-01/2706
but have no solution. I downgraded to previous version

AndreHamburger, 2022-11-05:
Sorry, being a newbi….can someone mention, how to downgrade to 2.18.2 on Rasperry, to make hue button working again?

IgnacioHR, 2022-11-05:
@AndreHamburger I downloaded the .deb file and executed apt install ….deb sorry for lack of formatting cause I’m on my mobile phone

Friedi1970, 2022-11-05:
> @AndreHamburger I downloaded the .deb file and executed apt install ….deb sorry for lack of formatting cause I’m on my mobile phone

Which .deb? Apt install? Could you describe more precise pls.?

IgnacioHR, 2022-11-05:
@Friedi1970 @AndreHamburger 

Head here:

http://deconz.dresden-elektronik.de/raspbian/beta/

And select the version you want to downgrade to.

Copy that file in the RPI you want to install it and run `sudo apt install ....deb` where ... is the file you just downloaded. The process will ask you if you are sure you want to downgrade.

Note, do it at your own risk! For example, if the new version updates the database schema, the old version might not work.  It is always a good practice to do a backup copy of the `.local/share/dresden-elektronik/deCONZ` directory (I'm not an expert so there might be other directories with important information). In my case, going back from 2.19.1 to 2.17.1 caused no issues.

Friedi1970, 2022-11-06:
> @Friedi1970 @AndreHamburger
> 
> Head here: ...

Thanks a lot!!!

Friedi1970, 2022-11-06:
> @Friedi1970 @AndreHamburger
> 
> Head here:
> 
> http://deconz.dresden-elektronik.de/raspbian/beta/
> 
> And select the version you want to downgrade to.
> 
> Copy that file in the RPI you want to install it and run `sudo apt install ....deb` where ... is the file 

When I try to run sudo apt the command apt is unknown ...

IgnacioHR, 2022-11-06:
just `apt` works for me, try `apt-get` instead

MSchmitt67, 2022-11-06:
> Sorry, being a newbi….can someone mention, how to downgrade to 2.18.2 on Rasperry, to make hue button working again?

i downgraded to 2.18.02  =

1.
sudo systemctl stop deconz **(Headless version)**, or
sudo systemctl stop deconz-gui **(GUI version)**

2.
wget -O deconz-2.18.02-qt5.deb https://deconz.dresden-elektronik.de/raspbian/stable/deconz-2.18.02-qt5.deb

3.
sudo dpkg -i deconz-2.18.02-qt5.deb

4.
sudo systemctl start deconz

Friedi1970, 2022-11-06:
I neither have the commands apt nor apt-get nor dpkg :(
I just get: 
sudo: dpkg: command not found

MSchmitt67, 2022-11-06:
wich system you are on ?

Friedi1970, 2022-11-06:
> wich system you are on ?

HA OS

MSchmitt67, 2022-11-06:
Raspberry Pi ?
Conbee 2 ?

Friedi1970, 2022-11-06:
yep

Mimiix, 2022-11-06:
Ha os doesn't have this options afaik.

Friedi1970, 2022-11-06:
> Ha os doesn't have this options afaik.

Unfortunately I guess you're right :-/

IgnacioHR, 2022-11-12:
I tested version 2.19.2 yesterday but the problem with the HUE Smart Switch ROM001 persist. I updated firmware of the HUE to match the one @ebaauw showed here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6461#issuecomment-1300977974 

I request to reopen this bug

IgnacioHR, 2022-11-12:
Downgrading to version 2.18.02 does NOT fix the issue. Downgrading to 2.17.01 fixes it

Tscherno, 2022-11-14:
> I tested version 2.19.2 yesterday but the problem with the HUE Smart Switch ROM001 persist. I updated firmware of the HUE to match the one @ebaauw showed here: [#6461 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6461#issuecomment-1300977974)
> 
> I request to reopen this bug

Same for me. All the Buttons stopped working after upgrading to 2.19.2. Reset / Readding not possible. After downgrade to 2.17.01 its working again. Had to reconfigure the actions of all buttons afterwards.

Wiggerl26, 2022-11-14:
I had exact same problem with my HUE Smart Switch, but it started working again after firmware update to 2.47.8_h2F968624. You can find the firmware here: https://github.com/dresden-elektronik/deconz-ota-plugin#hue-firmware.

Tscherno, 2022-11-15:
Does that work if they are not connected anymore or not connected at all?!?

Wiggerl26, 2022-11-16:
Let me be a bit more precise: I downgraded deconz to 2.17 and deleted the smart button in phoscon app. Then I upgraded deconz to 2.19.2 and paired the button again using phoscon app. After that my button started working in openhab again.

Sergeus, 2022-11-20:
I have also had to downgrade back to 2.17, because all of my Hue Smart Buttons stopped working on 2.19.1. Downgrading has restored their functionality.

I tried the steps mentioned by ebaauw, but several behaviors diverged from those instructions. The Configuration value on the Basic cluster for several of my siwtches is 0x0000. Reading the Basic cluster again did not change this. Rereading the Basic cluster caused the Manufacturer to change to Philips, not Signify. The DDF json for the buttons looks much more like the one posted by Pirol62 (though not exactly the same).
[image]

Several of my Hue Buttons are on older firmwares (none are on 2.47.8, which is what's listed on that firmware list page - the highest is 2.30, lowest seems to be 2.21), but trying to do any firmware updates via deConz did not work for me. Is there a reliance on a specific firmware version now for some updated information?

At the high level, even if we can fix it (and right now I can't) this seems like a significant breaking change for existing users. Is there anything that future versions can do for some backwards compatibility? Or a migration path to what seems to be a new way for deConz to communicate with these Hue devices?

Smanar, 2022-11-20:
Your DDf is not used, look at the status "Draft" if you want to use it set it to "Gold", and with the DDF the manufacture name need to be "Philips"

ebaauw, 2022-11-20:
>  Reading the Basic cluster again did not change this.

Make sure to press the button to wake the device.

> Is there a reliance on a specific firmware version now for some updated information?

I suspect Signify might have changed the _Manufacturer Name_ with the latest firmware version.  I only have one button, which is on the latest firmware, so I cannot verify this myself.

> Is there anything that future versions can do for some backwards compatibility?

Change the DDF to accept both "Signify Netherlands B.V."/"ROM001" and "Philips"/"ROM001" as `manufacturername`/`modelid`:
```
{
    "schema": "devcap1.schema.json",
    "manufacturername": [
        "$MF_SIGNIFY",
        "$MF_PHILIPS"
    ],
    "modelid": [
        "ROM001",
        "ROM001"
    ],
    "product": "Hue smart button",
    "status": "Gold",
    "sleeper": true,
    "subdevices": [
        ...
    ]
}
```

IgnacioHR, 2022-11-20:
Most of us (users) don’t understand how to modify DDF files. Maybe is a good idea to include an updated one with the next version?

ebaauw, 2022-11-20:
If someone can confirm that indeed the older firmware versions advertise "Philips" as _Manufacturer Name_.

IgnacioHR, 2022-11-20:
It is confirmed!

https://forum.phoscon.de/t/hue-smart-switch-rom001-no-longer-work-in-version-2-19-01/2706

ebaauw, 2022-11-20:
No, it hasn’t. I want to see a GUI screenshot of the _Basic_ cluster.  There’s still some crooked logic in the API plugin that sets manufacturername based on the MAC address prefix, before the _Basic_ cluster has been read.

Sergeus, 2022-11-21:
Here's confirmation:
[image]
That button is running 2.25.0_r30044 (according to Phoscon). The above screenshot was taken after pressing "Read" while holding down the button. This also shows that Configuration remains at 0x0000.

Before pressing "Read" most of those values were blank or invalid, like so:
[image]

ebaauw, 2022-11-21:
@Sergeus, thanks!

Damn Signify.  It would explain why I originally created the DDF for "Philips" in #6164, before changing it to "Signify Netherlands B.V." in #6347.  I thought I made an error, but apparently the manufacturer name changed between firmware versions.

@Rolf56 Why are you posting screenshots of the original model Hue dimmer switch?

ebaauw, 2022-11-21:
This PR should math both manufacturer names.  I sure hope we won't see too many other devices with this "feature".

Rolf56, 2022-11-21:
@ebaauw
because I opend this issue initualy and still got no answer for a solution!

ebaauw, 2022-11-21:
Damn, indeed.  I fear they hijacked your issue for the Hue smart button.  You're running older firmwares on your Hue dimmer switches, but the old model still advertises as "Philips" with the latest fiirmware, so your issue is unrelated to the issue with the smart button.

Still, the same questions as in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6461#issuecomment-1300977974, except for the _Manufacturer Name_ part.

ebaauw, 2022-11-21:
It’s not the predating, but the change that bothers me.  The Hue bridge exposes the new `manufacturername` for all Hue devices, but they didn’t change the _Manufacturer Name_ on the firmware of the old devices.  At least not for the old models of the Hue dimmer switch and Hue motion sensor.  And some of the newer devices (like latest Hue Bloom) still use “Philips” with the latest firmware.  It’s a jungle out there…

I wish the DDF would simply match both “Signify Netherlands B.V.” and “Philips” on either value.

Rolf56, 2022-11-23:
@https://github.com/ebaauw 
I have compared the DDF files of the two versions. There is no difference. I don't think that the problem arises because of the DDF file.
Therefore, something must have been changed in the deConz code that causes this different behavior.
Can you give the issue to SW development to check this?

ebaauw, 2022-11-23:
There is no difference in handling the Hue dimmer switch between recent versions of deCONZ.  We’re seeing a good deal of issues where deCONZ becomes unresponsive because of queues on the coordinator device filling up, and the API plugin (re-)queuing requests too frequently.  That’s what the PR that closed this issue addresses (in part).

Still we need to assess whether your issue is caused by the communication between the Hue dimmer switch and deCONZ, or between deCONZ and the lights.  Hence my questions above.  Also, do you see any errors in the deCONZ log?

Rolf56, 2022-11-23:
Here the messages from APS and APS_L2. ERROR and ERROR_L2 are empty

APS
====
16:15:24:274 APS-DATA.indication srcAddr: 0x9e63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x000C, lqi: 127, rssi: -75
16:15:24:294 APS-DATA.indication srcAddr: 0x9e63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x000C, lqi: 127, rssi: -75
16:15:24:336 APS-DATA.indication srcAddr: 0x9e63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x000C, lqi: 199, rssi: -66
16:15:24:416 APS-DATA.confirm id: 7, status: 0x00 SUCCESS
16:15:24:416 APS-DATA.confirm request id: 7 -> confirmed, timeout 625297673
16:15:24:448 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 98
16:15:24:448 APS-DATA.indication request id: 7 -> finished
16:15:24:448 APS-DATA.request id: 7 erase from queue
16:15:24:448 APS-DATA.request id: 28, addrmode: 0x03, addr: 0x00212effff01ba30, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 5 len: 2 tx.options 0x00
16:15:24:512 APS-DATA.confirm id: 22, status: 0x00 SUCCESS
16:15:24:513 APS-DATA.confirm request id: 22 -> erase from queue
16:15:24:523 APS-DATA.confirm id: 28, status: 0x00 SUCCESS
16:15:24:523 APS-DATA.confirm request id: 28 -> confirmed, timeout 625340459
16:15:24:539 reject aps request to enddevice node queue is full (4)
16:15:24:554 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 0, rssi: 98
16:15:24:554 APS-DATA.request id: 22 erase from queue
16:15:24:554 APS-DATA.request id: 31, addrmode: 0x02, addr: 0x0000, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 5 len: 18 tx.options 0x00
16:15:24:586 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 98
16:15:24:586 APS-DATA.indication request id: 28 -> finished
16:15:24:586 APS-DATA.request id: 28 erase from queue
16:15:24:586 APS-DATA.request id: 33, addrmode: 0x03, addr: 0x00212effff01ba30, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 5 len: 2 tx.options 0x00
16:15:24:656 APS-DATA.confirm id: 31, status: 0x00 SUCCESS
16:15:24:656 APS-DATA.confirm request id: 31 -> erase from queue
16:15:24:672 APS-DATA.confirm id: 33, status: 0x00 SUCCESS
16:15:24:672 APS-DATA.confirm request id: 33 -> confirmed, timeout 625340602
16:15:24:695 aps request id: 31 finished, erase from queue
16:15:24:695 APS-DATA.request id: 34, addrmode: 0x02, addr: 0x058f, profile: 0x0000, cluster: 0x0001, ep: 0x00 -> 0x00 queue: 5 len: 5 tx.options 0x00
16:15:24:699 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 0, rssi: 98
16:15:24:726 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 98
16:15:24:726 APS-DATA.indication request id: 33 -> finished
16:15:24:726 APS-DATA.request id: 33 erase from queue
16:15:24:726 APS-DATA.request id: 37, addrmode: 0x03, addr: 0x00212effff01ba30, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 5 len: 2 tx.options 0x00
16:15:24:785 APS-DATA.confirm id: 34, status: 0x00 SUCCESS
16:15:24:801 APS-DATA.confirm request id: 34 -> confirmed, timeout 625340705
16:15:24:805 APS-DATA.confirm id: 37, status: 0x00 SUCCESS
16:15:24:805 APS-DATA.confirm request id: 37 -> confirmed, timeout 625340746
16:15:24:836 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 98
16:15:24:836 APS-DATA.indication request id: 37 -> finished
16:15:24:836 APS-DATA.request id: 37 erase from queue
16:15:24:836 APS-DATA.request id: 39, addrmode: 0x03, addr: 0x00212effff01ba30, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 5 len: 2 tx.options 0x00
16:15:24:867 APS-DATA.indication srcAddr: 0x058f, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8001, lqi: 103, rssi: -78
16:15:24:867 APS-DATA.indication request id: 34 -> finished
16:15:24:867 APS-DATA.request id: 34 erase from queue
16:15:24:913 APS-DATA.confirm id: 39, status: 0x00 SUCCESS
16:15:24:929 APS-DATA.confirm request id: 39 -> confirmed, timeout 625340859
16:15:24:946 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 98
16:15:24:961 APS-DATA.indication request id: 39 -> finished
16:15:24:961 APS-DATA.request id: 39 erase from queue
16:15:24:961 APS-DATA.request id: 42, addrmode: 0x03, addr: 0x00212effff01ba30, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 4 len: 2 tx.options 0x00
16:15:25:258 APS-DATA.request id: 43, addrmode: 0x03, addr: 0x00158d0003f3e837, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 5 len: 5 tx.options 0x00

16:15:45:039 aps request id: 19 failed, erase from queue
16:16:20:134 APS-DATA.confirm id: 19, status: 0x00 SUCCESS
16:16:20:134 APS-DATA.confirm id: 19, status: 0x00, match: 0
16:16:20:165 APS-DATA.indication srcAddr: 0x9b14, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 39, rssi: -86
16:16:20:196 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 108, rssi: 0
16:16:20:196 APS-DATA.indication request id: 42 -> finished? not confirmed
16:16:20:231 APS-DATA.indication srcAddr: 0x9b14, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 39, rssi: -86
16:16:20:258 APS-DATA.indication srcAddr: 0x9b14, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0xFC00, lqi: 7, rssi: -91
[... cut ...]
16:17:29:758 Delay APS request id: 134 to 0x21BA, profile: 0x0000 cluster: 0x0021 node already has busy 2
16:17:29:820 Delay APS request id: 134 to 0x21BA, profile: 0x0000 cluster: 0x0021 node already has busy 2
16:17:29:883 Delay APS request id: 134 to 0x21BA, profile: 0x0000 cluster: 0x0021 node already has busy 2
16:17:29:945 Delay APS request id: 134 to 0x21BA, profile: 0x0000 cluster: 0x0021 node already has busy 2
16:17:30:008 Delay APS request id: 134 to 0x21BA, profile: 0x0000 cluster: 0x0021 node already has busy 2

Mimiix, 2022-12-02:
Re-opened as it seems to still affect users.

DoCheckor, 2022-12-02:
Hi guys! Updated HomeAssistant and the Phoscon firmware for my Conbee2 yesterday and now my Philips Dimmer Switches don't work anymore like they did for months. Better said, only the last one I rebind happens to work

Brian-lub, 2022-12-03:
i have lost the connection to all mine and i can't get them to reconnect, i have updated downgraded updated, and i have tried everything know. 
running HA with DeConz
Version 2.19.01 / 19.9.2022
Firmware 26720700

ebaauw, 2022-12-04:
> 16:17:29:289 Delay APS request id: 134 to 0x21BA, profile: 0x0000 cluster: 0x0021 node already has busy 2

That looks suspiciously like queus on the coordinator device filling up, due to issues in v2.19.0 to v2.19.2.  Hopefully, this won’t happen anymore in v2.19.3.

To clear the queus, you need to reset the coordinator device.  Easiest way to do this is to power cycle it.  For the ConBee: stop deCONZ, remove the stick, wait 5s, re-insert it, start deCONZ.  For the RaspBee: shutdown the Pi, remove the power adapter plug from the Pi, wait 5s, re-insert it and wait for the Pi to boot.

undulattice, 2022-12-08:
Just updated to 2.19.3.
Philips Hue Smart Button still isn't found. :-(

ebaauw, 2022-12-08:
That’s to be expected; PR hasn’t been merged.

tkintscher, 2022-12-09:
I have a similar issue, but not quite... for me the difference between working and non-working Hue dimmer switches is the firmware version.

Running deCONZ v2.19.3-beta.
I have three "old" Philips dimmer switches here (RWL021). All of them show the Manufacturer Name as "Philips".

- one has SW Build ID 5.45.1.16265
  - button presses show up on the 0xFC00 cluster in deCONZ
  - button presses show up on the WebSocket
  - rules are executed
  - all good :)

- two have SW Build ID 5.45.1.17846
  - the button presses are registered on the 0xFC00 cluster in deCONZ
  - button presses do *not* show up on the WebSocket
  - rules are *not* executed
  - they used to work before, now they don't ...

I don't understand why the firmware build on the switch would make this kind of difference, though.

Pirol62, 2022-12-10:
I already updated the conbee firmware to 26780700 but without effect.
Hue dimmer switches cannot be paired and the existing, already paired one, doesn't work with the origin switch modes

Pirol62, 2022-12-13:
Ok, I already updated deconz addon in HA to 2.19.3 and my Hue Dimmer Switch could be reconnected and works as usual.
So I deleted and reconnected the second Dimmer Switch and everything worked again as well.
For me, this part of the issue is solved.

ge8020, 2022-12-13:
Smart Button still not functional here with HA and deconz addon 6.17.1 (deconz 2.19.3). When i delete it and pair it new on 2.19.3 it ist recognized as a philips dimmer switch instead of smart button by phoscon. But there are still no events triggered in Home assistant...

Friedi1970, 2022-12-17:
> Smart Button still not functional here with HA and deconz addon 6.17.1 (deconz 2.19.3). 

Same here. I wonder whether this will solved this year ...

[1 more reply]

closed 2023-01-07 by manup