dresden-elektronik/deconz-rest-plugin #6463: Xiaomi Aqara curtain driver E1, model # CM-M01 not exposed to REST-API

zguy2008, 2022-10-24:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26780700)
 - deCONZ version: (2.18.2)
 - Device: RaspBee II
 - Is there any other USB or serial devices connected to the host system? None

## deCONZ Logs

## Additional context

zguy2008, 2022-10-24:
Looks like there is an issue with the template as the post doesn't show much.  trying to post the original text.

## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26780700)
 - deCONZ version: (2.18.2)
 - Device: RaspBee II
 - Is there any other USB or serial devices connected to the host system? None

## deCONZ Logs

## Additional context

zguy2008, 2022-10-24:
So what am I doing wrong here, practically all the information I provide doesn't show.

zguy2008, 2022-10-24:
Trying again...

rant
I posted an issue asking for which template to use.  Issue was closed as soon as it was posted.
You guys need to realize that sometimes we are not sure which template to use.
/rant

background (my initial post)
I recently bought 2 Xiaomi Aqara curtain driver E1, model # CM-M01.

The supported device page (https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#other-supported-devices) shows 2 different devices, a (supported I believe) Aqara smart curtain motor and a Aqara smart curtain motor B1 with an open issue. The later is not my device.

At this point, my assumption is my device is supported.

I added both devices to a Raspbee II:

Host system: Raspberry Pi
Running method: Raspbian
Firmware version: (26780700)
deCONZ version: (2.18.2)
Device: RaspBee II

I can read the clusters and control the start/stop and direction of the motor using the Window Covering cluster.

However, I do not see the devices in Phoscon app and they do not show up in the REST-API either. Seems like the devices are not exposed outside of deCONZ.

- Describe the bug
New devices added to deCONZ do not show up in REST-API.
They don't show up in Phoscon either but I don't really care about this.

- Steps to reproduce the behavior
  If the problem is reproducable, list the steps here:

  1. Using Postman send a GET request on
      - /config
      - /api
      - /sensors
      - /lights
  2. Search for MAC address or NWK address in Postman output produces no results.  The plugin using deCONZ REST-API doesn't see the new devices either.

- Expected behavior
  Having the devices exposed to REST-API so I can control them in Homeseer using Zigbee Jowihue plugin 

-- Screenshots
The 2 devices in deCONZ:
[image]

- Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26780700)
 - deCONZ version: (2.18.2)
 - Device: RaspBee II
 - Is there any other USB or serial devices connected to the host system? None

Mimiix, 2022-10-25:
Your probably putting the info between the comment lines.

Can you please share a screenshot of the basic cluster?

zguy2008, 2022-10-25:
Here is the basic cluster

[image]

[image]

Mimiix, 2022-10-25:
Hi,

 Looks like you have a new version. Can you please create a device request?

zguy2008, 2022-10-26:
OK, will do.

zguy2008, 2022-10-26:
Device request posted.

Mimiix, 2022-10-26:
Thanks!

Closing this case :)

closed 2022-10-26 by Mimiix