dresden-elektronik/deconz-rest-plugin #6464: Aqara Single Switch module not working since update to 2.19.01

evan-shaw, 2022-10-25:
## Describe the bug

After updating from 2.18.2 to 2.19.1 my Aqara T1 Switch doesn't work. [https://www.aqara.com/en/single_switch_T1_no-neutral.html](url)
In 2.18.2, and before, the switch was recognised as a light, I think this is correct.
But since the upgrade to 2.19.1, the switch appears as both a light and a switch, i.e. 2 entities are created. If you delete either one, the other is also removed. One of these entities is being exposed to home assistant, but it doesn't work. When attempting to switch it on/off in home assistant, I get the error message: -

 _[545860309824] 3 /lights/3 resource, /lights/3, not available_

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 4 (8GB)
 - Running method: Raspbian
 - Firmware version: 26720700
 - deCONZ version: 2.19.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

Smanar, 2022-10-25:
Unfortunately a DDf for this device was valided just 3 days ago https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6363

I don't think you have it on your deconz version so can you instal it manualy to test it ?

evan-shaw, 2022-10-25:
Hi Smanar, just checked out that link   "Add DDF for Aqara T1 single rocker switch (no neutral wire) SSM-U02"
That device is different, mine doesn't have a rocker switch, it's more a sort of relay that is placed in the wall behind an existing dumb rocker switch.

Mimiix, 2022-10-25:
It's in. but as the or said it's untested.

evan-shaw, 2022-10-25:
Ah OK, so seems not to be working at the moment then if #6363 is a similar issue to mine. Thanks

Mimiix, 2022-10-25:
@SwoopX you might be interested in testing with this user ?

Smanar, 2022-10-25:
@evan-shaw you have tested the swoopX DDF ?

evan-shaw, 2022-10-25:
Er no, I haven't seen it?

Smanar, 2022-10-25:
Here you have a DDF for "lumi.switch.l0agl1" https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6363

evan-shaw, 2022-10-25:
I would need  some instructions....

Smanar, 2022-10-26:
You have the theory here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet.

You have 2 folder used for DDF, one is used by deconz and will be reseted at every deconz uptade, the other is for "user"
You just need to create a text file inside, with this contain, name the file that_you_want.json and restart deconz.

If you have HA, have fun https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

evan-shaw, 2022-10-30:
@Smanar & @SwoopX  Hi Both, I tested the json from #6363 as suggested, it's currently in my deconz user directory. It worked. The "light" entity now works correctly in Home Assistant. Not sure if it's a problem, but the "switch" entity for the same device is still present in the phoscon web ui?

Smanar, 2022-10-30:
I think it's normal, I m reading the DDF and there is still the ZHASwitch probably to be used with decoupled mode.

Smanar, 2022-11-24:
From the comment the corrective is in the version 2.19.1 https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.19.1

samhed, 2022-12-19:
Do you have any idea when we can expect 2.19.1 to hit the repos?

This page only lists 2.19.0:
https://www.phoscon.de/en/changelog

And in raspbian repos I can only find 2.19.03.

Smanar, 2022-12-19:
http://deconz.dresden-elektronik.de/raspbian/beta/

I think there was a typo the 2.19.1 is the 2.19.01

samhed, 2022-12-20:
Hm, in that case, something is still amiss. I have deconz version `2.19.03-raspbian-buster-beta` and my Aqara T1 switches are still missing from phoscon. They disappear after deconz upgrade, and if I downgrade to 2.18.02 they reappear.

Smanar, 2022-12-21:
It's a "lumi.switch.l0agl1" ?
I can see the DDF in the folder `devices\xiaomi\xiaomi_ssm-u02_t1_switch.json`

The manufacture name is "LUMI" ?

samhed, 2022-12-22:
Yes, it is a `lumi.switch.l0agl1` and with version 2.19.03 I can also see the DDF here:
`/usr/share/deCONZ/devices/xiaomi\xiaomi_ssm-u02_t1_switch.json`

After downgrading to 2.18.02 I can see it in Phoscon:

[image]

As you can see, it lists "Vendor" as LUMI.

Smanar, 2022-12-22:
I don't see problem on, the DDF, deconz check onlu this to use the DDF

```
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.switch.l0agl1",
```

$MF_LUMI = LUMI

And if you try to re-include the device ?

samhed, 2022-12-23:
Do you mean removing it from the ZigBee network and then re-pairing it?

In that case, I'd remove it when on version 2.18.02, then upgrade to 2.19.03, and lastly attempt to re-pair it.

Smanar, 2022-12-23:
Yes, sorry if it's not a problem for you, because I don't see see what can be wrong on the DDF.

The manufacture name is good  "LUMI"
The model id is good "lumi.switch.l0agl1"
No special check in the DDF
The status is "Gold"
And no typo in the DDF

I can understand if the device is not working, but no it can't be included.

hermsi94, 2022-12-31:
The switch without neutral is actually just like the one with only without voltage, current and power measurement, if that helps. I have adjusted my DFF here and it is now recognized as a light and also displayed in Phoscon. Also it can now be controlled with NodeRed. Here is the DFF:
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.switch.l0agl1",
  "product": "Aqara Single Switch Module T1 without neutral",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}`

hermsi94, 2022-12-31:
The switch without neutral is actually just like the one with neutral only without voltage, current and power measurement. It is also an end device and not a router like the one with neutral, if that makes a difference ':D. 

I have adjusted my DFF here and it is now recognized as a light and also displayed in Phoscon. Also it can now be controlled with NodeRed. Here is the DFF if that helps:
```{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.switch.l0agl1",
  "product": "Aqara Single Switch Module T1 without neutral",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

[image]

hermsi94, 2022-12-31:
With the rest of the things I know unfortunately not so from :sweat_smile: . It must be because of the things that are different in the two versions like Type Switch or Clickmode or something.

It is actually like the working one with neutral only without all these measurement functions or not? :
[https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/xiaomi/xiaomi_lumi.switch.n0agl1_switch.json](url)

Smanar, 2022-12-31:
IDK, I don't know this device.

samhed, 2023-01-01:
@Smanar it's working now! I downgraded to the older version, removed my switches, then upgraded to the most recent version of Deconz. I then successfully could re-add them to the Zigbee network. Things work great now.

One difference that I noticed is that these devices have now started appearing both under "lights" and under "switches" in phoscon.

samhed, 2023-01-01:
@Smanar it's working now! I downgraded to the older version, removed my switches, then upgraded to the most recent version of Deconz. I then successfully could re-add them to the Zigbee network. Things work great now.

One difference that I noticed is that these devices have now started appearing both under "lights" and under "switches" in phoscon.

Anyway, thanks for the help!

Smanar, 2023-01-01:
>One difference that I noticed is that these devices have now started appearing both under "lights" and under "switches" in phoscon

Yep, it's from the Swoops's DDF, probably because you can "decouple' them.

closed 2023-01-31 by github-actions[bot]