dresden-elektronik/deconz-rest-plugin #6470: Universal Remote Control, UFO-R11, IR remote

DieuwerH, 2022-10-26:
## Device

- Product name: Universal Remote Control
- Manufacturer: _TZ3290_j37rooaxrcdcqosn
- Model identifier: TS1201
- Device type : Please remove all unrelated device types. 
  - Remote
  - Sensor

It is an IR remote that can learn from an existing remote and later transmit those signals.

Zigbee2mqtt page on this device: https://www.zigbee2mqtt.io/devices/UFO-R11.html

## Screenshots

### Basic

[image]

[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

### Power Configuration
[image]

Xyaren, 2023-02-02:
I bought a similar Device Called "Zigbee IR Remote Control"
Model **ZS-06** / **TS1201** / `_TZ3290_7v1k4vufotpowp9z`

Bought at https://www.aliexpress.com/item/1005003878194474.html
Zigbee2mqtt device link: https://www.zigbee2mqtt.io/devices/ZS06.html

### Device
[image]
### Basic
[image]
### On/OFF
[image]
### 
Node Info
[image]

I tested the device using zigbee2mqtt and it works.
But I prefer to use deconz :)

I don't see any other IR Blaster devices in the supported devices, so is this a first ?
How is the procedure? 
I would like to help, bringing IR Blasters to Deconz/Homeassistant.

Xyaren, 2023-02-06:
@Smanar you've been my savior in this project so far :D
I would like to help but I have no starting point.

Smanar, 2023-02-09:
Hu ... good question.
I don't think it will be possible yet, need a custom request to send code, and DDF can only use predefined one for the moment, (need a complete zigbee request, not just a n attribute read/write).
Have already the problem for that for covering request, so for completely new request ....

Xyaren, 2023-02-10:
The Zug we request seems proprietary.
Would this be accepted as code with an additional DDF?

Smanar, 2023-02-10:
>The Zigbee equest seems very proprietary.

I haven't say they don't respect zigbee rules, the problem is more on deconz side.

>Would this be accepted as code with an additional DDF?

"One day" the DDF core will be able to use custom request, but honnestly it can take so long time, and even you make all the c++ code yourself, no sure it will be merged as the c++ code need to move a lot for future version.

closed 2023-03-12 by github-actions[bot]