dresden-elektronik/deconz-rest-plugin #6471: Samotech SM313Z Temperature & Humidity Sensor

TheStork239, 2022-10-26:
## Device

- Product name: Samotech SM313Z temperature & humidity sensor
- Manufacturer: _TZ3000_5nrcorgu
- Model identifier: TS0201
- Device type :
  - Sensor

## Screenshots

### Basic
[image]

### Identify
[image]

### Other clusters that are not mentioned above
[image]
[image]
[image]
[image]

Smanar, 2022-10-27:
Hello can you try adding your model in this DDF to test it ? https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_yd2e749y_temp_hum_sensor.json

TheStork239, 2022-10-27:
> Hello can you try adding your model in this DDF to test it ? https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_yd2e749y_temp_hum_sensor.json

Thanks.  Not sure how I do that / what exactly you want me to do.  I run deConz via HomeAssistant add-on, and I don't seem to find a way to get to edit the relevant DDF file to try the edit.

And do you mean that in that file (if there is a way for me to get to it) I should simply edit lines 3 and 4 of the file to include a third item in the list, i.e.:  
"manufacturername": ["_TZ3000_yd2e749y", "_TZ3000_fllyghyj", "_TZ3000_5nrcorgu],
  "modelid": ["TS0201", "TS0201", "TS0201"],

But how do I get to the DDF file on a HomeAssistant install to do that?

Smanar, 2022-10-28:
Yep, it's exaclty that.
But not easy on HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

TheStork239, 2022-10-28:
Thanks.  I managed to make the edits to the required json file, but when I join the the device via deConz, it does not appear to be picking up the relevant DDF file for that device, and it is paired with no DDF file if I right click into the "edit DDF" after the pairing completes (picture attached).  And if I restart the deConz add-on, then all the edits to the suggested json doc are lost as the docker starts clean, so can't really test the usual "please see if a reboot helps"...
[image]

Smanar, 2022-10-29:
As it's just an edit, you can try that :
- select the node
- select "edit DDF"
- select file/load
- select the "devices/tuya/_TZ3000_yd2e749y_temp_hum_sensor.json"
- edit the string to add your device, need to edit manufacture and model id.
- try to save the file (idk what will happen here)
- make "file / hot relaod"

I think it will be enought to test the device, but the change will be not persisitent.

TheStork239, 2022-10-29:
Thanks.  That seems to work.  After the hot reload the device picks up a new name automatically e.g. "Temperature 27", and I can see the device also in the Phoscon app (see pic below).  
I haven't seen the device or its entities pop up in HomeAssistant yet, but that may require a restart of deConz or reboot of HA, which will make it lose the changes given they are not yet persistent.

[image]

[image]

Smanar, 2022-10-29:
>but that may require a restart of deConz or reboot of HA, which will make it lose the changes given they are not yet persistent.

Yeah, but if I m right there is a way in the HA plugin to resynchronise the plugin with deconz.

Just tell me if in 24h the battery is working or not ? And I will make the PR to add the device.

TheStork239, 2022-10-29:
Great - I managed to get the entities to appear in HA using the deConz device refresh service.
The battery reporting already started to work, showing 100% for one device, and 98% for another I also paired.
So it all seems to work using that DDF edit, and probably ready for the PR to add the device.
Thank you very much for your help!

[image]

Smanar, 2022-10-29:
Have added the device on this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

audiocrush, 2022-11-05:
Hi,

I'd also like to add the ID _TZ3000_ena8vgqb for this json file 
I have a Luminea Home Control Temperature and Humidity sensor that seems to be just a oem version of the same.
It worked fine for me when I added this ID to the json file you mentioned.

Smanar, 2022-11-05:
Thx, just added to the same PR

non092, 2022-11-22:
Hi,
Can you add _TZ3000_xr3htd96 as well ?
Seems to be a clone of the same device, I've edited the DDF yesterday and everything is reporting properly 👍 

[image]

Smanar, 2022-11-23:
Thx, just added (on same PR too)

TheStork239, 2022-12-18:
I guess this could be closed as soon as the PR mentioned above gets merged.  @Smanar do you know what the timelines for that merge might be?

Smanar, 2022-12-18:
Don't worry, the issu will be closed immediatly when the PR will be valided.

And no I don't have timeline, and somes devs are on holidays atm. I hope for a massive PR merge before Christmas

manuel-mauky, 2022-12-28:
Hi, I have a _TZ3000_ena8vgqb (Luminea Temperature sensor) and tested with the DDF file from the PR.
It seems to work. In DeConz I can see the newest measurement values (in Cluster Info) and they are updated every few minutes as expected.
However, in the Phoscon app, Home-Assistant and in the Rest-API these values aren't updated. There I can only see the value that was first read when I started DeConz but no following updates.

This whole zigbee topic is new for me so I'm not sure if this is related to the DDF file or is it a separate/unrelated issue?

Smanar, 2022-12-28:
I don't found special information about this device.
It don't have the tuya cluster ?
Can try to re include the device, without deleting it, the inclusion procedure can have failed and the bind/report was skipped.

manuel-mauky, 2022-12-28:
> It don't have the tuya cluster ?
I'm not sure what this means. Where can I see this?

Here are the screenshots, I hope this helps.

## Basic
[image]

## Node Info
[image]

## Identity
[image]

## Power
[image]

## Temperature
[image]

## Humidity
[image]

Smanar, 2022-12-29:
So your device look standard.

Perhaps it need the tuya unlock sequence (and this is not possible yet using deconz), but I can't find information about it on google.

And binds/report are fine in the DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/b2761cd529982fa13eaf4b56059221220a7f3237/devices/tuya/_TZ3000_yd2e749y_temp_hum_sensor.json (and work for other devices)

Have you tried to re-include the device without deleting it ?

@audiocrush you don't have problem on your side ?

manuel-mauky, 2022-12-29:
What do you mean with "re-include the device without deleting it"?
What should I do?

Smanar, 2022-12-30:
Just re pair it, in phoscon "add new sensor"

manuel-mauky, 2022-12-30:
Ok. I've tried this and it doesn't work either.

closed 2023-01-09 by manup