dresden-elektronik/deconz-rest-plugin #6473: Device name

mikeg-de, 2022-10-27:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2022-10-27:
Hello can you try to add your model this DDF to test it ? https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_n2egfsli_door_sensor.json

mikeg-de, 2022-10-27:
When trying to load a DDF via VNC I am getting an error stating the file could not get opened.

[image]

For testing and learning purposes I saved the json generated by deCONZ and by inspecitng in via vi I noticed the line breaks are "quite interesting"
[image]

Here is the file I created myself:
[image]

Any ideas?

Smanar, 2022-10-27:
>Here is the file I created myself:

But this one is the orignal one ? I don't see your device inside

```
  "manufacturername": ["_TZ3000_n2egfsli", "_TZ3000_oxslv1c9", "_TZ3000_7tbsruql", "_TZ3000_7d8yme6f", "_TZ3000_osu834un"],
  "modelid": ["TS0203", "TS0203", "TS0203", "TS0203", "TS0203"],
```

mikeg-de, 2022-10-27:
Thanks for the confirmation of the path. Just one note, though. Per the [DDF Cheat Sheet](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet) the paths can't be found:

[image]

The file you linked doesn't contain the manufacturer: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_n2egfsli_door_sensor.json

I added the recently shared code manually and will re-add the node soon. (taking care of my kids now). Will let you know once done.

Smanar, 2022-10-27:
>the paths can't be found:

What if your OS ?

```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ cd /usr/share/
pi@raspberrypi:/usr/share $ dir
accountsservice          glib-2.0                     perl
aclocal                  glvnd                        perl5
adduser                  gnome                        piclone
agnostics                gnome-background-properties  pi-greeter
alacarte                 gnome-control-center         pi-language-selector
alsa                     gnupg                        pi-package
.........
...........

```

>The file you linked doesn't contain the manufacture

? line 3 !

mikeg-de, 2022-10-27:
```# cat /etc/os-release
NAME="Home Assistant OS"
VERSION="9.3 (RaspberryPi 4 64bit)"
ID=haos
VERSION_ID=9.3
PRETTY_NAME="Home Assistant OS 9.3"
CPE_NAME=cpe:2.3:o:home-assistant:haos:9.3:*:production:*:*:*:rpi4-64:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS RaspberryPi 4 64bit"
VARIANT_ID=rpi4-64
SUPERVISOR_MACHINE=raspberrypi4-64
SUPERVISOR_ARCH=aarch64

# cd /usr/share/
# /usr/share $ dir
-sh: /usr/share: Permission denied 
```

The device is still not recognized
[image]

Smanar, 2022-10-28:
Ha it s not the "Home Assistant Operating System" ?
But one other installation ? Still using docker ?
Because you are the first one I see making like that with an HA OS ....

>Quick question is there a documentation available which would guide someone crating a DDF and then proposing it to cut a corner for you?

For the moment the better link is this one (there is a video too) https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
And the better way is using an existing DDF used for a similar device.

After that you just need to make a PR https://github.com/dresden-elektronik/deconz-rest-plugin/pulls

BTW wana make the PR or wana I make it ? (to submit officialy, because i don't think your file will be persistant on your OS)

closed 2022-11-26 by github-actions[bot]