dresden-elektronik/deconz-rest-plugin #6476: Bosch Radiator Thermostat II

h1k3r, 2022-10-27:
## Device

- Product name: Bosch Radiator Thermostat II.
- Manufacturer: clusters not showing (see screenshot)
- Model identifier: clusters not showing (see screenshot)
- Device type : Please remove all unrelated device types. 
  - Sensor
  - Thermostat

## Screenshots
[image]

[image]

## Problem

This is the new Bosch Radiator thermostat II. Unfortunately deCONZ can not retrieve the clusters of the ZigBee device. Probably for the proper paring ZigBee Install Code has to be used (it can be found on the device battery cover).

There is similar issue for another Bosch Smart Home ZigBee-based device: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6421.

SwoopX, 2022-11-06:
I should get the device in a couple of days, than I can see what I can do

h1k3r, 2022-11-11:
There is a new device request in zigbee2mqtt for the same model: https://github.com/Koenkk/zigbee2mqtt/issues/14005. Maybe it can be useful to get more information on device specification.

mmattel, 2022-11-27:
Also see: https://github.com/Koenkk/zigbee2mqtt/issues/14926 ([New device support]: Bosch Radiator Thermostat II) which includes references to integration PR's

besanur, 2022-12-09:
Did the DDF file working in your PR?

is the problem here with the install code?

loefchen, 2022-12-30:
I'm looking for a solution too. Anyone got this working with deCONZ?

closed 2023-01-18 by manup

jensflorian, 2023-03-06:
@h1k3r How did you pair the device in phoscon (as a light)?

h1k3r, 2023-03-06:
@jensflorian unfortunately I was not able to pair the device yet (also with the version 2.21.00).

SwoopX, 2023-03-07:
I just finished some working code to use install codes, which hopefully can be released to public without any issues. Probably not pretty, but gives the expected results. Hope we can internally discuss this and push it out real fast...

besanur, 2023-04-15:
Maybe the best option is to switch to zigbee2mqtt

besanur, 2023-06-27:
No Battery State is shown
Phoscon:
[image]
deCONZ:
[image]