dresden-elektronik/deconz-rest-plugin #6484: Ikea blinds Fyrtur and Kadrilj cannot be added

Siggi0904, 2022-10-30:
Hi,

since one of the latest versions, the Ikea blinds Fyrtur and Kadrilj cannot be added.
The DDF file is lost, I think. At release date in 2019 of the blinds, all was fine.
Checking and re-adding is needed.

See forum as well.
https://forum.phoscon.de/t/tradfri-blind-s-not-visible-in-web-app/2552

or
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121

Thanks.

## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

SwoopX, 2022-10-30:
The template is there for a reason and dancing around it is not exactly helpful

closed 2022-10-30 by SwoopX

Mimiix, 2022-10-30:
Hi

I'm closing this issue as it is not compliant with #5113. 

Swoop was ahead of me.

Siggi0904, 2022-10-30:
What shall I do? I have no point to fill out the template.
I can only say, what the problem is.