dresden-elektronik/deconz-rest-plugin #6485: Tuya Smart ZigBee Energy Meter  Single Phase 80A Din Rail Smart Energy Meter TS0601 _TZE200_lsanae15

doreks, 2022-10-30:
## Device

- Product name: Zigbee Tuya Single Phase 80A Din Rail Smart Energy Meter timer Monitor kWh Meter Wattmeter 110V 220V 
TS0601 _TZE200_lsanae15

https://es.aliexpress.com/item/1005004399475951.html?spm=a2g0o.order_list.0.0.2b31194dxGWGio&gatewayAdapt=glo2esp

- Manufacturer: _TZE200_lsanae15
- Model identifier: TS0601
- Device type : 
  -  Other: Smart Energy Meter

## Screenshots

Smanar, 2022-10-30:
It's not a clone of this one https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6465 ?

Can you try with adding your manufacture name to the list ?

doreks, 2022-10-30:
thanks Smanar for your support!!
im new with this zigbee and i have not clear how to add into PR my current manufacture. I can see how to access to DDF but once im in i don't have idea how to proceed... could you let me know with some info,.

thanks

Smanar, 2022-10-31:
Depend of OS.
If you have HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
Else you just need to create a file named "that_you_want.json" in the "devices" folder" with this contain (The PR and adding your model)

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

Then just restart deconz and include your device.

rejoe2, 2022-11-02:
Would you mind adding Manufacturername "_TZE200_hkdl5fmv" as well? I got the 100A version; see https://github.com/Koenkk/zigbee-herdsman-converters/issues/2962 for details on zigbee2mqtt.

rejoe2, 2022-11-02:
OK. Will take some time...

closed 2022-12-01 by github-actions[bot]