dresden-elektronik/deconz-rest-plugin #6486: Ikea blinds Fyrtur and Kadrilj cannot be added

Siggi0904, 2022-10-30:
## Describe the bug
Try to add the Ikea blinds Fyrtur and Kadrilj at phoscan app.
The blinds won't be found.
In 2019 at release of the blinds, all works fine.

## Steps to reproduce the behavior
Click searching for lights at phoscon app, press the two buttons at the blind for reset.
Nothing will be found at phoscon app, blind won't be connected to deconz.

## Expected behavior
Adding and controlling of ikea blinds Fyrtur and Kadrilj as it worked in latest versions around 2019.

## Screenshots
[image]

## Environment

 - Host system: Proxmox VM with Debian Bullseye
 - Running method: Debian bullseye VM with routed usb stick to VM
 - Firmware version: 26780700
 - deCONZ version: 2.19.01 / 19.9.2022
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Nothing else

## deCONZ Logs
have no, because of running headless.

## Additional context
It was working in older versions, around 2.16. Looks like DDF was missed.

Smanar, 2022-10-30:
But they are not supported by the API or not supported by phoscon ?

Mimiix, 2022-10-30:
Looks like this is an issue in phoscon, not deconz.

If there was a issue in the ddf, the device wouldn't show its name.

Can you check if it's showing in the rest api? You can find that out in phoscon unser help - api informatie

Siggi0904, 2022-10-30:
Where I can find the ddf?

I cannot (re-)adding the blinds to deconz / phoscon like the link in the forum.

Siggi0904, 2022-10-30:
They was supported and worked long time but now it won't.
See:
https://forum.phoscon.de/t/ikea-fyrtur-blinds/1858/5
https://dresden-elektronik.github.io/deconz-rest-doc/devices/ikea/kadrilj_roller_blind/

and 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121

Smanar, 2022-10-30:
You don't have other third than phoscon ?
Can you take a look in the API if the device is here, using for exemple Phoscon/help/API Information/light

Mimiix, 2022-10-30:
> No, only deconz and phoscon.
> 
> In the API page of phoscon I cannot see it, because device won't be added. The the screenshot from forum. It's the same like me, but I have no gui.
> 
> Other people have the same problem. See iobroker Forum for example or look at FHEM forum: https://forum.iobroker.net/topic/59432/conbee-ii-ikea-rollo

Please show us your deconz gui.

Siggi0904, 2022-10-30:
As I wrote, I have no deconz gui. I'm headless.
Only phoscon app.

And in deconz source file I found only ddf for KADRILJ not for Fyrtur
See: https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/ikea

Smanar, 2022-10-30:
>The the screenshot from forum. It's the same like me, but I have no gui.

On this capture the device is added else the node will have as name something like 0xXXXX

And the DDF have bronze status so never used, can try this DDF, enabled for both model

```
    "manufacturername": ["$MF_IKEA", "$MF_IKEA"],
    "modelid": ["KADRILJ roller blind", "FYRTUR block-out roller blind"],

```
```
{
    "schema": "devcap1.schema.json",
    "manufacturername": ["$MF_IKEA", "$MF_IKEA"],
    "modelid": ["KADRILJ roller blind", "FYRTUR block-out roller blind"],
    "product": "KADRILJ and FYRTUR",
    "status": "Gold",
    "supportsMgmtBind": true,
    "sleeper": false,
    "subdevices": [
        {
            "type": "$TYPE_WINDOW_COVERING_DEVICE",
            "restapi": "/lights",
            "uuid": ["$address.ext", "0x01"],
            "items": [
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "config/checkin",
                    "default": 480
                },
                {
                    "name": "state/lift",
                    "parse": {"cl": "0x0102", "at": "0x0008", "eval": "Item.val = Attr.val"},
                    "read": {"cl": "0x0102", "at": "0x0008"},
                    "refresh.interval": 700
                },
                {
                    "name": "state/open",
                    "parse": {"cl": "0x0102", "at": "0x0008", "eval": "Item.val = Attr.val < 100"}
                },
                {
                    "name": "state/reachable"
                },
                {
                    "name": "state/bri",
                    "description": "The lift state in earlier versions, use <code class=\"api-attribute\">lift</code> instead.",
                    "parse": {"cl": "0x0102", "at": "0x0008", "eval": "Item.val = Attr.val * 254 / 100"},
                    "read": {},
                    "deprecated": "2020-04-08"
                },
                {
                    "name": "state/on",
                    "description": "The open state in earlier versions, use <code class=\"api-attribute\">open</code> instead.",
                    "parse": {"cl": "0x0102", "at": "0x0008", "eval": "Item.val = (Attr.val * 254 / 100) > 0"},
                    "read": {},
                    "deprecated": "2020-04-08"
                }
            ],
            "example":{
                "etag": "ec590d6e73342fb20fb284111522e48d",
                "lastannounced": "2021-04-12T18:37:56Z",
                "lastseen": "2021-04-12T18:47Z",
                "manufacturername": "IKEA of Sweden",
                "modelid": "KADRILJ roller blind",
                "name": "Window covering device 4",
                 "state": {
                    "bri": 254,
                    "lift": 100,
                    "on": true,
                    "open": false,
                    "reachable": true
                },
                "swversion": "2.2.009",
                "type": "Window covering device",
                "uniqueid": "cc:cc:cc:ff:fe:b3:d9:e6-01"
            }
        },
        {
            "type": "$TYPE_BATTERY_SENSOR",
            "restapi": "/sensors",
            "uuid": [ "$address.ext", "0x01", "0x0001"],
            "fingerprint": { "profile": "0x0104", "device": "0x0402", "endpoint": "0x01", "in": ["0x0001", "0x0020", "0x0102"] },
            "items": [
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "config/on"
                },
                {
                    "name": "config/reachable"
                },
                {
                    "name": "state/battery",
                    "parse": {"cl": "0x0001", "at": "0x0021", "eval": "Item.val = Attr.val"},
                    "read": {"cl": "0x0001", "at": "0x0021"},
                    "refresh.interval": 1800
                },
                {
                    "name": "state/lastupdated"
                }
            ]
        }
    ],
    "bindings": [
        {
            "bind": "unicast",
            "src.ep": 1,
            "cl": "0x0001",
            "report": [
                {"at": "0x0021", "dt": "0x20", "min": 300, "max": 600, "change": "0x01" }
            ]
        },
        {
            "bind": "unicast", "src.ep": 1, "cl": "0x0020"
        },
        {
            "bind": "unicast",
            "src.ep": 1,
            "cl": "0x0102",
            "report": [
                {"at": "0x0008", "dt": "0x20", "min": 1, "max": 600, "change": "0x01" }
            ]
        }
    ]
}
```

Siggi0904, 2022-10-30:
@Smanar sounds great, where should I store these DDF?
What's the installation progress for the new DDF?

PR in waiting list? Sounds not good for devs.

Mimiix, 2022-10-30:
Hi guys,

I am not sure if this is a bug in deconz. It sure does not comply with #5113.

Closing this issue. I suggest moving to the forums for support.

Kind regards,
Dennis

closed 2022-10-30 by Mimiix