dresden-elektronik/deconz-rest-plugin #6492: Zll.db growing in size by itself. At over 800MB today... 20min+ deconz service startup time

bjornatic, 2022-11-02:
## Describe the bug
 noticed that the deconz service took more and more time at each reboot. Now it over 20min. The zll.db file is 875,1Mo today and was 849,6Mo yesterday.

I opened the database with "DB Browser for SQLite" and notices that the "groups" and "sensors" table has shows "49999" rows.. (wich isn't right I think). If I try to mass delete some rows from "groups" or "senors", I get a "database disk image is malformed" error.

I tried a backup from the Phoscon GUI, the file I get is 12,5Mo big. If I restore from this file not a single device is shown as if I was starting from scratch.

The closest similiar issue I found was this :  [Database corruption - 10000s of duplicate sensors in the sensors table · Issue #3972 · dresden-elektronik/deconz-rest-plugin · GitHub 1] #https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3972 and I tried the “solution” given with no success (I could not import the sql file into a database… to many “UNIQUE constraint failed”.

I reaaally don't want to start from scratch and pair all my 90 devices again. I'm begging for your help.

## Steps to reproduce the behavior
No idea. I guess I corrupted the database by upgrading / downgrading the firmware / software a couple of time (month ago).

## Expected behavior
Just a working config' :)

## Screenshots
[image]
[image]

## Environment
- Host system: Rpi 4B running from SSD on USB3 (over long cable)
 - Running method: Raspbian Bullseye
 - Firmware version: 26780700
 - deCONZ version: 2.19.01
 - Device:  RaspBee II (89 nodes connected among them a lot of Aqara stuff)
 - Is there any other USB or serial devices connected to the host system : yes Arduino MEGA with RFLink module

## deCONZ Logs
Output of APS,ZDP, ZCL  INFO and ERROR : https://pastebin.com/BrkFjVPU

## Additional context

RobGri001, 2022-11-04:
i´v exactly the same problem

madradrakie, 2022-11-05:
i have the same problem

bjornatic, 2022-11-07:
Today, the database is at 933Mo..

`pi@piDomo:~/.local/share/dresden-elektronik/deCONZ $ ls -la -h
total 945M
drwxr-xr-x 3 pi pi 4.0K Nov  7 10:22 .
drwxr-xr-x 3 pi pi 4.0K Aug 31 17:26 ..
-rw-r--r-- 1 pi pi 2.5K Nov  7 10:22 config.ini
-rw------- 1 pi pi  13M Nov  2 10:21 deCONZ.tar.gz
drwxr-xr-x 2 pi pi 4.0K Aug 31 17:26 devices
-rw-r--r-- 1 pi pi  16K Nov  2 09:51 session.default
-rw-r--r-- 1 pi pi   35 Nov  2 09:51 zcldb.txt
-rw-r--r-- 1 pi pi 933M Nov  7 10:15 zll.db
`

deluxestyle, 2023-02-12:
I see the same issue. can this be re-opened?