dresden-elektronik/deconz-rest-plugin #6495: Nedis Zigbee motion sensor ZBSM10WT

joggs, 2022-11-04:
## Device

- Product name: Nedis Zigbee motion sensor ZBSM10WT
- Manufacturer: _TZ3000_nss8amz9
- Model identifier: TS0202
- Device type : Please remove all unrelated device types. 
 
  - Sensor
 

## Screenshots

### Basic
[image]
### Identify
[image]
### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off
EMPTY
### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]
[image]
[image]
[image]

Smanar, 2022-11-05:
Hello can you try this DDF

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": ["_TZ3000_msl6wxk9", "_TZ3000_otvn3lne", "_TZ3000_mcxw5ehu", "_TZ3000_nss8amz9"],
   "modelid":["TS0202", "TS0202", "TS0202"],
   "product":"TS0202 Presence sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint": {
           "profile": "0x0104",
           "device": "0x0402",
           "endpoint": "0x01",
           "in": [
             "0x0000",
             "0x0001",
             "0x0500"
           ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "default":0
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":60,
               "max":3600,
               "change":"0x00000001"
            }
         ]
      }
   ]
}
```

joggs, 2022-11-05:
This is the first time i play with DDF files.
I run deconz in docker, so I've created a json file in the docker container an then load the json from the deconz vnc by selecting the nedis device, go to edit device. load and select the DDF. A lot of nice colors and settings appears, but I do not think the device has been updated as nothing happens and everything looks like before. What step am I missing?
When I added the nedis device from phoscon, I did never get the device found message. I only saw this new unknown device in the vnc interface. I tried to scan for sensors once again, but no hit. Please point me in the right direction. thanks

Smanar, 2022-11-05:
No, it's not a new DDF, I have kust added your model.
And yes, if you can confirm me the battery is working after 24h ^^, I will make the PR just after.

joggs, 2022-11-06:
Hi!
Battery still on 0%
[image]
[image]

Smanar, 2022-11-06:
Not good, it s like you have a battery drain, or an used battery, you need to divide the value by 2, so 156 mean 78%

Try to re-include the device, no need to delete it, just re-include, like this the procedure will be invisible
And if it still not working try this DDF

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": ["_TZ3000_msl6wxk9", "_TZ3000_otvn3lne", "_TZ3000_mcxw5ehu", "_TZ3000_nss8amz9"],
   "modelid":["TS0202", "TS0202", "TS0202"],
   "product":"TS0202 Presence sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint": {
           "profile": "0x0104",
           "device": "0x0402",
           "endpoint": "0x01",
           "in": [
             "0x0000",
             "0x0001",
             "0x0500"
           ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "awake":true,
               "parse":{
                  "at":"0x0021",
                  "cl":"0x0001",
                  "ep":1,
                  "eval":"Item.val = Attr.val / 2",
                  "fn":"zcl"
               },
               "read":{
                  "ep":1,
                  "cl":"0x0001",
                  "at":"0x0021"
               },
               "refresh.interval": 3700,
               "default":0
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":60,
               "max":3600,
               "change":"0x00000001"
            }
         ]
      }
   ]
}
```

joggs, 2022-11-09:
still battery 0%. batteryvalue is now 122, so it seems to drain pretty fast even though the pir is brand new and probably also the battery

Smanar, 2022-11-09:
BTW, after reflexion, if you read the value in the GUI, it need to update it in the DDF (if the bind/reporting don't work)

And I m seeing an error too in the DDF, miss a model id, the DDF is probalby not used

Pls try this DDF

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": ["_TZ3000_msl6wxk9", "_TZ3000_otvn3lne", "_TZ3000_mcxw5ehu", "_TZ3000_nss8amz9"],
   "modelid":["TS0202", "TS0202", "TS0202", "TS0202"],
   "product":"TS0202 Presence sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint": {
           "profile": "0x0104",
           "device": "0x0402",
           "endpoint": "0x01",
           "in": [
             "0x0000",
             "0x0001",
             "0x0500"
           ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "awake":true,
               "parse":{
                  "at":"0x0021",
                  "cl":"0x0001",
                  "ep":1,
                  "eval":"Item.val = Attr.val / 2",
                  "fn":"zcl"
               },
               "read":{
                  "ep":1,
                  "cl":"0x0001",
                  "at":"0x0021",
                  "fn":"zcl"
               },
               "refresh.interval": 3700,
               "default":0
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":60,
               "max":3600,
               "change":"0x00000001"
            }
         ]
      }
   ]
}
```

joggs, 2022-11-10:
Still 0% battery  unfortunately

Smanar, 2022-11-10:
Even if you read the battery in the GUI ? attribute 0x0021 in the cluster 0x0001 ?
Check first in the json, before forcing the update.

joggs, 2022-11-10:
See screens below. Just to be clear. The way I update the DDF is to select the sensor in the vnc gui, edit, edit DDF, File->open (the json that I have saved in a ddf file), File->Save. And then I set the sensor in pairing mode without deleting it and search for new sensors, and the sensor is the discovered. After that I suppose the device should be configured with the new ddf?

[image]
[image]

Mimiix, 2022-11-10:
You need to do a hot reload or restart deconz afaik.

joggs, 2022-11-11:
Ok, I'll give it a shot. I noticed now that the sensor also goes from detected to not detected every 5 seconds or so, so it looks like the sensing part is strange. Is that also configurable in a ddf ?

Smanar, 2022-11-11:
Not normal, and it's the "standard" DDF used for all Tuya motion sensor devices, I have checked other zigbee projct, there is nothing special for this device.

I don't see the problem, nothing usefull in logs (in deconz/help/debug view) with flag "info" "info_l2" or "error" and "error_l2" ?

Smanar, 2022-11-25:
Hello, I have perhpas found something, can you try this   DDF

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": ["_TZ3000_msl6wxk9", "_TZ3000_otvn3lne", "_TZ3000_mcxw5ehu", "_TZ3000_nss8amz9"],
   "modelid":["TS0202", "TS0202", "TS0202", "TS0202"],
   "product":"TS0202 Presence sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint": {
           "profile": "0x0104",
           "device": "0x0402",
           "endpoint": "0x01",
           "in": [
             "0x0000",
             "0x0001",
             "0x0500"
           ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "awake":true,
               "parse":{
                  "at":"0x0021",
                  "cl":"0x0001",
                  "ep":1,
                  "eval":"Item.val = Attr.val / 2",
                  "fn":"zcl"
               },
               "read":{
                  "ep":1,
                  "cl":"0x0001",
                  "at":"0x0021",
                  "fn":"zcl"
               },
               "refresh.interval": 3700,
               "default":0
            },
            {
               "name":"config/duration",
               "default": 120
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":60,
               "max":3600,
               "change":"0x00000001"
            }
         ]
      }
   ]
}
```

joggs, 2022-11-25:
Hi!
Thanks.
I think the detection is working with this ddf. It triggers a motion
detected for 5 seconds and then get back to not detected after 5 seconds,
which is fine for me.
The battery percentage is still 158% and shows 0% in phoscon, so that is
not working, but I suppose it is not possible for you to divide the value
with 2 in the ddf, which possibly could result in a correct value in
phoscon as well as it then wil be below 100, but that is just a guess.

Smanar, 2022-11-25:
5s, hu, it need to be 120s.

> but I suppose it is not possible for you to divide the value
with 2 in the ddf

It's possible but the value is 0 in the API.
There is a poll if the report not working `"refresh.interval": 3700,` so after 3700s deconz is making a poll to retreive the value.
Can you check direclty in the API to see if the problem is not from phosocn, can see it in phoscon/help/API Information/sensors

Jalle19, 2023-01-14:
This should not be closed - this device does _not_ work out of the box with Deconz at the moment.

Smanar, 2023-01-14:
Hello, if you have the device, can you check if the battery is working on deconz ?

Jalle19, 2023-01-16:
@Smanar with the DDF you posted in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6495#issuecomment-1327295694 I was able to get the device added in Phoscon, and as far as I can tell everything including battery level looks good (it's a new device so I expect battery to be near full):
[image]
What's the next step here? Can you make a pull request with your DDF? I'm not familiar with the process myself.

Smanar, 2023-01-17:
PR done.

closed 2023-01-18 by manup