dresden-elektronik/deconz-rest-plugin #6497: _TZE200_hkdl5fmv 4 Phases DIN Rail switch with power metering

rejoe2, 2022-11-05:
## Device

- Product name:  ZJSBL7-100Z
- Manufacturer: _TZE200_hkdl5fmv
- Model identifier: TS0601
- Device type : 
  - Switch (4 Phases with Power Metering, 100A)

## Screenshots

[image]
[image]
[image]
[image]
Tried to just add the device like mentionned in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6485#issuecomment-1300444232, but didn't work.

For zigbee2mqtt see https://github.com/Koenkk/zigbee-herdsman-converters/issues/2962.

Smanar, 2022-11-05:
Can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_hkdl5fmv",
  "modelid": "TS0601",
  "product": "ZJSBL7-100Z",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
      {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 113, "eval": "Item.val = Attr.val * 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 102, "eval": "Item.val = Attr.val / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 101, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 103, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

Value need convertion ofc, but from Z2M it seem returned value are array that need calculation.
So if you can share Logs for tension for exemple (The easier to know the real value) using "DDF" in deconz / help / debug view

rejoe2, 2022-11-05:
Hope this ist sth. like you are looking for:
`15:13:56:149 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:13:56:521 update ddf TRADFRI bulb E27 WW 806lm index 169
15:13:57:904 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:01:165 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:02:902 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:06:149 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:07:902 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:09:857 update ddf TRADFRI bulb E27 WW 806lm index 169
15:14:11:148 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:12:901 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:16:150 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:16:861 update ddf HG06338 index 170
15:14:16:910 update ddf HG06338 index 170
15:14:17:904 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:18:313 update ddf TRADFRI bulb E27 WW 806lm index 169
15:14:21:150 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:22:907 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:24:008 update ddf A60 TW Value II index 176
15:14:26:149 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:27:905 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:31:150 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:32:903 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:36:148 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:37:907 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:40:978 00:0d:6f:ff:fe:ae:15:ba-01/attr/swversion expression: let v = Attr.val;
Item.val = String((v & 192) >> 6) + '.' + String((v & 48) >> 4) + '.' + String(v & 15);
 --> 1.0.8
15:14:40:981 00:0d:6f:ff:fe:ae:15:ba-01-0b04/attr/swversion expression: let v = Attr.val;
Item.val = String((v & 192) >> 6) + '.' + String((v & 48) >> 4) + '.' + String(v & 15);
 --> 1.0.8
15:14:40:983 00:0d:6f:ff:fe:ae:15:ba-01-0702/attr/swversion expression: let v = Attr.val;
Item.val = String((v & 192) >> 6) + '.' + String((v & 48) >> 4) + '.' + String(v & 15);
 --> 1.0.8
15:14:41:149 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:42:904 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:46:148 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:47:905 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:51:153 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:52:924 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:14:56:148 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:14:57:903 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:15:01:148 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:15:02:903 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:15:06:148 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
15:15:07:904 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
15:15:11:147 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
^A15:15:12:901 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
`
There's been attached just a single 6W LED Bulb on the first of the three power out connectors. Voltage should have been sth. around 233V.

Smanar, 2022-11-05:
```
15:14:52:924 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0 
15:14:57:903 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0 
```

Bad luck ^^, value is 0, so hard to convert.
And no debug line for tension.

You can enable "info", you will see raw value in log starting by "TY_DATA_"

rejoe2, 2022-11-06:
So this ist with info, one time switching on and off:
`
19:56:56:539 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
19:56:56:540 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 1
19:56:56:541 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 1
19:56:56:542 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 1
19:56:56:542 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 1
19:56:56:543 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 1
19:56:56:795 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> false
19:56:56:796 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:56:56:804 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:56:56:805 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:56:56:806 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:56:56:807 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:56:57:546 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> false
19:56:57:547 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:56:57:548 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:56:57:549 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:56:57:550 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:56:57:550 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:56:57:799 TY_DATA_RESPONSE: seq 515, dpid: 0x09, type: 0x02, length: 4, val: 0
19:56:57:800 TY_DATA_RESPONSE: seq 515, dpid: 0x09, type: 0x02, length: 4, val: 0
19:56:57:801 TY_DATA_RESPONSE: seq 515, dpid: 0x09, type: 0x02, length: 4, val: 0
19:56:57:802 TY_DATA_RESPONSE: seq 515, dpid: 0x09, type: 0x02, length: 4, val: 0
19:56:57:802 TY_DATA_RESPONSE: seq 515, dpid: 0x09, type: 0x02, length: 4, val: 0
19:56:58:303 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0
19:56:58:305 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0
19:56:58:306 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0
19:56:58:306 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0
19:56:58:307 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0
19:56:58:332 max transmit errors for node 0xA4C138801F16B2AA, last seen by neighbors 69 s
19:56:58:564 TY_DATA_RESPONSE: seq 515, dpid: 0x69, type: 0x02, length: 4, val: 29
19:56:58:565 TY_DATA_RESPONSE: seq 515, dpid: 0x69, type: 0x02, length: 4, val: 29
19:56:58:566 TY_DATA_RESPONSE: seq 515, dpid: 0x69, type: 0x02, length: 4, val: 29
19:56:58:567 TY_DATA_RESPONSE: seq 515, dpid: 0x69, type: 0x02, length: 4, val: 29
19:56:58:567 TY_DATA_RESPONSE: seq 515, dpid: 0x69, type: 0x02, length: 4, val: 29
19:56:58:808 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
19:56:58:809 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
19:56:58:810 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
19:56:58:811 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
19:56:58:811 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
19:56:59:062 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 0
19:56:59:063 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 0
19:56:59:064 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 0
19:56:59:065 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 0
19:56:59:067 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
19:56:59:068 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 0
19:56:59:309 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 0
19:56:59:310 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 0
19:56:59:311 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 0
19:56:59:311 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 0
19:56:59:312 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 0
19:57:01:540 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> false
19:57:01:541 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:57:01:542 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:57:01:543 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:57:01:543 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:57:01:544 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:57:02:560 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0
19:57:02:561 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0
19:57:02:562 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0
19:57:02:562 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0
19:57:02:563 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0
19:57:02:811 TY_DATA_RESPONSE: seq 515, dpid: 0x69, type: 0x02, length: 4, val: 30
19:57:02:812 TY_DATA_RESPONSE: seq 515, dpid: 0x69, type: 0x02, length: 4, val: 30
19:57:02:813 TY_DATA_RESPONSE: seq 515, dpid: 0x69, type: 0x02, length: 4, val: 30
19:57:02:813 TY_DATA_RESPONSE: seq 515, dpid: 0x69, type: 0x02, length: 4, val: 30
[... cut ...]
19:57:26:542 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> false
19:57:26:543 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:57:26:544 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:57:26:545 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:57:26:545 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
19:57:26:546 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 0
`
Still quite a lot of zeros...

Smanar, 2022-11-07:
>Still quite a lot of zeros...

Yeah, not good, I hope the device is working, no zero value are

The state on / off

```
19:56:56:543 TY_DATA_RESPONSE: seq 515, dpid: 0x01, type: 0x01, length: 1, val: 1 
19:56:56:795 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> false 
```

From the documentation, this one is the temperature

`19:56:58:567 TY_DATA_RESPONSE: seq 515, dpid: 0x69, type: 0x02, length: 4, val: 29`

So 29°C

Used but with zero values
```

19:56:57:801 TY_DATA_RESPONSE: seq 515, dpid: 0x09, type: 0x02, length: 4, val: 0 > timer
19:56:58:303 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0 > leakage current
19:56:58:808 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0 > Remaining Energy
19:56:59:062 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 0  > power
19:56:59:309 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 0 > reverse active power
```

At least the power need to work, you have a device connected to it, you are sure for the wiring ?

rejoe2, 2022-11-07:
The 29°C might be ok, that's some degrees above the room temperature.

Wrt. to the wiring, I'm absolutely **not sure**, as I just connected one of the three 230V lines and neutral and attached a LED bulb to the respective out port. This was ok to get the switch integrated in the ZigBee network and check the basic switching functions. Will have to check how to set up a reasonable test wiring (hopefully this evening)...
(PS: how to format the debug outputs to make it better readable?)

Smanar, 2022-11-07:
>(PS: how to format the debug outputs to make it better readable?)

It's the linux copy/paste ^^, try on pastebin site.
But you can just look at your logs for the good line in live ( a line with better value than 0 ^^), better for your SD card.

rejoe2, 2022-11-07:
Sorry, no luck with other values than zero (apart from temperature and state)... Attached the switch to a 3-Phase line+neutral and a second bulb on each of phases 1 and 2. Still only zeros. Did we miss something or might the switch be just defect?

Smanar, 2022-11-08:
From that I m reading this device expose only 1 value for current, mean it need to be used with triphase devices (same consumption by phase) no ?

Else you can try to enable log "aps" and aps_l2", you will have a strange log like

`asdu (length: 9): 10e400050508050b05`

It's raw value, I can translate it (easier if you can share too the line just after with "TY_DATA_RESPONSE" ^^) to see if the zero is realy a zero or if it's a problem in deconz.

rejoe2, 2022-11-09:
OK, so here's some lines that may contain some more info:

I also addressed the seller's support. They told me, it would only be possible to get this data using the Tuya app or smart life app, (hahah) and advised to use sth. consuming more than my LED bulb. So this ist with a classic 60W light bulb combined with some energy saving bulb...
________________
20:57:12:231 APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -49
20:57:12:232    asdu: 09140102036802000400000000
20:57:12:234 APS-DATA.request id: 28, addrmode: 0x02, addr: 0xac63, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04
20:57:12:235    asdu (length: 5): 10140b0100
20:57:12:261 APS-DATA.confirm id: 28, status: 0x00 SUCCESS
20:57:12:262 APS-DATA.confirm request id: 28 -> erase from queue
20:57:12:324 aps request id: 28 finished, erase from queue
20:57:12:484 APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -49
20:57:12:485    asdu: 09150102036902000400000019
20:57:12:487 APS-DATA.request id: 31, addrmode: 0x02, addr: 0xac63, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04
20:57:12:488    asdu (length: 5): 10150b0100
20:57:12:512 APS-DATA.confirm id: 31, status: 0x00 SUCCESS
20:57:12:513 APS-DATA.confirm request id: 31 -> erase from queue
20:57:12:564 aps request id: 31 finished, erase from queue
20:57:12:733 APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -49
20:57:12:734    asdu: 09160102036a02000400000000
20:57:12:736 APS-DATA.request id: 34, addrmode: 0x02, addr: 0xac63, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04
20:57:12:737    asdu (length: 5): 10160b0100
20:57:12:766 APS-DATA.confirm id: 34, status: 0x00 SUCCESS
20:57:12:767 APS-DATA.confirm request id: 34 -> erase from queue
20:57:12:804 aps request id: 34 finished, erase from queue
20:57:12:980 APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -49
20:57:12:981    asdu: 09170102037102000400000000
20:57:12:984 00:0d:6f:ff:fe:ae:15:ba-01-0702/state/consumption expression: Item.val = Attr.val * 10; --> 0
20:57:12:986 APS-DATA.request id: 37, addrmode: 0x02, addr: 0xac63, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04
20:57:12:988    asdu (length: 5): 10170b0100
20:57:13:018 APS-DATA.confirm id: 37, status: 0x00 SUCCESS
20:57:13:019 APS-DATA.confirm request id: 37 -> erase from queue
20:57:13:044 aps request id: 37 finished, erase from queue
20:57:13:205 APS-DATA.request id: 39, addrmode: 0x03, addr: 0xf0d1b80000118ab1, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
20:57:13:206    asdu (length: 2): d206
20:57:13:236 APS-DATA.confirm id: 39, status: 0x00 SUCCESS
20:57:13:237 APS-DATA.confirm request id: 39 -> confirmed, timeout 524074951
20:57:13:249 APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -49
20:57:13:250    asdu: 09180102037502000400000000
20:57:13:252 APS-DATA.request id: 41, addrmode: 0x02, addr: 0xac63, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x04
20:57:13:254    asdu (length: 5): 10180b0100
20:57:13:270 APS-DATA.indication srcAddr: 0x0478, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -52
20:57:13:271    asdu: d2000e0603fe8704ffff2e21003a6268feff57b414efa925020ffffe8704ffff2e2100ba15aefeff6f0d0063ac25020ffefe8704ffff2e2100e240170000b8d1f0adc925020fff
20:57:13:272 APS-DATA.indication request id: 39 -> finished
20:57:13:272 APS-DATA.request id: 39 erase from queue
20:57:13:290 APS-DATA.confirm id: 41, status: 0x00 SUCCESS
20:57:13:291 APS-DATA.confirm request id: 41 -> erase from queue
20:57:13:364 aps request id: 41 finished, erase from queue
20:57:16:225 APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -49
20:57:16:226    asdu: 09190102030101000101
20:57:16:229 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
20:57:16:231 APS-DATA.request id: 56, addrmode: 0x02, addr: 0xac63, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04
20:57:16:232    asdu (length: 5): 10190b0100
20:57:16:259 APS-DATA.confirm id: 56, status: 0x00 SUCCESS
20:57:16:260 APS-DATA.confirm request id: 56 -> erase from queue
20:57:16:326 aps request id: 56 finished, erase from queue
20:57:16:484 APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -49
20:57:16:485    asdu: 091a01020365000006091c000b0005
20:57:16:487 APS-DATA.request id: 59, addrmode: 0x02, addr: 0xac63, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04
20:57:16:488    asdu (length: 5): 101a0b0100
20:57:16:513 APS-DATA.confirm id: 59, status: 0x00 SUCCESS
20:57:16:514 APS-DATA.confirm request id: 59 -> erase from queue
20:57:16:564 aps request id: 59 finished, erase from queue
20:57:16:735 APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 212, rssi: -49
20:57:16:736    asdu: 091b01020366000009000105000000000000
[... cut ...]
20:58:09:364 APS-DATA.request id: 32, addrmode: 0x03, addr: 0xbc33acfffe479b82, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
20:58:09:394 APS-DATA.confirm id: 32, status: 0x00 SUCCESS
20:58:09:395 APS-DATA.confirm request id: 32 -> confirmed, timeout 524131110
20:58:09:420 APS-DATA.indication srcAddr: 0x5a05, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 240, rssi: -76
20:58:09:421 APS-DATA.indication request id: 32 -> finished
20:58:09:422 APS-DATA.request id: 32 erase from queue
20:58:11:229 00:0d:6f:ff:fe:ae:15:ba-01/state/on expression: Item.val = Attr.val; --> true
____________________

rejoe2, 2022-11-11:
Maybe it's really because it is not properly connected yet? I wanted to make sure it is ok prior to let it be included in my steady installation... It will still take some weeks until this will happen (or I might opt to use the "bigger brother" with additional "FI" functionality).

How could I do the "check again it's the dp 113 (0x71)"? 
I have to admitt to be not very deep in the ZigBee protocol details nor the debugging things around it.

closed 2022-12-10 by github-actions[bot]

rejoe2, 2023-01-19:
Comming back to this...

I atm have two of the _TZE200_hkdl5fmv running, the original one and another one with additional residual current circuit breaker function. Both of them seem so measure something, especially the value sent as total consumed energy seems to be fairly plausible, using state/consumption for dpid 113. Stepwidth seems to be 10Wh. 
Unfortunately, there still are only zero values for the other things, especially power. 

For later reference, I found https://raw.githubusercontent.com/kkossev/Hubitat/main/Drivers/Tuya%20Zigbee%20Metering%20Plug/Tuya%20Zigbee%20Metering%20Plug, that seems to implement a more complex data anlysis for the DIN rail variants.

Smanar, 2023-01-19:
Yes the calculation is more complex.

For exemple for current we are using

```
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 102, "eval": "Item.val = Attr.val / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
```

And need to use something almost like that (need to convert the value in byte array first)

        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 102, "eval": "Item.val = (Attr.val[2] | Attr.val[1] << 8) / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },

But need to have other value than 0 as result, else even using the good convertion will do nothing 

>19:56:59:062 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 0  > power

rejoe2, 2023-01-20:
There's no display, but atm there's definitely some current, as consumption is also rising. There's attached a heavy consumer (around 7kW) switching on periodically.
Unfortunately, the proposed parse fn for state/current seems not to deliver any results yet. Will be abroad some days, then I'll have again a look on the communication details.

Reporting of the gadget seems to be rather complex.

Smanar, 2023-01-20:
>19:56:59:062 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 0 > power

Using this log, you can read all returns from the device, and they are 0 too on your previous logs

```
19:56:57:801 TY_DATA_RESPONSE: seq 515, dpid: 0x09, type: 0x02, length: 4, val: 0 > timer
19:56:58:303 TY_DATA_RESPONSE: seq 515, dpid: 0x68, type: 0x02, length: 4, val: 0 > leakage current
19:56:58:808 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0 > Remaining Energy
19:56:59:062 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 0  > power
19:56:59:309 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 0 > reverse active power
```

If you enable APS+APS_L2 you will have raw zigbee value, but I m almost sure they are 0 too

rejoe2, 2023-03-05:
One more try...

```12:52:42:155 TY_DATA_RESPONSE: seq 516, dpid: 0x69, type: 0x02, length: 4, val: 33
12:52:42:155 TY_DATA_RESPONSE: seq 516, dpid: 0x69, type: 0x02, length: 4, val: 33
12:52:42:156 TY_DATA_RESPONSE: seq 516, dpid: 0x69, type: 0x02, length: 4, val: 33
12:52:42:156 TY_DATA_RESPONSE: seq 516, dpid: 0x69, type: 0x02, length: 4, val: 33
12:52:42:157 TY_DATA_RESPONSE: seq 516, dpid: 0x69, type: 0x02, length: 4, val: 33
12:52:42:158 Websocket 127.0.0.1:60988 send message: {"e":"changed","id":"90","r":"sensors","state":{"current":0,"lastupdated":"2023-03-05T11:52:42.156","power":0,"temperature":3300,"voltage":0},"t":"event","uniqueid":"2c:11:65:ff:fe:4c:c8:26-01-0b04"} (ret = 199)
12:52:42:189 Skip idle timer callback, too early: elapsed 170 msec
12:52:42:206 skip verify rule Rule TOOGLE_ON for sensor 90:fd:9f:ff:fe:fe:2b:91-01-1000 (available = 0, node = (nil), sensorNode = (nil))
12:52:42:251 TY_DATA_RESPONSE: seq 517, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:252 TY_DATA_RESPONSE: seq 517, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:253 TY_DATA_RESPONSE: seq 517, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:253 TY_DATA_RESPONSE: seq 517, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:254 TY_DATA_RESPONSE: seq 517, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:254 TY_DATA_RESPONSE: seq 517, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:301 skip verify rule Rule TOOGLE_OFF for sensor 90:fd:9f:ff:fe:fe:2b:91-01-1000 (available = 0, node = (nil), sensorNode = (nil))
12:52:42:345 TY_DATA_RESPONSE: seq 517, dpid: 0x68, type: 0x02, length: 4, val: 0
12:52:42:347 TY_DATA_RESPONSE: seq 517, dpid: 0x68, type: 0x02, length: 4, val: 0
12:52:42:348 TY_DATA_RESPONSE: seq 517, dpid: 0x68, type: 0x02, length: 4, val: 0
12:52:42:348 TY_DATA_RESPONSE: seq 517, dpid: 0x68, type: 0x02, length: 4, val: 0
12:52:42:349 TY_DATA_RESPONSE: seq 517, dpid: 0x68, type: 0x02, length: 4, val: 0
12:52:42:350 TY_DATA_RESPONSE: seq 517, dpid: 0x68, type: 0x02, length: 4, val: 0
12:52:42:364 TY_DATA_RESPONSE: seq 518, dpid: 0x71, type: 0x02, length: 4, val: 54331
12:52:42:366 TY_DATA_RESPONSE: seq 518, dpid: 0x71, type: 0x02, length: 4, val: 54331
12:52:42:367 TY_DATA_RESPONSE: seq 518, dpid: 0x71, type: 0x02, length: 4, val: 54331
12:52:42:368 TY_DATA_RESPONSE: seq 518, dpid: 0x71, type: 0x02, length: 4, val: 54331
12:52:42:368 TY_DATA_RESPONSE: seq 518, dpid: 0x71, type: 0x02, length: 4, val: 54331
12:52:42:370 TY_DATA_RESPONSE: seq 518, dpid: 0x71, type: 0x02, length: 4, val: 54331
12:52:42:372 Websocket 127.0.0.1:60988 send message: {"e":"changed","id":"89","r":"sensors","state":{"consumption":543310,"lastupdated":"2023-03-05T11:52:42.370"},"t":"event","uniqueid":"2c:11:65:ff:fe:4c:c8:26-01-0702"} (ret = 167)
12:52:42:458 TY_DATA_RESPONSE: seq 518, dpid: 0x69, type: 0x02, length: 4, val: 24
12:52:42:459 TY_DATA_RESPONSE: seq 518, dpid: 0x69, type: 0x02, length: 4, val: 24
12:52:42:460 TY_DATA_RESPONSE: seq 518, dpid: 0x69, type: 0x02, length: 4, val: 24
12:52:42:460 TY_DATA_RESPONSE: seq 518, dpid: 0x69, type: 0x02, length: 4, val: 24
12:52:42:461 TY_DATA_RESPONSE: seq 518, dpid: 0x69, type: 0x02, length: 4, val: 24
12:52:42:462 TY_DATA_RESPONSE: seq 518, dpid: 0x69, type: 0x02, length: 4, val: 24
12:52:42:477 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 54331
12:52:42:478 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 54331
12:52:42:478 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 54331
12:52:42:479 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 54331
12:52:42:479 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 54331
12:52:42:480 TY_DATA_RESPONSE: seq 515, dpid: 0x75, type: 0x02, length: 4, val: 54331
12:52:42:566 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:567 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:567 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:568 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:568 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:568 TY_DATA_RESPONSE: seq 515, dpid: 0x6A, type: 0x02, length: 4, val: 0
12:52:42:652 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 5566
12:52:42:653 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 5566
12:52:42:654 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 5566
12:52:42:654 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 5566
12:52:42:655 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 5566
12:52:42:656 TY_DATA_RESPONSE: seq 515, dpid: 0x71, type: 0x02, length: 4, val: 5566```
seems this ist no longer only zero values, right? 

(reactive?) Power should be something in the range of 770W.

Smanar, 2023-03-05:
So you have value for 
0x69 > Temperature
0x71 > Power > Ok for this one
0x75 > Reactive Power ? (It's called reversed)

But I think I have understand why some value are not displayed in logs
For the current

```
20:58:06:736 APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -49
20:58:06:737 asdu: 097501020366000009000105000000000000
```

097501 02 03 6600 00 09 000105000000000000

Dp = 0x66 and Type 0x00
Raw values are ignored by deconz.

So you will see this line
>APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00

But not this one after
>TY_DATA_RESPONSE:

If I make a code modification are you able to compile the code to test it ? You need a complete linux machine with full OS ?

rejoe2, 2023-03-05:
> So you have value for 0x69 > Temperature 
Temp is there, and also consumption (? there seems not to be any distinction wrt. to the one used for PV and the other one with real consumption):
```
{
      "name": "state/current",
      "parse": {"fn": "tuya", "dpid": 102, "eval": "Item.val = (Attr.val[2] | Attr.val[1] << 8) / 1000;" },
      "read": {"fn": "none"},
      "default": 0
    }, 

```
> 0x71 > Power 
> Ok for this one 0x75 
> Reactive Power ? (It's called reversed)
Didn't do more parse instructions than for consuption and temp until now. Added power recently, but it's to dark now to get results.
> But I think I have understand why some value are not displayed in logs For the current
>
> ```
> 20:58:06:736 APS-DATA.indication srcAddr: 0xac63, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: [...]
deconz is running on a debian bullseye x64 machine, but I'm not sure if there's all dev tools on it (it's more like a minimal system), and my experience in compiling entire applications is more or less zero...
My laptop is running on kubuntu linux, but using a different machine might cause other problems, so I'd prefer to give it a try on the bullseye server.

How to proceed?

Remark: There seem to exist two different hardwares for the same zigbee identifier, if I got the code in https://raw.githubusercontent.com/kkossev/Hubitat/main/Drivers/Tuya%20Zigbee%20Metering%20Plug/Tuya%20Zigbee%20Metering%20Plug right - one for three/four phases and one for just single phase.

Wrt. to the wiring, I'm quite sure it's ok at least now.

Smanar, 2023-03-06:
>There seem to exist two different hardwares for the same zigbee identifier

Lol, I hate tuya.

>Temp is there, and also consumption

Yep those one are here, but I never see the tension, from Z2M it's the dp 110 (0x6E), not visible in yout log at all.
For the current, the dpid 0x66 is sometime.

>How to proceed?

So first you are right, the compilation is host dependendt, so you can't compile on a host and copy the file on another.

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing that explain what are doing command line.
So for you, after having installed deconz (The code is based on realy last one, so need a recent version):

```
sudo apt install deconz-dev
git clone --branch tuya_raw https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

It don't impact your zigbee database, but can use a lot the CPU (at least the first time, compilation will be faster after).
To rollback fastly you can make a backup of the file /usr/share/deCONZ/plugins/libde_rest_plugin.so in another folder

If it work you will be able to see more debug line like this one
>12:52:42:458 TY_DATA_RESPONSE: seq 518, dpid: 0x69, type: 0x02, length: 4, val: 24

The lines with type = 0x00 (actually not present in your logs)