dresden-elektronik/deconz-rest-plugin #6498: Woox R7054 4 buttons

jordanderoubaix, 2022-11-05:
## Device

- Product name: Woox Smart Security Remote Control (4 buttons)
- Manufacturer: _TZ3000_Ozrecfgx
- Model identifier: TS0215A
- Device type :
  - Switch

This switch is very similar to Woox R7052 (this one has 4 buttons instead of 1), I have tried to use the R7052 DDF from #4323 and I am able to use only 1 button (it was expected :) ).

And it seems to be the same device than #4524 but I was not able to find any solution for this one.

## Screenshots

### Node info

(after setting manually R7052 DDF)
[image]

### Basic

[image]

### Identify

[image]

### Power configuration

[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2022-11-05:
Hello, still missing the cluster list, to see wich ones the devices is able to use.

You can too share the DDF you are using, even it work for only 1 button, I will adapt it.
Can try that to see if we can skip the use of the button map

```
		{
			"name": "state/buttonevent",
			"awake": true,
			"parse": {
				"cl": "0x0501",
				"ep": 1,
				"eval": "Item.val = Attr.val",
				"fn": "zcl"
			}
		},
```

With replacing the `"name": "state/buttonevent"` you already have.
You will have bad value but the result in log (if it work) can help to make a fast script.

jordanderoubaix, 2022-11-05:
Hello Smanar,

Thank you for your reply !

Sorry, I am still new with deconz GUI, where can I find the cluster list ?
Is it this one ?

[image]
[image]

Here is the DDF

`
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_0zrccfgx",
  "modelid": "TS0215A",
  "vendor": "Woox",
  "product": "R7054 4 gang",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/woox/woox_r7054_button2.json",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0500",
          "0x0501"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0x0501",
            "cmd": "0x02",
            "eval": "Item.val = 1002"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0501"
    }
  ]
}
`

I have replaced the node in the DDF, and I have the following error in the log (INFO_L2)
`
17:49:33:535 [INFO] - No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None, zclSeq: 138

17:49:33:562 Node data 0x0004740000090c83 profileId: 0x0104, clusterId: 0x0B04

17:49:33:562 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 0B05290D00, zclSeq: 43
17:49:33:562 	payload: 0b05290d00
17:49:34:208 [INFO] - No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: ARM (0x00), payload: 030064, zclSeq: 139

`

There is a lot a log in this level, so quite complicated to get only this device

Smanar, 2022-11-05:
It's just the picture with the node, the endpoints, and all clusters (the blue and grey number)

And you have just the lines used for the file buttonevent :(

>I have replaced the node in the DDF, and I have the following error in the log (INFO_L2)

So you have used this DDF ?

{
   "schema":"devcap1.schema.json",
   "manufacturername":"_TZ3000_0zrccfgx",
   "modelid":"TS0215A",
   "vendor":"Woox",
   "product":"R7054 4 gang",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_SWITCH",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint":{
            "profile":"0x0104",
            "device":"0x0401",
            "endpoint":"0x01",
            "in":[
               "0x0000",
               "0x0500",
               "0x0501"
            ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
		{
			"name": "state/buttonevent",
			"awake": true,
			"parse": {
				"cl": "0x0501",
				"ep": 1,
				"eval": "Item.val = Attr.val",
				"fn": "zcl"
			}
		},
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0501"
      }
   ]
}

jordanderoubaix, 2022-11-05:
Oh that's all !

Here is the cluster (screenshot taken before changing the DDF): 
[image]

[image]

And yes, it is this DDF

Smanar, 2022-11-06:
>And yes, it is this DDF

Oups my bad, the log can appear with "DDF", so can you enable flag "DDF" "info" ( and "info_l2" if they are not too talkative) then press a button.

If you have nothing (can be an error), I will explain you how to edit the buttonmap file, what is your OS ?

jordanderoubaix, 2022-11-06:
Hum, I have already try to update the button map file (I'm on Debian)

I have added (from the R7052 ticket): 
```
  "TuyaKeyfobMap": {
            "vendor": "Woox",
            "modelids": ["TS0215A"],
            "map": [
                [1, "0x01", "IAS_ACE", "ARM", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm day/home zones only"],
                [1, "0x01", "IAS_ACE", "ARM", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Disarm"],
                [1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],
                [1, "0x01", "IAS_ACE", "ARM", "3", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm all zones"]
            ]
        }
```

On the DDF log, nothing and on info, I have this (but I am not sure if these messages come from the button click)
```
2:12:44:993 ZCL attribute report 0xCC86ECFFFE4E9D65 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
12:12:45:792 ZCL attribute report 0xCC86ECFFFE4E9D5A for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
12:12:45:000 ZCL attribute report 0xCC86ECFFFE3CD5D3 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
12:12:50:002 ZCL attribute report 0xCC86ECFFFE4E9D65 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
```

And the info_l2
```
12:14:08:349 [INFO] - No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None, zclSeq: 191

12:14:08:996 [INFO] - No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: ARM (0x00), payload: 030064, zclSeq: 192

12:14:09:100 [INFO] - No button map for: TS0121, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 050521ED00, zclSeq: 104
```

Smanar, 2022-11-06:
Ok so I give up my idea, new DDF (classic one)

```
{
   "schema":"devcap1.schema.json",
   "manufacturername":"_TZ3000_0zrccfgx",
   "modelid":"TS0215A",
   "vendor":"Woox",
   "product":"R7054 4 gang",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_SWITCH",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint":{
            "profile":"0x0104",
            "device":"0x0401",
            "endpoint":"0x01",
            "in":[
               "0x0000",
               "0x0500",
               "0x0501"
            ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
            	"name": "state/buttonevent"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0501"
      }
   ]
}
```

And you are right for the buttonmap

```
        "TuyaKeyfobMap": {
            "vendor": "Woox",
            "modelids": ["TS0215A"],
            "map": [
                [1, "0x01", "IAS_ACE", "ARM", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm day/home zones only"],
                [1, "0x01", "IAS_ACE", "ARM", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Disarm"],
                [1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],
                [1, "0x01", "IAS_ACE", "ARM", "3", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm all zones"]
            ]
        },
```

But strange you have 

>12:14:08:349 [INFO] - No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None, zclSeq: 191

as you have 
`[1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],`

Perhaps you haven't disabled the DDF core in the DDF ?

```
            {
            	"name": "state/buttonevent"
            },
```

The buttonmap file is in the "devices" folder too.

jordanderoubaix, 2022-11-06:
I have implemented the last DDF you gave me and check the button map file. Everything seems fine. (I have only run a "hot reload on Deconz GUI to be sure).

But I still have the issue "no button map" when I click on any of the switch buttons.

```
14:43:22:627 [INFO] - No button map for: TS0215A, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: ARM (0x00), payload: 010064, zclSeq: 233
```

And I didn't understand your last question

> Perhaps you haven't disabled the DDF core in the DDF ?

Do I need to change another DDF ? Or as I have implemented the one you gave me in your last reply, it should be OK

Smanar, 2022-11-06:
>Do I need to change another DDF ? Or as I have implemented the one you gave me in your last reply, it should be OK

Yes it was disabled on last DDF

But `14:43:22:627 [INFO] - No button map for` mean there is a problem with the buttonmap file else the error will be` [INFO] - No button handler for:`

What is the path of your buttonmap file ? /usr/share/deCONZ/devices/button_maps.json ?

No error on start when loading the buttonmap file ? (typo in the file)

jordanderoubaix, 2022-11-06:
Yes the path is correct for the button_maps file. 

Where can I find error on start ?

I tried to move lines in the button map, just before the last one but it does not change anything

```
        "TuyaKeyfobMap": {
            "vendor": "Woox",
            "modelids": ["TS0215A"],
            "map": [
                [1, "0x01", "IAS_ACE", "ARM", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm day/home zones only"],
                [1, "0x01", "IAS_ACE", "ARM", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Disarm"],
                [1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],
                [1, "0x01", "IAS_ACE", "ARM", "3", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm all zones"]
            ]
        },
        "terncySd01Map": {
            "vendor": "Terncy",
            "doc": "Terncy Smart Dial SD01",
            "modelids": ["TERNCY-SD01"],
            "buttons": [
                {"S_BUTTON_1": "Action button"}
            ],
            "map": [
                [1, "0x01", "XIAOYAN", "COMMAND_0", "0x0001", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Single press"],
                [1, "0x01", "XIAOYAN", "COMMAND_0", "0x0002", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Double press"],
                [1, "0x01", "XIAOYAN", "COMMAND_0", "0x0003", "S_BUTTON_1", "S_BUTTON_ACTION_TREBLE_PRESS", "Tripple press"],
                [1, "0x01", "XIAOYAN", "COMMAND_0", "0x0004", "S_BUTTON_1", "S_BUTTON_ACTION_QUADRUPLE_PRESS", "Quadruple press"],
                [1, "0x01", "XIAOYAN", "COMMAND_20", "0x01", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Long press"],
                [1, "0x01", "XIAOYAN", "ATTRIBUTE_REPORT", "0x00", "S_BUTTON_1", "S_BUTTON_ACTION_ROTATE_CLOCKWISE", "Rotate clockwise"],
                [1, "0x01", "XIAOYAN", "ATTRIBUTE_REPORT", "0x01", "S_BUTTON_1", "S_BUTTON_ACTION_ROTATE_COUNTER_CLOCKWISE", "Rotate counter-clockwise"]
            ]
        }
```

There is no "reload" to be executed after changing the button map file ? Only restart stop / start deconz' service ?

Smanar, 2022-11-06:
Haaaaa, sorry my bad

```
    if (isTuyaManufacturerName(r->item(RAttrManufacturerName)->toString()))
    {
        // for Tuya devices use manufacturer name as modelid
        return qHash(r->item(RAttrManufacturerName)->toString());
    }
```

It's because it's a tuya device, so much clone so for them we are using the manufacture name instead

```
        "TuyaKeyfobMap": {
            "vendor": "Woox",
            "modelids": ["_TZ3000_0zrccfgx"],
            "map": [
                [1, "0x01", "IAS_ACE", "ARM", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm day/home zones only"],
                [1, "0x01", "IAS_ACE", "ARM", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Disarm"],
                [1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],
                [1, "0x01", "IAS_ACE", "ARM", "3", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm all zones"]
            ]
        },
```

jordanderoubaix, 2022-11-06:
Perfect !!

I have now all 4 values as expected for each click (1002/2002/3002/4002) !

Thank you for your help !

Do you need any more info from my side ?

Smanar, 2022-11-06:
Nope, it's fine, PR sumbitted here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6506

closed 2022-11-08 by manup