dresden-elektronik/deconz-rest-plugin #6499: Aubess 20A TS011F _TZ3000_hdopuwv6

mcormick, 2022-11-05:
## Device

- Product name: Aubess 20A Smart Socket
- Manufacturer: _TZ3000_hdopuwv6
- Model identifier: TS011F
- Device type : Smart Plug

## Screenshots
[image]

### Basic
[image]
[image]

### Identify
[image]

### On/Off
[image]

### Simple Metering
[image]

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2022-11-06:
I think you can have some information here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6423

Make a try with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_hdopuwv6",
  "modelid": "TS011F",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "read": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          },
          "refresh.interval": 100
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "read": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          },
          "refresh.interval": 100
        },
        {
          "name": "state/voltage",
          "read": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          },
          "refresh.interval": 100
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "read": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 0,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 0,
            "eval": "Item.val = Attr.val * 10"
          },
          "refresh.interval": 10
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

rejoe2, 2022-11-14:
This json basically seems to work, but the current value is a bit strange. This is how the measurement is interpreted in FHEM (power with 6W is realistic): 

current | 40 | 2022-11-14 20:21:37
lastseen | 2022-11-14T19:21Z | 2022-11-14 20:21:37
power | 6 | 2022-11-14 20:21:37
reachable | 1 | 2022-11-14 20:21:37
voltage | 232 | 2022-11-14 20:21:37

Smanar, 2022-11-14:
6W is a little, it s like a LED bulb, agree with that ?
For the current you need to have something around 0.02  A

So we can divide tha vlaue by  1000 to have 0.04 but not realy precise.

Perhaps with a device wiht more consumption ?

To divide the value by 1000

```
        {
          "name": "state/current",
          "read": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 1000"
          },
          "refresh.interval": 100
        },
```

Can make direclty with the editor in the GUI.

rejoe2, 2022-11-14:
I'm deconz running headless on a non standard port...
Changed it by editing the json directly and restart the service. Wattage now shows 61 (different bulb, now classical 60W instead of the LED bulb from my first post, so both values should be correct).

Now current is shown as "0", so there's some problem in the calculation (didn't look into the log until now). Typo?

Smanar, 2022-11-14:
Yep, it's for compatibilities, It only report integer, so you need to have at least 1 A

rejoe2, 2022-11-14:
OK, so the last proposal with the 1000 division seems to be ok.

Do you need any additional info from my side or is it sufficient what we know by now?

rejoe2, 2022-11-14:
Or might that be integrated in some similar device? (didn't check if there's already a suitable one)

Smanar, 2022-11-15:
Ok so my bad, have take a look in the API and current is in mA

<html><body>

current | UInt16 | The measured current (in mA). | RW
-- | -- | -- | --

</body>
</html>

So I will remove the / 1000

https://dresden-elektronik.github.io/deconz-rest-doc/devices/innr/sp_120_plug/#state-attributes_1

And there is a Aubess folder, So I will make a PR just for this device https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6537

closed 2022-12-15 by github-actions[bot]