dresden-elektronik/deconz-rest-plugin #6501: ZHAThermostat change schedule is using wrong http method

luka6000, 2022-11-06:
## Describe the bug

ZHAThermostat sensor is using method POST to configure schedule but the docs says it should be PUT as described here https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/#change-sensor-config
Method POST is not documented at all.

## Steps to reproduce the behavior

I'm trying to update Danfoss Ally thermostats schedule with HA deconz.configure service. This service is using PUT method to update sensor config. And I get error from deconz
```
pydeconz.errors.RequestError: 4 /sensors/86/config/schedule/W124 method, PUT, not available for resource, /sensors/86/config/schedule/W124
```

This can also be replicated with POSTMAN
[image]

Works ok with POST but this is not documented so not implemented in HA service
[image]

## Expected behavior

Sensor schedule configuration (and all configurations) should be possible with PUT method

## Screenshots

## Environment

 - Host system: Raspberry Pi4 32bit
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: 2.19.1
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? esphome board

## deCONZ Logs

## Additional context

Smanar, 2022-11-06:
Ha right and it's alone

```
    // POST, DELETE /api/<apikey>/sensors/<id>/config/schedule/Wbbb
    else if ((req.path.size() == 7) && (req.hdr.method() == QLatin1String("POST") || req.hdr.method() == QLatin1String("DELETE")) && (req.path[4] == QLatin1String("config")) && (req.path[5] == QLatin1String("schedule")))
    {
        return changeThermostatSchedule(req, rsp);
    }
```

luka6000, 2022-11-28:
@Smanar would you allow something non breaking, like
```
    // PUT, POST, DELETE /api/<apikey>/sensors/<id>/config/schedule/Wbbb
    else if ((req.path.size() == 7) && (req.hdr.method() == QLatin1String("PUT") || req.hdr.method() == QLatin1String("POST") || req.hdr.method() == QLatin1String("DELETE")) && (req.path[4] == QLatin1String("config")) && (req.path[5] == QLatin1String("schedule")))
    {
        return changeThermostatSchedule(req, rsp);
    }
```
?

Smanar, 2022-11-28:
Long story, but for the moment it will not move.
From @ebaauw this request is used only to add or remove a day, not to change it, so for him it's POST or DELETE but not PUT

closed 2022-12-28 by github-actions[bot]