dresden-elektronik/deconz-rest-plugin #6504: Niko Connectable Motor Control 3A (552-72311)

jecako, 2022-11-06:
## Device

- Product name: Niko Connectable Motor Control 3A (552-72311)
- Manufacturer: NIKO NV
- Model identifier: Connectable Motor Control,3A 
- Device type : 
  - Switch (see: https://www.niko.eu/en/article/552-72311)

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2022-11-06:
Please add all screenshots.

jecako, 2022-11-06:
> Please add all screenshots.

Done!

Mimiix, 2022-11-06:
Please make sure all attributes are showing . That's at least not the case for the window covering cluster.

Can you also provide a overview of all clusters ?

Smanar, 2022-11-06:
But the device don't work natively ? all seem fine with this device.

closed 2022-12-06 by github-actions[bot]