dresden-elektronik/deconz-rest-plugin #6505: Girier Tuya Zigbee 3.0 Switch Module 2CH no neutral

Marzmarz1, 2022-11-06:
## Device

- Product name: Girier Tuya Zigbee 3.0 Switch Module 2CH
- Manufacturer: _TZ3000_jl7qyupf
- Model identifier: TS0012
- Device type : 
    - Switch
  

## Screenshots
[image]

### Basic
[image]
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]
[image]
### Scenes
[image]
[image]
[image]
[image]

### On/Off
01
[image]
[image]

02
[image]
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2022-11-06:
Hello can try this DDF

```
{
   "schema":"devcap1.schema.json",
   "manufacturername":"_TZ3000_jl7qyupf",
   "modelid":"TS0012",
   "product":"Girier Tuya Zigbee 3.0 Switch Module 2CH no neutral",
   "sleeper":false,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_ON_OFF_LIGHT",
         "restapi":"/lights",
         "uuid":[
            "$address.ext",
            "0x01"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"state/on",
               "refresh.interval":5
            },
            {
               "name":"state/reachable"
            }
         ]
      },
      {
         "type":"$TYPE_ON_OFF_OUTPUT",
         "restapi":"/lights",
         "uuid":[
            "$address.ext",
            "0x02"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"state/on",
               "refresh.interval":5
            },
            {
               "name":"state/reachable"
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "dst.ep":1,
         "cl":"0x0006",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x10",
               "min":1,
               "max":300
            }
         ]
      },
      {
         "bind":"unicast",
         "src.ep":2,
         "dst.ep":1,
         "cl":"0x0006",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x10",
               "min":1,
               "max":300
            }
         ]
      }
   ]
}
```

If it don't work can try direclty with the API to see if the device use the cluster 0x0006, just test the "on" and "off" command.

Marzmarz1, 2022-11-06:
Hello, 
I don't how I get this DDF file direct in the node, so tried to compile the file my self.  I dragged the on off lights under the device tab (2 times; for switch 1 and switch 2). and I gave switch 2 another number (0x02)
Here are some screenshots:
[image]

[image]

[image]

So far so good. But i'm missing the last part of your DDF. The text with "bindings". 
[image]

Here is my json text:
[image]
[image]

In Phoncon I see 2 lights : 
[image]
(don't mind the names)
but in HA, I only see one light: 
[image]
But when I open this light, I see 2 controls. 
[image]
And now the strange part, when I switch plug 1, in no time plug 2 is automatically also switched on. And when I switch plug 1 off, switch 2 automatically switches off.
Can somebody give me some help to solve this? I would like to get 2 seperate switches, so I can make some automations.

Smanar, 2022-11-07:
Yep, you miss the binding, but will see this part after, without it you will just miss state return, but you need to be able to send request.

In fact your device is special, it have too 2 manufacture clusters, and perhaps it use them instead of using the cluster we are using in the DDF (the 0x0006)

So to be sure, can you use the GUI, select the node, select the endpoint 01(it's the big number), select the cluster "on/off" the 0x0006, then in the cluster panel try the "on" and "off" request ?

To see if using the GUI you have the same issue ?

Marzmarz1, 2022-11-07:
I've used the GUI and on both endpoints (01 & 02) and the on/off requests working separetely. In the GUI i don't have the issue that both lights are switching on and off together.
On/off properties01:
[image]

On/off properties 02:
[image]
As you can see, they are exactly the same.

Smanar, 2022-11-08:
So if it work with GUI it's a good news.

Try to make the DDF more selective
- select the state/on
- fill value in "parse" and read", with the good endpoint (1 for the first one and 2 for the second one) instead of "auto"
- save the DDF
- make hot reload

Marzmarz1, 2022-11-08:
> I've used the GUI and on both endpoints (01 & 02) and the on/off requests working separetely. In the GUI i don't have the issue that both lights are switching on and off together. On/off properties01: [image]
> 
> On/off properties 02: [image] As you can see, they are exactly the same.

I was wrong. the 2 seprate connections are working as one. Your solution is not working. I can't switch them separately.  I'm back at quare one...

Smanar, 2022-11-08:
Have found some information.

And unfortunately this device need a special sequence to be unlocked.
So not possible using only DDF.

closed 2022-12-08 by github-actions[bot]