dresden-elektronik/deconz-rest-plugin #6512: Xiaomi Aqara relay controller lumi.relay.c2acn01 stopped working

espen4001, 2022-11-07:
## Describe the bug

Xiaomi Aqara relay controller lumi.relay.c2acn01, has stopped working. In previous versions this was working somewhat OK, being able to turn on/off both relays as lights via API. Not sure which version it stopped working, but I see it was done some work on it in version 2.05.79.
Tried modifying DDF in GUI, but file is not loaded when container is restarted/recreated, even if file is persistent in the correct location - /home/deconz/.local/share/dresden-elektronik/deCONZ/devices

## Steps to reproduce the behavior

Device is showing in the deCONZ VNC-viewer, and it is possible to turn on/off/toggle from cluster menu. But device not listed in api/lights/

## Expected behavior

Ability to toggle on/off from API.
Device listed when doing API-call for lights
Device shown as light in Phoscon App, as it was earlier.

## Screenshots

[image]
[image]

## Environment

 - Host system: Ubuntu VM, ESXi
 - Running method: Ubuntu Docker container, deconzcommunity image
 - Firmware version: 26780700
 - deCONZ version: 2.19.1
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
[_deconz_logs-2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9955895/_deconz_logs-2.txt)

## Additional context

Smanar, 2022-11-08:
Probably same issue than here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6464

Can try this DDF (power measurement is not enabled)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.relay.c2acn01",
  "product": "Xiaomi Aqara relay controller",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
     {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/temperature"
        },
        {
          "name": "state/current"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power"
        },
        {
          "name": "state/voltage"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

>Tried modifying DDF in GUI, but file is not loaded when container is restarted/recreated, even if file is persistent in the correct location - /home/deconz/.local/share/dresden-elektronik/deCONZ/devices

What is your OS ?
On HA the persistent folder is `/data/.local/share/dresden-elektronik/deCONZ/devices/` cf https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

espen4001, 2022-11-21:
I am sorry for the late reply. I missed out on the notification from your comment. I will try this and let you know! Thank you.

closed 2022-12-21 by github-actions[bot]

Smanar, 2023-02-12:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.relay.c2acn01",
  "product": "wireless relay with neutral",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
   {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
     },
     {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x000C"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x000C"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/temperature"
        },
        {
          "name": "state/current",
          "parse": {
            "at": "0x00F7",
            "eval": "Item.val = Math.round(Attr.val);",
            "fn": "xiaomi:special",
            "idx": "0x97"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 10,
          "read": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 1,
            "eval": "Item.val = Math.round(Attr.val);"
          }
        },
        {
          "name": "state/voltage",
          "parse": {
            "at": "0x00F7",
            "eval": "Item.val = Math.round(Attr.val / 10);",
            "fn": "xiaomi:special",
            "idx": "0x96"
          },
          "read": {
            "fn": "none"
          }
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x000C"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x000C"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/temperature"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 300,
          "read": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 1,
            "eval": "Item.val = Math.round(Attr.val * 1000);"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 300
        }
      ]
    }
  ]
}
```