dresden-elektronik/deconz-rest-plugin #6515: Tuya Zigbee Smoke Detector

Valader, 2022-11-08:
## Device

- Product name: Tuya Zigbee Smoke Sensor
- Manufacturer: _TZE200_dq1mfjug
- Model identifier: TS0601
- Device type :   - Sensor   - Siren

bought at: https://de.aliexpress.com/item/1005003564649452.html

## Screenshots

### Basic
[image]
[image]
[image]

### Node Info

[image]

### Working Device Temperature and Humidity from Aqara
### Basic

[image]
### Node Info

[image]

Smanar, 2022-11-09:
Hello can you try this DDf

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_dq1mfjug",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Smoke sensor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 15, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire"
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val == 1 ? 1 : 0;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        }
      ]
    }
  ]
}
```

Valader, 2022-11-10:
thanks for the hint. How can I test this?

Smanar, 2022-11-10:
Depend of OS.
If you have HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
Else you just need to create a text file called what_you_want.json with this contain in a folder "devices" 

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

Then re-include the device

Valader, 2022-11-10:
Thanks. What's the problem now?

[image]

[image]

[image]

**Here I want to include the new file...**

[image]

**...then the error message follows**

[image]

Smanar, 2022-11-11:
:(, was my bad 2 typo in the same DDF, sorry have made it with copy/paste without test.

I have updated it on the previous post.

Valader, 2022-11-11:
No problem. Now it looks like this... What is the next step?

[image]
[image]

[image]

[image]

**It is now also recognized in the Phoscon app but not listed yet**

[image]

[image]

Smanar, 2022-11-12:
Fire sensor are not visible in phoscon.
So need to use third app or look direclty in the API (like you make using help/API Information)

But there is "last updated" to "none" so the device have probably never send a notification.

I have just found that https://github.com/Koenkk/zigbee2mqtt/issues/11119#issuecomment-1112488421

Perhaps this device don't use same DP than the one I m using.

To be sure there is only way.
Enable log in deconz/help/debug view with flag "info", make the device react, (with real smoke or button test)
And look for the changed log starting by "TY_DATA_"

Valader, 2022-11-12:
Oh, thank you. Have triggered an alarm with a button and get this information:

[image]

Smanar, 2022-11-13:
It's perfect, it's exactly what we need.

So the device is using :
- the dpid 0x65 a bool value ( = 101)
- the dpid 0x01 enum value ( = 1)
- the dpid 0x0B bitmap value ( = 11)
- the dpid 0x0C a bool value again (= 12)

On the Z2M link he have 0x65 = 0 and you = 1 (because you have trigger a detection), all other values are for him, so I think it's the value to use for fire detection.

So you edit edit the DDF to use the value 101 here
```

        {
          "name": "state/fire",
          "parse": {"fn": "tuya", "dpid": 101, "eval": "Item.val == 1 ? 1 : 0;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
```

Now for the battery is another problem, you haven't an used battery to make the same thing ?

Valader, 2022-11-13:
Now:

{
    "13:49:06:939": {
        "config": {
            "battery": 0,
            "on": true,
            "pending": [],
            "reachable": false
        },
        "e": "changed",
        "id": "11",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    },
    "14:17:53:252": {
        "config": {
            "battery": 0,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "e": "changed",
        "id": "11",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    }
}

Smanar, 2022-11-13:
>Now:

You don't have notification with state / fire when making the same thing that you have done before ?

Need to make "hot relaod" in the ddf editor menu after editing the DDF, or relaod deconz.

Valader, 2022-11-13:
17:27:06:715 TY_DATA_RESPONSE: seq 1, dpid: 0x65, type: 0x01, length: 1, val: 1
17:27:06:716 TY_DATA_RESPONSE: seq 1, dpid: 0x65, type: 0x01, length: 1, val: 1
17:27:06:747 Daylight now: dusk, status: 210, daylight: 0, dark: 1
17:27:06:748 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
17:27:06:749 Skip automatic channel change, TODO warn user
17:27:06:760 GW firmware version: 0x26720700
17:27:06:763 GW firmware version is up to date: 0x26720700
17:27:06:923 APS-DATA.indication from child 0xC7B2
17:27:06:926 TY_DATA_RESPONSE: seq 2, dpid: 0x01, type: 0x04, length: 1, val: 0
17:27:06:926 TY_DATA_RESPONSE: seq 2, dpid: 0x01, type: 0x04, length: 1, val: 0
17:27:07:933 APS-DATA.indication from child 0xC7B2
17:27:07:936 TY_DATA_RESPONSE: seq 3, dpid: 0x65, type: 0x01, length: 1, val: 1
17:27:07:937 TY_DATA_RESPONSE: seq 3, dpid: 0x65, type: 0x01, length: 1, val: 1
17:27:08:358 APS-DATA.indication from child 0xC7B2
17:27:08:365 TY_DATA_RESPONSE: seq 4, dpid: 0x65, type: 0x01, length: 1, val: 1
17:27:08:368 TY_DATA_RESPONSE: seq 4, dpid: 0x65, type: 0x01, length: 1, val: 1
17:27:08:949 APS-DATA.indication from child 0xC7B2
17:27:08:952 TY_DATA_RESPONSE: seq 5, dpid: 0x01, type: 0x04, length: 1, val: 0
17:27:08:953 TY_DATA_RESPONSE: seq 5, dpid: 0x01, type: 0x04, length: 1, val: 0
17:27:09:969 APS-DATA.indication from child 0xC7B2
17:27:09:972 TY_DATA_RESPONSE: seq 6, dpid: 0x0B, type: 0x05, length: 1, val: 0
17:27:09:973 TY_DATA_RESPONSE: seq 6, dpid: 0x0B, type: 0x05, length: 1, val: 0
17:27:10:987 APS-DATA.indication from child 0xC7B2
17:27:10:992 TY_DATA_RESPONSE: seq 7, dpid: 0x0C, type: 0x01, length: 1, val: 1
17:27:10:993 TY_DATA_RESPONSE: seq 7, dpid: 0x0C, type: 0x01, length: 1, val: 1
17:27:11:303 APS-DATA.indication from child 0xC7B2
17:27:12:011 APS-DATA.indication from child 0xC7B2
17:27:12:019 TY_DATA_RESPONSE: seq 8, dpid: 0x0E, type: 0x04, length: 1, val: 2
17:27:12:021 TY_DATA_RESPONSE: seq 8, dpid: 0x0E, type: 0x04, length: 1, val: 2
17:27:13:025 APS-DATA.indication from child 0xC7B2
17:27:13:028 TY_DATA_RESPONSE: seq 8, dpid: 0x0E, type: 0x04, length: 1, val: 2
17:27:13:029 TY_DATA_RESPONSE: seq 8, dpid: 0x0E, type: 0x04, length: 1, val: 2
17:27:14:043 APS-DATA.indication from child 0xC7B2
17:27:14:050 TY_DATA_RESPONSE: seq 9, dpid: 0x10, type: 0x01, length: 1, val: 0
17:27:14:052 TY_DATA_RESPONSE: seq 9, dpid: 0x10, type: 0x01, length: 1, val: 0
17:27:16:748 Daylight now: dusk, status: 210, daylight: 0, dark: 1
17:27:16:749 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
17:27:16:750 Skip automatic channel change, TODO warn user
17:27:21:747 DB save zll database items 0x00000081
17:27:21:749 DB saved in 1 ms
17:27:23:175 Websocket 192.168.0.60:60884 send message: {"attr":{"id":"1","lastannounced":null,"lastseen":"2022-11-13T16:27Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","swversion":"0x26720700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:08:24:a4-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:08:24:a4-01"} (ret = -521768192)
17:27:23:747 Idle timer triggered
17:27:23:964 APS-DATA.indication from child 0xC7B2
17:27:23:968 TY_DATA_RESPONSE: seq 10, dpid: 0x65, type: 0x01, length: 1, val: 0
17:27:23:969 TY_DATA_RESPONSE: seq 10, dpid: 0x65, type: 0x01, length: 1, val: 0
17:27:24:988 APS-DATA.indication from child 0xC7B2
17:27:24:995 TY_DATA_RESPONSE: seq 10, dpid: 0x65, type: 0x01, length: 1, val: 0
17:27:24:998 TY_DATA_RESPONSE: seq 10, dpid: 0x65, type: 0x01, length: 1, val: 0
17:27:26:008 APS-DATA.indication from child 0xC7B2
17:27:26:013 TY_DATA_RESPONSE: seq 11, dpid: 0x01, type: 0x04, length: 1, val: 1
17:27:26:014 TY_DATA_RESPONSE: seq 11, dpid: 0x01, type: 0x04, length: 1, val: 1
17:27:26:747 Daylight now: dusk, status: 210, daylight: 0, dark: 1
17:27:26:749 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
17:27:26:751 Skip automatic channel change, TODO warn user
17:27:27:029 APS-DATA.indication from child 0xC7B2
17:27:27:034 TY_DATA_RESPONSE: seq 12, dpid: 0x0B, type: 0x05, length: 1, val: 0
17:27:27:035 TY_DATA_RESPONSE: seq 12, dpid: 0x0B, type: 0x05, length: 1, val: 0
17:27:28:045 APS-DATA.indication from child 0xC7B2
17:27:28:052 TY_DATA_RESPONSE: seq 13, dpid: 0x0C, type: 0x01, length: 1, val: 1
17:27:28:054 TY_DATA_RESPONSE: seq 13, dpid: 0x0C, type: 0x01, length: 1, val: 1
17:27:29:063 APS-DATA.indication from child 0xC7B2
17:27:29:066 TY_DATA_RESPONSE: seq 14, dpid: 0x0E, type: 0x04, length: 1, val: 2
17:27:29:067 TY_DATA_RESPONSE: seq 14, dpid: 0x0E, type: 0x04, length: 1, val: 2
17:27:30:081 APS-DATA.indication from child 0xC7B2
17:27:30:088 TY_DATA_RESPONSE: seq 15, dpid: 0x10, type: 0x01, length: 1, val: 0
17:27:30:090 TY_DATA_RESPONSE: seq 15, dpid: 0x10, type: 0x01, length: 1, val: 0
17:27:31:112 APS-DATA.indication from child 0xC7B2
17:27:31:115 TY_DATA_RESPONSE: seq 15, dpid: 0x10, type: 0x01, length: 1, val: 0
17:27:31:116 TY_DATA_RESPONSE: seq 15, dpid: 0x10, type: 0x01, length: 1, val: 0
17:27:36:747 Daylight now: dusk, status: 210, daylight: 0, dark: 1
17:27:36:749 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
17:27:36:750 Skip automatic channel change, TODO warn user
17:27:46:522 APS-DATA.indication from child 0xC7B2
17:27:46:747 Daylight now: dusk, status: 210, daylight: 0, dark: 1
[... cut ...]
17:27:56:750 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
17:27:56:751 Skip automatic channel change, TODO warn user
17:28:06:715 Websocket 192.168.0.60:60884 send message: {"attr":{"id":"11","lastannounced":"2022-11-13T16:24:42Z","lastseen":"2022-11-13T16:28Z","manufacturername":"_TZE200_dq1mfjug","modelid":"TS0601","name":"Fire 11","swversion":null,"type":"ZHAFire","uniqueid":"a4:c1:38:1a:40:d3:3a:75-01-ef00"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"a4:c1:38:1a:40:d3:3a:75-01-ef00"} (ret = -1092804456)

Smanar, 2022-11-13:
BTW you have a setting issue too

>17:27:06:748 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration

You haven't an old backup somewhere ? Else can set the value using the deconz setting window (with pressing F9), need to have something like here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help#in-case-the-network-does-still-not-come-up

And BTW the DDF is not working, try this one pls.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_dq1mfjug",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Smoke sensor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 15, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire",
          "parse": {"fn": "tuya", "dpid": 101, "eval": "Item.val = (Attr.val == 1);" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        }
      ]
    }
  ]
}
```

And have new dpid

>17:27:29:066 TY_DATA_RESPONSE: seq 14, dpid: 0x0E, type: 0x04, length: 1, val: 2
17:27:30:088 TY_DATA_RESPONSE: seq 15, dpid: 0x10, type: 0x01, length: 1, val: 0

So the device is using :

-    the dpid 0x65 a bool value ( = 101)
-    the dpid 0x01 enum value ( = 1)
-    the dpid 0x0B bitmap value ( = 11)
-    the dpid 0x0C a bool value again (= 12)
-    the dpid 0x0E a enum value again (= 14)
-    the dpid 0x10 a bool value again (= 16)

Valader, 2022-11-14:
Thank you very much for being so patient with me... (I don’t speak English or almost no English and I have to translate it first) 

I followed the instructions...
Do I always have to re-integrate the sensor?

Debug Info:

12:37:09:748 Skip automatic channel change, TODO warn user
12:37:14:385 TY_DATA_RESPONSE: seq 7, dpid: 0x65, type: 0x01, length: 1, val: 1
12:37:14:387 TY_DATA_RESPONSE: seq 7, dpid: 0x65, type: 0x01, length: 1, val: 1
12:37:14:597 TY_DATA_RESPONSE: seq 8, dpid: 0x65, type: 0x01, length: 1, val: 1
12:37:14:599 TY_DATA_RESPONSE: seq 8, dpid: 0x65, type: 0x01, length: 1, val: 1
12:37:15:197 TY_DATA_RESPONSE: seq 9, dpid: 0x01, type: 0x04, length: 1, val: 0
12:37:15:198 TY_DATA_RESPONSE: seq 9, dpid: 0x01, type: 0x04, length: 1, val: 0
12:37:16:213 TY_DATA_RESPONSE: seq 10, dpid: 0x0B, type: 0x05, length: 1, val: 0
12:37:16:213 TY_DATA_RESPONSE: seq 10, dpid: 0x0B, type: 0x05, length: 1, val: 0
12:37:17:237 TY_DATA_RESPONSE: seq 11, dpid: 0x0C, type: 0x01, length: 1, val: 1
12:37:17:238 TY_DATA_RESPONSE: seq 11, dpid: 0x0C, type: 0x01, length: 1, val: 1
12:37:18:255 TY_DATA_RESPONSE: seq 12, dpid: 0x0E, type: 0x04, length: 1, val: 2
12:37:18:255 TY_DATA_RESPONSE: seq 12, dpid: 0x0E, type: 0x04, length: 1, val: 2
12:37:19:280 TY_DATA_RESPONSE: seq 13, dpid: 0x10, type: 0x01, length: 1, val: 0
12:37:19:280 TY_DATA_RESPONSE: seq 13, dpid: 0x10, type: 0x01, length: 1, val: 0
12:37:19:748 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
12:37:19:748 Skip automatic channel change, TODO warn user
12:37:22:761 TY_DATA_RESPONSE: seq 14, dpid: 0x65, type: 0x01, length: 1, val: 1
12:37:22:762 TY_DATA_RESPONSE: seq 14, dpid: 0x65, type: 0x01, length: 1, val: 1
12:37:23:370 TY_DATA_RESPONSE: seq 15, dpid: 0x01, type: 0x04, length: 1, val: 0
12:37:23:371 TY_DATA_RESPONSE: seq 15, dpid: 0x01, type: 0x04, length: 1, val: 0
12:37:24:379 TY_DATA_RESPONSE: seq 16, dpid: 0x0B, type: 0x05, length: 1, val: 0
12:37:24:380 TY_DATA_RESPONSE: seq 16, dpid: 0x0B, type: 0x05, length: 1, val: 0
12:37:25:393 TY_DATA_RESPONSE: seq 17, dpid: 0x0C, type: 0x01, length: 1, val: 1
12:37:25:393 TY_DATA_RESPONSE: seq 17, dpid: 0x0C, type: 0x01, length: 1, val: 1
12:37:26:417 TY_DATA_RESPONSE: seq 18, dpid: 0x0E, type: 0x04, length: 1, val: 2
12:37:26:418 TY_DATA_RESPONSE: seq 18, dpid: 0x0E, type: 0x04, length: 1, val: 2
12:37:27:433 TY_DATA_RESPONSE: seq 18, dpid: 0x0E, type: 0x04, length: 1, val: 2
12:37:27:434 TY_DATA_RESPONSE: seq 18, dpid: 0x0E, type: 0x04, length: 1, val: 2
12:37:28:462 TY_DATA_RESPONSE: seq 19, dpid: 0x10, type: 0x01, length: 1, val: 0
12:37:28:463 TY_DATA_RESPONSE: seq 19, dpid: 0x10, type: 0x01, length: 1, val: 0
12:37:29:748 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
12:37:29:749 Skip automatic channel change, TODO warn user
12:37:39:748 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
12:37:39:749 Skip automatic channel change, TODO warn user
12:37:43:929 TY_DATA_RESPONSE: seq 20, dpid: 0x65, type: 0x01, length: 1, val: 0
12:37:43:931 TY_DATA_RESPONSE: seq 20, dpid: 0x65, type: 0x01, length: 1, val: 0
12:37:44:749 GW firmware version is up to date: 0x26720700
12:37:44:950 TY_DATA_RESPONSE: seq 20, dpid: 0x65, type: 0x01, length: 1, val: 0
12:37:44:952 TY_DATA_RESPONSE: seq 20, dpid: 0x65, type: 0x01, length: 1, val: 0
12:37:45:971 TY_DATA_RESPONSE: seq 21, dpid: 0x01, type: 0x04, length: 1, val: 1
12:37:45:971 TY_DATA_RESPONSE: seq 21, dpid: 0x01, type: 0x04, length: 1, val: 1
12:37:46:988 TY_DATA_RESPONSE: seq 22, dpid: 0x0B, type: 0x05, length: 1, val: 0
12:37:46:989 TY_DATA_RESPONSE: seq 22, dpid: 0x0B, type: 0x05, length: 1, val: 0
12:37:48:007 TY_DATA_RESPONSE: seq 23, dpid: 0x0C, type: 0x01, length: 1, val: 1
12:37:48:008 TY_DATA_RESPONSE: seq 23, dpid: 0x0C, type: 0x01, length: 1, val: 1
12:37:49:032 TY_DATA_RESPONSE: seq 24, dpid: 0x0E, type: 0x04, length: 1, val: 2
12:37:49:033 TY_DATA_RESPONSE: seq 24, dpid: 0x0E, type: 0x04, length: 1, val: 2
12:37:49:748 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
12:37:49:749 Skip automatic channel change, TODO warn user
12:37:50:050 TY_DATA_RESPONSE: seq 25, dpid: 0x10, type: 0x01, length: 1, val: 0
12:37:50:051 TY_DATA_RESPONSE: seq 25, dpid: 0x10, type: 0x01, length: 1, val: 0
12:37:51:071 TY_DATA_RESPONSE: seq 25, dpid: 0x10, type: 0x01, length: 1, val: 0
12:37:51:071 TY_DATA_RESPONSE: seq 25, dpid: 0x10, type: 0x01, length: 1, val: 0
12:37:59:748 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
12:37:59:748 Skip automatic channel change, TODO warn user
12:38:09:748 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
12:38:09:748 Skip automatic channel change, TODO warn user

In the Phoscon App: (Rest API)

{
    "config": {
[... 1398 lines cut ...]
        "e": "changed",
        "id": "11",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    },
    "14:04:15:548": {
        "attr": {
            "id": "11",
            "lastannounced": "2022-11-14T12:57:34Z",
            "lastseen": "2022-11-14T14:04Z",
            "manufacturername": "_TZE200_dq1mfjug",
            "modelid": "TS0601",
            "name": "Fire 11",
            "swversion": null,
            "type": "ZHAFire",
            "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
        },
        "e": "changed",
        "id": "11",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    }
}

Smanar, 2022-11-14:
> 12:37:14:385 TY_DATA_RESPONSE: seq 7, dpid: 0x65, type: 0x01, length: 1, val: 1
> 12:37:44:950 TY_DATA_RESPONSE: seq 20, dpid: 0x65, type: 0x01, length: 1, val: 0

```
"state": {
"fire": false,
"lastupdated": "2022-11-14T11:37:44.951",
```

You have a lastupdated values, So I think the state/fire is working no ?

You haven't an old battery to try to find the "low battery" or the "battery level" dpid ?

BTW you have a problem in your setup (and don't worry I have an awful english too ^^)

> 12:37:29:748 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration

[image]

Valader, 2022-11-14:
I connect 2 sensors from Aqara with the adapter “DeConz” in the iobroker. 
I uninstalled the adapter Zigbee from iobroker. In addition, I always have to enter the commands in the console of the Pi:
sudo systemctl stop deconz
sudo systemctl start deconz-gui
and back again with
sudo systemctl stop deconz-gui
sudo systemctl start deconz

The screenshots from above I have always started directly from the DeConz-Gui, top right.

Now that I have e.g. stopped deconz-gui and started deconz.... I see the following in the Phoscon app under Events:

{
    "19:57:31:887": {
        "attr": {
            "id": "17",
            "lastannounced": null,
            "lastseen": "2022-11-14T18:57Z",
            "manufacturername": "_TZE200_dq1mfjug",
            "modelid": "TS0601",
            "name": "Fire 17",
            "swversion": null,
            "type": "ZHAFire",
            "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
        },
        "e": "changed",
        "id": "17",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    },
    "19:58:00:242": {
        "e": "changed",
        "id": "17",
        "r": "sensors",
        "state": {
            "fire": true,
            "lastupdated": "2022-11-14T18:58:00.533",
            "lowbattery": null
        },
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    },
    "19:58:00:355": {
        "e": "changed",
        "id": "17",
        "r": "sensors",
        "state": {
            "fire": true,
            "lastupdated": "2022-11-14T18:58:00.752",
            "lowbattery": null
        },
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    },
    "19:58:06:871": {
        "e": "changed",
        "id": "17",
        "r": "sensors",
        "state": {
            "fire": true,
            "lastupdated": "2022-11-14T18:58:07.267",
            "lowbattery": null
        },
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    },
    "19:58:30:510": {
        "e": "changed",
        "id": "17",
        "r": "sensors",
        "state": {
            "fire": false,
            "lastupdated": "2022-11-14T18:58:30.796",
            "lowbattery": null
        },
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    },
    "19:58:31:427": {
        "e": "changed",
        "id": "17",
        "r": "sensors",
        "state": {
            "fire": false,
            "lastupdated": "2022-11-14T18:58:31.824",
            "lowbattery": null
        },
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    },
    "19:58:32:441": {
        "attr": {
            "id": "17",
            "lastannounced": null,
            "lastseen": "2022-11-14T18:58Z",
            "manufacturername": "_TZE200_dq1mfjug",
            "modelid": "TS0601",
            "name": "Fire 17",
            "swversion": null,
            "type": "ZHAFire",
            "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
        },
        "e": "changed",
        "id": "17",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    },
    "19:59:33:155": {
        "attr": {
            "id": "17",
            "lastannounced": null,
            "lastseen": "2022-11-14T18:59Z",
            "manufacturername": "_TZE200_dq1mfjug",
            "modelid": "TS0601",
            "name": "Fire 17",
            "swversion": null,
            "type": "ZHAFire",
            "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
        },
        "e": "changed",
        "id": "17",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:1a:40:d3:3a:75-01-ef00"
    }
}

Valader, 2022-11-14:
In iobroker I see the following under objects:

[image]

Valader, 2022-11-14:
Alarm triggered and now stands .... 

[image]

Valader, 2022-11-14:
I have old batteries in it now, it beeps at regular intervals... How can we test this?

Smanar, 2022-11-14:
>Alarm triggered and now stands

No automatic return to false ? It can happen on some device that make a new notification at every detection but never return back, but from your log this device need to return to false itself.

>I have old batteries in it now, it beeps at regular intervals... How can we test this?

Like usual, looking for tuya return  ^^.

By deduction I think it will be this one

>12:37:26:418 TY_DATA_RESPONSE: seq 18, dpid: 0x0E, type: 0x04, length: 1, val: 2

And you will have a lower value than 2, 1 or 0

Valader, 2022-11-14:
21:18:44:574 TY_DATA_RESPONSE: seq 30, dpid: 0x01, type: 0x04, length: 1, val: 0
21:18:44:574 TY_DATA_RESPONSE: seq 30, dpid: 0x01, type: 0x04, length: 1, val: 0
21:18:45:593 APS-DATA.indication from child 0xC7B2
21:18:45:594 TY_DATA_RESPONSE: seq 31, dpid: 0x0B, type: 0x05, length: 1, val: 0
21:18:45:594 TY_DATA_RESPONSE: seq 31, dpid: 0x0B, type: 0x05, length: 1, val: 0
21:18:46:607 APS-DATA.indication from child 0xC7B2
21:18:46:608 TY_DATA_RESPONSE: seq 32, dpid: 0x0C, type: 0x01, length: 1, val: 1
21:18:46:608 TY_DATA_RESPONSE: seq 32, dpid: 0x0C, type: 0x01, length: 1, val: 1
21:18:46:750 DB save zll database items 0x00000081
21:18:46:754 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('11', 'Fire 11', 'ZHAFire', 'TS0601', '_TZE200_dq1mfjug', 'a4:c1:38:1a:40:d3:3a:75-01-ef00', '', '{"fire":true,"lastupdated":"2022-11-14T20:18:43.561","lowbattery":null}', '{"battery":0,"enrolled":0,"on":true,"pending":null,"reachable":true}', '', 'normal', '1', '2022-11-14T20:18Z', '2022-11-14T20:18:25Z')
21:18:46:894 DB saved in 144 ms
21:18:47:622 APS-DATA.indication from child 0xC7B2
21:18:47:622 TY_DATA_RESPONSE: seq 33, dpid: 0x0E, type: 0x04, length: 1, val: 0
21:18:47:623 TY_DATA_RESPONSE: seq 33, dpid: 0x0E, type: 0x04, length: 1, val: 0
21:18:47:750 Skip idle timer callback, too early: elapsed 854 msec
21:18:48:637 APS-DATA.indication from child 0xC7B2
21:18:48:638 TY_DATA_RESPONSE: seq 34, dpid: 0x10, type: 0x01, length: 1, val: 0
21:18:48:638 TY_DATA_RESPONSE: seq 34, dpid: 0x10, type: 0x01, length: 1, val: 0
21:18:49:653 APS-DATA.indication from child 0xC7B2

Smanar, 2022-11-14:
>21:18:47:622 TY_DATA_RESPONSE: seq 33, dpid: 0x0E, type: 0x04, length: 1, val: 0

And it's the only one that have changed, so I m almost sure it's this one, not the first time I see this dpid

So I think the final DDF, for state/fire and state/battery
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_dq1mfjug",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Smoke sensor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 14, "eval": "Item.val = Attr.val * 50;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire",
          "parse": {"fn": "tuya", "dpid": 101, "eval": "Item.val = (Attr.val == 1);" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

You don't need to re-include the device.

I m using the value in the 0x0E request and  * 50 so you will have as battery level 100/50/0 for values (2/1/0)

Valader, 2022-11-14:
Great, looks good. . . I don’t have any batteries, which are about 50%, but it shows 100 and 0

With good batteries:

22:34:07:116 Websocket 192.168.0.60:39184 send message: {"attr":{"id":"11","lastannounced":"2022-11-14T21:27:24Z","lastseen":"2022-11-14T21:34Z","manufacturername":"_TZE200_dq1mfjug","modelid":"TS0601","name":"Fire 11","swversion":null,"type":"ZHAFire","uniqueid":"a4:c1:38:1a:40:d3:3a:75-01-ef00"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"a4:c1:38:1a:40:d3:3a:75-01-ef00"} (ret = -1092779880)
22:34:07:746 APS-DATA.indication from child 0xC7B2
22:34:07:752 DeviceAnnce of SensorNode: 0xA4C1381A40D33A75 [1]
22:34:07:753 Websocket 192.168.0.60:39184 send message: {"attr":{"id":"11","lastannounced":"2022-11-14T21:34:07Z","lastseen":"2022-11-14T21:34Z","manufacturername":"_TZE200_dq1mfjug","modelid":"TS0601","name":"Fire 11","swversion":null,"type":"ZHAFire","uniqueid":"a4:c1:38:1a:40:d3:3a:75-01-ef00"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"a4:c1:38:1a:40:d3:3a:75-01-ef00"} (ret = -1092780768)
22:34:07:755 Websocket 192.168.0.60:39184 send message: {"config":{"battery":100,"on":true,"pending":[],"reachable":true},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"a4:c1:38:1a:40:d3:3a:75-01-ef00"} (ret = -1092780768)

**and "others"**

22:41:01:847 TY_DATA_RESPONSE: seq 11, dpid: 0x0E, type: 0x04, length: 1, val: 0
22:41:01:848 TY_DATA_RESPONSE: seq 11, dpid: 0x0E, type: 0x04, length: 1, val: 0
22:41:03:494 Websocket 192.168.0.60:39184 send message: {"config":{"battery":0,"on":true,"pending":[],"reachable":true},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"a4:c1:38:1a:40:d3:3a:75-01-ef00"} (ret = -1092780768)

Valader, 2022-11-15:
What’s the next step? The sensor should then be displayed under DeCONZ or in the Phoscon APP. From there I can also integrate it into the iobroker. Unfortunately, it no longer works in iobroker today. There was still Fire 18. So I deleted it in the iobroker. But I can not integrate the Fire 11, which is displayed in the deCONZ Gui. How this was with the Fire 18 I can’t explain myself either.

Smanar, 2022-11-15:
>I don’t have any batteries, which are about 50%, but it shows 100 and 0

Yeah it's with the value 1, probably with medium charger battery ...

>I can’t explain myself either.

Lol,me too, It was not working or are not visible ? a synchronisation issue ?

Valader, 2022-11-15:
In deCONZ the sensor is visible and when I press the Phoscon app in the upper right corner and connect it to, then the information is available in the API. Then I enter the commands in the console: stop deconz-gui and start deconz. Then I open the Phoscon app and there is no information in the API and the sensor is not listed. In addition, the sensor does not appear in the iobroker and I do not know how I could integrate it there....?

Smanar, 2022-11-16:
Ha ?
So you mean the device is visible in the API, then you restart deconz and the device is no more visible in the API ?
So it was not saved in the database ?

But it was still in deconz ?

What is your OS ? docker  ?

Valader, 2022-11-17:
Best from the start:

In order to get deCONZ-gui and the connection to the ConBee II, the following commands are entered in the console:
sudo systemctl stop deconz
sudo systemctl start deconz-gui

Then I connect via Remotdesctop to the Pi and the connection is established. There I see the sensor e.g.Fire 12. If I start the Phoscon APP from there with the button in the upper right, then I see the data from this sensor in the API.

Then I close the deCONZ-gui, disconnect the connection and enter the following commands:
sudo systemctl stop deconz-gui
sudo systemctl start deconz

Then I start the Phoscon app directly in the browser and there is no data in the API and the sensor is not present.

Here’s the thing.

I had to create a separate user on the Pi for access via Desctop. With this data I log in via Remotdesctop.

I use a Pi4 with Linux.

Yes, I have integrated 2 Aqara temperature and humidity sensors and they work perfectly. I also see this in the deCONZ-gui. Pictures of these are shown above.

Valader, 2022-11-18:
> Have you try to include the device on the headless mode ? without the GUI , one time the DDf is on a folder "devices" it need to work for both mode.

**Sorry, can I ask how to do this?**

If I see correctly, the Sonsor is in here:

[image]

> It's not just a tip ^^, you realy need to correct that.

**I’ve done that now, thank you.**

Valader, 2022-11-18:
I just want the following:
I want to be able to integrate this sensor into iobroker! So that from there I can query the status with scripts or be alerted.

Smanar, 2022-11-18:
>I want to be able to integrate this sensor into iobroker! So that from there I can query the status with scripts or be alerted.

Yeah but I don't understand why the inclusion is not memorised in the data base, probabably a "user issue".

But for me it's fine, if you put the file like on the capture, it need to be used for the user "nano" for all type of running deconz (headless and desktop), so just try including the device with this user (on headless mode if its the mode used with iobroker)

But I hope iobroker is running for the "nano" user too ? By defaut its "pi" for deconz. To use deconz with other user than "pi" there is some change to done, I hope you have correclty set your OS ?

If you type "groups" you need to see "dialout"

Valader, 2022-11-19:
@Smanar, you to the best and my hero! Thank you very much.

It was actually due to the user’s settings. After I checked that both users are present in the dialout, the Pi restarted the sensor appears now in the iobroker! Wow. Although I don’t see it in the Phoscon app, but I think that’s normal, since now deconz have to record this sensor, right?

I was also able to integrate a second sensor successfully. Question:
Where and how can I change the name (Fire XY)? In the deCONZ-gui and if so where and how exactly?

Smanar, 2022-11-19:
>I don’t see it in the Phoscon app, but I think that’s normal, since now deconz have to record this sensor,

the problem is only from Phoscon, it's a third app, but less powerfull than Iobroker.

>Where and how can I change the name (Fire XY)? In the deCONZ-gui and if so where and how exactly?

Not possible with the GUI, you need to use direclty the API, and as phoscon don't support them, you haven't the possibility using iobroker ?

else I can explain you how to do with the API ?

For information:
>v2.19.2: Name on the node in the GUI is no longer updated when name attribute is changes through the API.

So probably need a restart too.

If the DDF is working wanan make a PR to submit it, or I can make it if needed ?

Valader, 2022-11-20:
> else I can explain you how to do with the API ?

yes please

> If the DDF is working wana make a PR to submit it, or I can make it if needed ?

I don’t quite understand your question, sorry

Smanar, 2022-11-20:
You have the long sory here https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

Else can use command line (from your old capture), the device id (3) can be different now

`curl -H 'Content-Type: application/json' -X PUT -d '{"name": "newname"}' http://192.168.0.60:8080/api/01751AFD89/sensors/3`

To submit the DDF, to have it officially in the code https://github.com/dresden-elektronik/deconz-rest-plugin/pulls if you make it, it will be under your name.

Valader, 2022-11-20:
All right, Smanar.
Then can I use this command to rename each sensor?

Please do this for me.
It’s your credit, not mine!

Valader, 2022-11-22:
> Not possible with the GUI, you need to use direclty the API, and as phoscon don't support them, you haven't the possibility using iobroker

Yes, in iobroker it works. I can rename the sensors there!

Smanar, 2022-11-23:
Pr submitted https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6567

>Yes, in iobroker it works. I can rename the sensors there!

And easier than with the API ^^

Valader, 2022-11-23:
> And easier than with the API ^^

That’s right, I think so.

I would like to thank you very sincerely for all your efforts. Without your help, I wouldn’t have been able to integrate my sensors into my system. Do you think DeConz will integrate them now?

closed 2022-12-23 by github-actions[bot]